christianrauch/camera_ros #1: error building in Ubuntu 22.04 raspberry pi ARM64

anfederman, 2022-07-01:
o suppress this warning ignore these packages in the workspace:
--packages-ignore camera_calibration_parsers camera_info_manager
--- stderr: camera_ros                          
CMake Error at /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:603 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  CMakeLists.txt:19 (pkg_check_modules)

I am not sure what wasn't found everything seems to be in the CMakeLists - but  although libcamera is installed, it appears not to work.

christianrauch, 2022-07-02:
Your installation of libcamera might be installed in a non-standard path or the `libcamera.pc` file might be missing, Note that the Ubuntu package (`libcamera-dev`) installs a `camera.pc`, while the upstream project installs a `libcamera.pc`. The ROS package is looking for the `libcamera` package and not the `camera` package.

With [`colcon-meson`](https://github.com/colcon/colcon-meson), you can now build the [libcamera upstream repo](https://libcamera.org/getting-started.html) as part of your workspace. If you install  `colcon-meson` via pip and setup a workspace with `libcamera` (https://git.libcamera.org/libcamera/libcamera.git) and `camera_ros`, then this should work.

Otherwise, you can just globally install `libcamera` from https://git.libcamera.org/libcamera/libcamera.git and this should then also work.

Also, note that the libcamera API isn't stable yet and does change over time. I am rebasing and force-pushing my repo to adapt to these API changes and to incorporate other fixes. Expect the git repo and the code to break once in a while. I will change this once the API stabilises or too many people complain about breaking the git history :-)

anfederman, 2022-07-02:
Thanks. Since no one else has posted issues, I wasn't sure this repo was
still active.

I also tried the raspicam2 node, but since I can't get raspistill working,
I guess that's a non-starter. BTW I did get the Noetic raspicam_node
working on ARM64 Ubuntu 20.04 on a PI 4.

christianrauch, 2022-07-02:
The old API for interfacing with the camera on the Raspberry Pi is [deprecated](https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-and-the-legacy-raspicam-camera-stack). So this will only work for older distributions. Eventually, you will have to switch to the libcamera stack.

Anyway, since you found a workaround, I am going to close this. If you still encounter issues with building the node due to libcamera dependency issues, just reopen this.

closed 2022-07-02 by christianrauch