christianrauch/camera_ros #2: Last set of commits don't compile in Humble:-(

mw46d, 2022-08-01:
Hello, I tried to compile your sources in stock Humble and it seems, the latest set of commits does not work:-( src/CameraNode.cpp @ 6ae554895f056e4b19cfd4084f4dc1fad6f99ceb works but everything else runs into compile errors:-( 
* I have only a camera_info_manager.h (not a *.hpp)
* sensor_msgs/image_encodings.hpp seems to clash with the cv_bridge.h ... :-(

Any idea?
Thanks,
-- Marco

mw46d, 2022-08-01:
Part of the problem is, that you need the ros-*-image-common and the ros-*-cv-bridge packages instead of the OS/Ubuntu packages. At least, now I have the *.hpp;_) 
Now it compiled;-)

mw46d, 2022-08-01:
User error. Make sure, you only have the related ros packages installed, not the Ubuntu/base ones!

closed 2022-08-01 by mw46d

christianrauch, 2022-08-01:
Yeah, mixing the official ROS packages and those in Ubuntu is not a good idea, especially if they are from ROS1 and ROS2. In general, binary dependencies should be installed via rosdep (`rosdep install --from-paths src --ignore-src -y`).