christianrauch/camera_ros #4: raspbian bullseye, humble, error on node startup with arducam-pivariety camera

JohK, 2022-09-05:
I have arducam-pivariety BO0353 (AR0234) running on bullseye (verified with libcamera-raw).

Tried to run camera_ros (73f7cedf44d37a512688ac2ee0ac8069e7716a4e), but getting a runtime_error:

`
ros2 run camera_ros camera_node -p 'width:=1920' -p 'height:=1200' -p 'format:=RGB888'
[WARN] [1662365217.732113618] [rcl]: Found remap rule 'format:=RGB888'. This syntax is deprecated. Use '--ros-args --remap format:=RGB888' instead.
[65:27:37.825079472] [30485]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3875-5eb7be6a
[65:27:37.874930011] [30498]  WARN CameraSensorProperties camera_sensor_properties.cpp:174 No static properties available for 'arducam-pivariety'
[65:27:37.875071712] [30498]  WARN CameraSensorProperties camera_sensor_properties.cpp:176 Please consider updating the camera sensor properties database
[65:27:37.885341451] [30498] ERROR IPAProxy ipa_proxy.cpp:149 Configuration file 'arducam-pivariety.json' not found for IPA module 'raspberrypi'
[65:27:38.337684062] [30498]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/arducam_pivariety@c to Unicam device /dev/media2 and ISP device /dev/media3
>> cameras:
0: arducam-pivariety (/base/soc/i2c0mux/i2c@1/arducam_pivariety@c)
>> stream configurations:

[... long list of configurations, including   "- Pixelformat: RGB888 (64x64 - 1920x1200)" ...]

[65:27:38.347462273] [30485]  INFO Camera camera.cpp:1035 configuring streams: (0) 1920x1080-BGR888
[65:27:38.422188796] [30498]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/arducam_pivariety@c - Selected sensor format: 1920x1080-SGRBG10_1X10 - Selected unicam format: 1920x1080-pgAA
camera "/base/soc/i2c0mux/i2c@1/arducam_pivariety@c" configured with 1920x1080-BGR888 stream
[WARN] [1662365218.962093272] [camera]: control NoiseReductionMode (39) not handled
terminate called after throwing an instance of 'std::runtime_error'
  what():  unsupported ControlValue cast
[ros2run]: Aborted`

christianrauch, 2022-09-05:
You need the `default_info_init` branch of my fork at https://github.com/christianrauch/libcamera/tree/default_info_init. I submitted this to the `libcamera-devel` mailing list and hope this gets merged soon.

Generally, I am tracking the most recent libcamera version and I am also rebasing the `main` history until an initial release. If you continue to use this repo, you will have to hard reset from time to time.

JohK, 2022-09-06:
Ok, understood. Thank you. I have to on another project first and will revisit my ROS setup in a few days, maybe I get lucky and it's already merged then. Is there a way on the _official repo_ to follow merge requests?

JohK, 2022-09-06:
Works as expected.

closed 2022-09-06 by JohK

christianrauch, 2022-09-06:
> Ok, understood. Thank you. I have to on another project first and will revisit my ROS setup in a few days, maybe I get lucky and it's already merged then. Is there a way on the _official repo_ to follow merge requests?

For the complete discussion, you have to watch the [`libcamera-devel` mailing list](https://lists.libcamera.org/pipermail/libcamera-devel/). To get an overview of the submitted patches, you can check out [patchwork](https://patchwork.libcamera.org/project/libcamera/list/). E.g. the latest version of my patches is at https://patchwork.libcamera.org/project/libcamera/list/?series=3467. You can use the [`pwclient`](https://patchwork.libcamera.org/project/libcamera/) to apply and test patches.

Unfortunately, everything works via mailing lists. There is no web interface to follow and contribute to discussions.