christianrauch/camera_ros #5: Memory leak?

JohK, 2022-09-06:
Just noticed that the VIRT value in top was increasing monotonically for `camera_node` and it's chomping away on my free memory (slowly). A pointer that's not freed?

I attached you pmap output, is it the list of dmabuf that is growing indefinately?

[pmap.CameraNode.txt](https://github.com/christianrauch/camera_ros/files/9494875/pmap.CameraNode.txt)

christianrauch, 2022-09-06:
> I attached you pmap output, is it the list of dmabuf that is growing indefinately?

Well, you have to monitor this over time :-)

I've never used pmap to monitor memory consumption. I usually just look at htop and friends to see if a process leaks memory, and ASan and valgrind to debug this.

> Just noticed that the VIRT value in top was increasing monotonically for `camera_node` and it's chomping away on my free memory (slowly). A pointer that's not freed?

I've seen memory leaks around the publishing of images before. When I removed the image message creation, conversion and publishing, the leaks go away. But this of course defies the purpose of the ROS node. I was never able to properly pinpoint the issue.

It may also depend on the camera. I have an integrated RGB and IR camera (a.k.a. "Windows Hello"). Using the "640x480-MJPEG" stream for the IR camera causes the leak and using "1280x720-MJPEG" for the RGB camera does not leak memory. It may be as well that something else in the stack leaks.

christianrauch, 2022-09-19:
@JohK You were right about `dmabuf`. I forgot to call `munmap` on the `mmap`-ed buffers returned by libcamera. This is fixed now on the newest `main`.

You can compare this with something like `pmap $(pidof camera_node) | grep dmabuf | wc -l`, which will show the number of `dmabuf` entries in the process map. This should be 0 now. Before it was growing indefinitely. Also the total amount of memory reported by `pmap` should now be more or less constant.

closed 2022-09-19 by christianrauch