christianrauch/camera_ros #7: /usr/include/camera_info_manager/camera_info_manager.h

alain-lnpcrd, 2022-10-04:
hi,
Linux pi-desktop 5.15.0-1015-raspi #17-Ubuntu SMP PREEMPT Mon Sep 12 13:14:51 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
ubuntu 22.04
ros2 humble
In CameraNode.ccp, I have to add :
#include <sensor_msgs/msg/camera_info.hpp>, 
the header is in source /opt/ros/humble/include  otherwise th header /usr/include/sensor_msgs is used and CameraInfo is not member of sensor_msgs/msg but sensor_msg.
But, I still have errors with colcon build
`Starting >>> camera_ros
[Processing: camera_ros]                             
--- stderr: camera_ros                               
/home/pi/mes_ros2/src/camera_ros/src/CameraNode.cpp: In constructor ‘camera::CameraNode::CameraNode(const rclcpp::NodeOptions&)’:
/home/pi/mes_ros2/src/camera_ros/src/CameraNode.cpp:176:87: error: no matching function for call to ‘camera_info_manager::CameraInfoManager::CameraInfoManager(camera::CameraNode*)’
  176 | onst rclcpp::NodeOptions &options) : Node("camera", options), cim(this)
      |                                                               ^~~~~~~~~

In file included from /home/pi/mes_ros2/src/camera_ros/src/CameraNode.cpp:6:
/usr/include/camera_info_manager/camera_info_manager.h:192:3: note: candidate: ‘camera_info_manager::CameraInfoManager::CameraInfoManager(ros::NodeHandle, const string&, const string&)’
  192 |   CameraInfoManager(ros::NodeHandle nh,
      |   ^~~~~~~~~~~~~~~~~
/usr/include/camera_info_manager/camera_info_manager.h:192:37: note:   no known conversion for argument 1 from ‘camera::CameraNode*’ to ‘ros::NodeHandle’
  192 |   CameraInfoManager(ros::NodeHandle nh,
      |                     ~~~~~~~~~~~~~~~~^~
/home/pi/mes_ros2/src/camera_ros/src/CameraNode.cpp: In member function ‘void camera::CameraNode::requestComplete(libcamera::Request*)’:
/home/pi/mes_ros2/src/camera_ros/src/CameraNode.cpp:530:56: error: conversion from ‘sensor_msgs::CameraInfo’ {aka ‘sensor_msgs::CameraInfo_<std::allocator<void> >’} to non-scalar type ‘sensor_msgs::msg::CameraInfo’ {aka ‘sensor_msgs::msg::CameraInfo_<std::allocator<void> >’} requested
  530 |     sensor_msgs::msg::CameraInfo ci = cim.getCameraInfo();
      |                                       ~~~~~~~~~~~~~~~~~^~
gmake[2]: *** [CMakeFiles/camera_component.dir/build.make:76 : CMakeFiles/camera_component.dir/src/CameraNode.cpp.o] Erreur 1
gmake[1]: *** [CMakeFiles/Makefile2:248 : CMakeFiles/camera_component.dir/all] Erreur 2
gmake: *** [Makefile:146 : all] Erreur 2
---
Failed   <<< camera_ros [33.2s, exited with code 2]

Summary: 0 packages finished [34.0s]
  1 package failed: camera_ros
  1 package had stderr output: camera_ros
`

christianrauch, 2022-10-04:
You are probably mixing ROS2 from the official repo with ROS1 from the Ubuntu repo. The headers (and all files in general) from official ROS distributions are installed under `/opt/ros/humble/$ROS_DISTRO`. The header `/usr/include/camera_info_manager/camera_info_manager.h` is installed by package [`libcamera-info-manager-dev`](https://packages.ubuntu.com/jammy/amd64/libcamera-info-manager-dev/filelist).

If you install multiple packages that provide the same header and those are also in the search path, then the compiler will not know which header to use. You have to uninstall `libcamera-info-manager-dev` or otherwise make sure that `/usr/include/camera_info_manager` is not in the search path.

alain-lnpcrd, 2022-10-05:
hi,
thank you very much.
after some changes on an other SSD  with ubuntu 22.;4 anr ros2 humble more clean
```
pi@pi-desktop:~/mes_ros2$ colcon build --packages-select camera_ros
Starting >>> camera_ros
[Processing: camera_ros]                             
[Processing: camera_ros]                                     
Finished <<< camera_ros [1min 5s]                             

Summary: 1 package finished [1min 6s]
pi@pi-desktop:~/mes_ros2$ . install/setup.bash

```
```
pi@pi-desktop:~/mes_ros2$ ros2 run camera_ros  camera_node 
[0:31:35.081577843] [17657]  WARN IPAManager ipa_manager.cpp:113 Public key not valid
[0:31:35.098176560] [17657]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3910-0a4d1b8d
[0:31:35.190875574] [17671]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media1 and ISP device /dev/media2
>> cameras:
0: ov5647 (/base/soc/i2c0mux/i2c@1/ov5647@36)
>> stream configurations:
0: 1920x1080-YUV420
  - Pixelformat: NV21 (64x64 - 2592x1944)
    Sizes:
     - 160x120
     - 240x160
     - 320x240
     - 400x240
     - 480x320
     - 640x360
     - 640x480
     - 720x480
     - 768x480
     - 854x480
     - 720x576
     - 800x600
     - 960x540
     - 1024x576
     - 960x640
     - 1024x600
     - 1024x768
     - 1280x720
     - 1152x864
     - 1280x800
     - 1360x768
     - 1366x768
     - 1440x900
[......]
  - Pixelformat: UYVY (64x64 - 2592x1944)
    Sizes:
     - 160x120
     - 240x160
     - 320x240
     - 400x240
     - 480x320
     - 640x360
     - 640x480
     - 720x480
     - 768x480
     - 854x480
     - 720x576
     - 800x600
     - 960x540
     - 1024x576
     - 960x640
     - 1024x600
     - 1024x768
     - 1280x720
     - 1152x864
[... 48 lines cut ...]
[0:54:45.366599589] [17724]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media1 and ISP device /dev/media2
[0:54:45.370642612] [17714]  INFO Camera camera.cpp:1026 configuring streams: (0) 1296x972-YUV420
[0:54:45.372846734] [17724]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 19
Abandon (core dumped)
pi@pi-desktop:~/mes_ros2$ 

```

```
pi@pi-desktop:~/mes_ros2$ libcamera-hello --qt-preview
Made QT preview window
[0:34:02.517722224] [17677]  WARN IPAManager ipa_manager.cpp:113 Public key not valid
[0:34:02.522520700] [17677]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3910-0a4d1b8d
[0:34:02.597206052] [17684]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media1 and ISP device /dev/media2
[0:34:02.601270973] [17677]  INFO Camera camera.cpp:1026 configuring streams: (0) 1296x972-YUV420
[0:34:02.603464674] [17684]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
pi@pi-desktop:~/mes_ros2$ 
```
I've video with  "libcamera-hello --qt-preview"

p.s.
as said of me 55 years ago one of my teachers
An island of knowledge in an ocean of ignorance

christianrauch, 2022-10-05:
Are you still having issues with the camera node? The issue you pasted here comes from another program and is probably caused by something else.

It seems the node is now working for you. Can you close the issue if your problem is solved?

alain-lnpcrd, 2022-10-05:
No I'm still on camera_ros. I searched how to configure the camera to remove the error below but only found the package for "camera legacy".
```
[ERROR] [1664987522.360175651] [camera_calibration_parsers]: Unable to open camera calibration file [/home/pi/.ros/camera_info/ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_1920x1080.yaml]
[WARN] [1664987522.360242929] [camera]: Camera calibration file /home/pi/.ros/camera_info/ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_1920x1080.yaml not found

```
I still have to figure out if the node is indeed sending a video and how to exploit it. (I have not studied the code)
Thank you very much for the help you have given me.

christianrauch, 2022-10-05:
The warning is from the camera info manager, telling you that there are no intrinsics for the camera. If you need intrinsics, you have to calibrate the camera and store them.

If you want to read images from the camera you have to subscribe to the respective topics.

I think your remaining issues are generic ROS questions that are better asked at https://answers.ros.org. I also assume that your original issue has been solved. I am going to close this issue for now. If you are encountering issues that are specific to this repo, then please open another issue.

closed 2022-10-05 by christianrauch