christianrauch/camera_ros #8: Documentation on camera calibration file

light-tech, 2022-10-07:
I was able to [build and run the node on Raspberry Pi 3B+ (running Ubuntu 22.04)](https://github.com/light-tech/TurtleBot3-On-RPi). I can stream to the Mac but there is some warning about missing camera calibration file.

```
[INFO] [1665126984.521708471] [camera]: using default calibration URL
[INFO] [1665126984.522385146] [camera]: camera calibration URL: file:///home/USERNAME/.ros/camera_info/ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_1920x1080.yaml
[ERROR] [1665126984.524095166] [camera_calibration_parsers]: Unable to open camera calibration file [/home/USERNAME/.ros/camera_info/ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_1920x1080.yaml]
[WARN] [1665126984.524412357] [camera]: Camera calibration file /home/USERNAME/.ros/camera_info/ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_1920x1080.yaml not found
```

Is there a place to learn more about this to finetune the experience? Right now, the framerate is low at 1920x1080 (I am using raw topic `camera/image_raw` to test things out) so I want to at least reduce the resolution.

christianrauch, 2022-10-07:
You are touching two issues here:
- camera calibration
- settings

The warning "Camera calibration file [...] not found" comes from the camera info manager. It manages the camera intrinsics. Those can be provided directly by the camera hardware or stored on disk. In this case, they are stored in a yaml file, which does not exist because the camera has not been calibrated. I haven't tested this myself, but https://navigation.ros.org/tutorials/docs/camera_calibration.html seems to be a good start.

If you start the node without configuration, it will use default values. To see all available parameters, run `ros2 param list`. You can set parameters on the command line at the start or dynamically during the runtime. See https://docs.ros.org/en/foxy/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Parameters/Understanding-ROS2-Parameters.html for details. To change the resolution at the start, append `--ros-args -p width:=320 -p height:=180` to the command line.

light-tech, 2022-10-08:
Thanks for the help. Adding `--ros-args` makes the video stream smooth enough for me.

I will try out the camera calibration once I get access to a printer to print out the pattern.

closed 2022-10-08 by light-tech

christianrauch, 2022-10-08:
> I will try out the camera calibration once I get access to a printer to print out the pattern.

You can also use a screen for this :-) As long as the surface is flat and you can measure the lengths, you can use any surface for this.

light-tech, 2022-10-09:
> You can also use a screen for this :-) As long as the surface is flat and you can measure the lengths, you can use any surface for this.

That's a great idea.