christianrauch/camera_ros #9: colcon build error, camera-info-manager not found.

anfederman, 2022-12-18:
Hello, I am trying to run camera_ros on a raspberry pi 4 with the new rtos kernel.  I had i running under humble on the standard kernel, but now I am getting this during colcon build:

/home/ubuntu/ros2_ws/src/camera_ros/src/CameraNode.cpp:6:10: fatal error: camera_info_manager/camera_info_manager.hpp: No such file or directory
    6 | #include <camera_info_manager/camera_info_manager.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

next I installed:
sudo apt install ros-humble-camera-info-manager
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  kpartx liburcu8 sg3-utils-udev

I still get the error.

Any suggestions? Perhaps it is looking for camera_info_manager in the wrong location?

christianrauch, 2022-12-18:
Which Linux distribution and version are you using?

`rosdep` should resolve all the dependencies:
```
rosdep install --from-paths src --ignore-src -y
```
and after this `dpkg` will tell you that the header `camera_info_manager.hpp` is installed in package `ros-humble-camera-info-manager`:
```
$ dpkg -S /opt/ros/humble/include/camera_info_manager/camera_info_manager/camera_info_manager.hpp
ros-humble-camera-info-manager: /opt/ros/humble/include/camera_info_manager/camera_info_manager/camera_info_manager.hpp
```

Does the header `camera_info_manager.hpp` exist anywhere on your filesystem? You can check this with:
```
sudo updatedb && locate camera_info_manager/camera_info_manager.hpp
```

Can you reproduce this on a "standard PC", i.e. not the Raspberry Pi? I don't really see why the behaviour should change by switching only the kernel.

anfederman, 2022-12-19:
***@***.***:~/ros2_ws$ dpkg -S
/opt/ros/humble/include/camera_info_manager/camera_info_manager/camera_info_manager.hpp

ros-humble-camera-info-manager:
/opt/ros/humble/include/camera_info_manager/camera_info_manager/camera_info_manager.hpp

***@***.***:~/ros2_ws$ colcon build

Starting >>> camera_ros
--- stderr: camera_ros
/home/ubuntu/ros2_ws/src/camera_ros/src/CameraNode.cpp:6:10: fatal error:
camera_info_manager/camera_info_manager.hpp: No such file or directory
    6 | #include <camera_info_manager/camera_info_manager.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/camera_component.dir/build.make:76:
CMakeFiles/camera_component.dir/src/CameraNode.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:248:
CMakeFiles/camera_component.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2

So the file is therehe, but it isn't being found.
---
***@***.***:~$ uname -a
Linux ubuntu 5.15.39-rt42-raspi #1 SMP PREEMPT_RT Sat Jul 16 08:34:58 UTC
2022 aarch64 aarch64 aarch64 GNU/Linux
Failed   <<< camera_ros [1.79s, exited with code 2]

anfederman, 2022-12-19:
rosdep install -i --from-path src --rosdistro humble -y
ERROR: the following packages/stacks could not have their rosdep keys
resolved
to system dependencies:
camera_ros: Cannot locate rosdep definition for [libcamera]

and if
dpkg -L libcamera-info-manager-dev
/.
/usr
/usr/include
/usr/include/camera_info_manager
/usr/include/camera_info_manager/camera_info_manager.h
/usr/lib

christianrauch, 2022-12-19:
In future, can you use the code blocks in markdown to mark code, commands, terminal output etc? This makes it much easier to read and reproduce.

christianrauch, 2022-12-19:
> dpkg -L libcamera-info-manager-dev

This is the wrong package. You have to install `ros-humble-camera-info-manager` from the ROS repo and not `libcamera-info-manager-dev` from the Ubuntu repo. `rosdep` takes care of the official packages from the ROS repo but you have to make sure that there are no other interfering packages yourself. See https://github.com/christianrauch/camera_ros/issues/7#issuecomment-1267536568 for an earlier discussion on this.

christianrauch, 2022-12-19:
> ok by putting libcamera and camera_ros in the same workspace it built:

> You may document and close, but please put something in the git intro page.

I doubt that the issue with the included header search path for `camera_info_manager.hpp` is related to libcamera. libcamera is a ROS-independent library. You can just build and install it as suggested at https://libcamera.org/getting-started.html. The `camera_ros` package should then discover and link the library in the same way. libcamera should not change the way in which ROS headers are found.

If you can give me two full reproducible examples, one where this issue occurs with libdecor outside the workspace and one where this issue does not occur with libdecor inside the workspace, then I may look into this to debug what ultimately causes this issue.

anfederman, 2022-12-20:
I installed libcamera after I already had Ros humble desktop pre installed,
and it took a couple of tries until I got libcamera-still working. It may
have something to do with how libcamera was built. After a ros_ws colcon
build of your repo for libcamera and camera_ros it worked. So my guess is
something wrong with my original libcamera install. If I recall correctly,
there are versions from linux.tv and raspberry pi, I probably tried both
until libcamera_still worked.

My first attempt at building camera ros, I got the camera info manager not
found, so I did an apt install, but the error message persisted.

 To get it to work, I did a rm -rf ros_ws, and started fresh. Cloned
libcamera and camera_ros in the same src, did a rosdep, colcon build.
Libcamera took about 6 minutes, and camera_ros about two. I was able to
test with rqt_image, and it worked fine for the raw image.

I will install image transport next to get compressed working.

Thanks for your help.

christianrauch, 2022-12-20:
I am trying to get `libcamera` and `camera_ros` released (bloomed) to the ROS repo. This should make it much easier to install and use it.

closed 2022-12-20 by christianrauch

anfederman, 2022-12-20:
Thanks for your reply. Where are you located? My company, Ubiquity Robotics
currently runs our robots on Noetic, but interested in upgrading to humble
on the Pi 4 Arm 64 RTOS. I am not sure if you are interested in a
collaboration, but I think David would like to have a conversation with you.

Note to David, Christian just helped me get the Raspicam working on the
RTOS Pi 4 ARM64 Pi 4.

anfederman, 2022-12-21:
Unable to load plugin for transport 'image_transport/compressed_sub', error
string:
According to the loaded plugin descriptions the class
image_transport/compressed_sub with base class type
image_transport::SubscriberPlugin does not exist. Declared types are
 image_transport/raw_sub

raw image works, but colors reversed r/b?   can't transport compressed

christianrauch, 2022-12-21:
Do you have all the image-transport plugins installed? There are multiple plugins.

I am also not sure why the channels would be switched. What pixel format/encoding are you using?

anfederman, 2022-12-21:
I installed the plugins, and comprfessed now works, however, the colors are
still reversed. in camera_ros, normal in libcamera-still.