christianrauch/camera_ros #10: rgb to bgr switch in camera ros

anfederman, 2022-12-22:
I have in separate work spaces built libcamera and camera_ros fro your github.  libcamera_still displays colors properly, camera ros swaps rgb for bgr.  This is on a Pi 4 Ubuntu 22.04 RTOS ARM64 

[image]
[image]

anfederman, 2022-12-22:
[WARN] [1671735223.499671649] [camera]: no pixel format selected, using default:
 "BGR888"

is in the camera_node log file.

it gets this from my default set in config.txt

# Autoload overlays for any recognized cameras or displays that are attached
# to the CSI/DSI ports. Please note this is for libcamera support, *not* for
# the legacy camera stack
camera_auto_detect=0
display_auto_detect=1
dtoverlay=vc4-kms-v3d,cma-512
dtoverlay=imx219
dtoverlay=imx219,cam0

# Config settings specific to arm64
arm_64bit=1

christianrauch, 2022-12-22:
There is no configuration file in this repo. By default, the node will select the first available pixel format. You can change that.

What is the `encoding` of the published mage? What happens when you change the format? How can I reproduce this?

christianrauch, 2022-12-29:
I can reproduce this issue with the [`vivid` driver](https://www.linuxtv.org/wiki/index.php/VIVI) on my desktop. After loading the kernel module via `sudo modprobe vivid` the new V4L2 device will show a test pattern with coloured vertical bars (note the red, blue, black bar order). You can check this also with VLC which lets you open a capture device:
[image]

To use this with `libcamera`, you have to use the `vivid` branch from the `vivid` fork https://git.libcamera.org/libcamera/vivid.git. 

When I then use this device with the `BGR888` format:
```sh
ros2 run camera_ros camera_node --ros-args -p camera:=vivid -p format:=BGR888
```
the image will be published with the correct `bgr8` `encoding`:
```
$ ros2 topic echo --once /camera/image_raw --field encoding
bgr8
---
```
but `rqt` will show the test pattern with switched red and blue colour bars:
[image]

anfederman, 2022-12-29:
Thanks for your help  I can launch

ros2 run camera_ros camera_node --ros-args -p format:=BGR888

or

ros2 run camera_ros camera_node --ros-args -p format:=RGB888

Results

 ros2 topic echo --once /camera/image_raw --field encoding
bgr8
---
ros2 topic echo --once /camera/image_raw --field encoding
rgb8

But in either case the red/blue   in rqt_image view is reversed -

here is the config.sys settings:
# Autoload overlays for any recognized cameras or displays that are attached
# to the CSI/DSI ports. Please note this is for libcamera support, *not* for
# the legacy camera stack
camera_auto_detect=0
display_auto_detect=1
dtoverlay=vc4-kms-v3d,cma-512
dtoverlay=imx219
dtoverlay=imx219,cam0

I get a warning about no config yaml file on launch,  but it does state
format:

***@***.***:~/ros2_ws$ ros2 run camera_ros camera_node --ros-args -p
format:=RGB888
[0:11:13.262331104] [1972]  INFO Camera camera_manager.cpp:299 libcamera
v0.0.2+61-3c70cae5
[0:11:13.314068508] [1983]  WARN RPI raspberrypi.cpp:1270 Mismatch between
Unicam and CamHelper for embedded data usage!
[0:11:13.315946837] [1983]  INFO RPI raspberrypi.cpp:1387 Registered camera
***@***.******@***.*** to Unicam device /dev/media1 and ISP
device /dev/media0
[INFO] [1672331294.177846336] [camera]:
>> cameras:
   0: imx219 ***@***.******@***.***)
[WARN] [1672331294.178177483] [camera]: no camera selected, using default:
***@***.******@***.***"
[INFO] [1672331294.179300296] [camera]:
>> RGB888 format sizes:
   - 160x120
   - 240x160
   - 320x240
   - 400x240
   - 480x320
   - 640x360
   - 640x480
   - 720x480
   - 768x480
   - 854x480
   - 720x576
   - 800x600
   - 960x540
   - 1024x576
   - 960x640
   - 1024x600
   - 1024x768
   - 1280x720
   - 1152x864
   - 1280x800
   - 1360x768
   - 1366x768
   - 1440x900
   - 1280x1024
   - 1536x864
   - 1280x1080
   - 1600x900
   - 1400x1050
   - 1680x1050
   - 1600x1200
   - 1920x1080
   - 2048x1080
   - 1920x1200
   - 2160x1080
   - 2048x1152
   - 2560x1080
   - 2048x1536
   - 2560x1440
   - 2560x1600
   - 2960x1440
   - 2560x2048
   - 3200x1800
   - 3200x2048
   - 3200x2400
[WARN] [1672331294.179617702] [camera]: no dimensions selected,
auto-selecting: "1920x1080"
[WARN] [1672331294.180015090] [camera]: stream configuration adjusted from
"1920x1080-RGB888" to "1920x1080-RGB888"
[0:11:13.318860164] [1972]  INFO Camera camera.cpp:1026 configuring
streams: (0) 1920x1080-RGB888
[0:11:13.319518922] [1983]  INFO RPI raspberrypi.cpp:767 Sensor:

1920x1080-SBGGR10_1X10 - Selected unicam format: 1920x1080-pBAA
[INFO] [1672331294.183301583] [camera]: camera ***@***.***
[WARN] [1672331294.184508266] [camera]: control NoiseReductionMode (39) not
handled
[INFO] [1672331294.746255674] [camera]: using default calibration URL
[INFO] [1672331294.746461433] [camera]: camera calibration URL:
file:///home/ubuntu/.ros/camera_info/imx219__base_soc_i2c0mux_i2c_1_imx219_10_1920x1080.yaml
[ERROR] [1672331294.746756229] [camera_calibration_parsers]: Unable to open
camera calibration file
[/home/ubuntu/.ros/camera_info/imx219__base_soc_i2c0mux_i2c_1_imx219_10_1920x1080.yaml]
[WARN] [1672331294.747002876] [camera]: Camera calibration file
/home/ubuntu/.ros/camera_info/imx219__base_soc_i2c0mux_i2c_1_imx219_10_1920x1080.yaml
not found
^C[INFO] [1672334399.649295161] [rclcpp]: signal_handler(signum=2)
[1:02:58.840854105] [1983] ERROR DeviceEnumerator device_enumerator.cpp:166
Removing media device /dev/media0 while still in use
[1:02:58.840922586] [1983] ERROR DeviceEnumerator device_enumerator.cpp:166
Removing media device /dev/media1 while still in use

_____

I am thinking this is a rqt_image_view issue.

christianrauch, 2022-12-29:
This problem is not specific to the Raspberry Pi camera as I showed above with the `vivid` test driver.

When you mention the "config yaml file" do you mean the "Camera calibration file"? If you did not calibrate your camera yet, then this warning is normal. I just says that the camera calibration file cannot be found at the expected location because the camera was never calibrated.

anfederman, 2022-12-29:
Thanks for your reply. Is there a parameter to pass to camera_ros to flip
the channels?

christianrauch, 2022-12-29:
> Thanks for your reply. Is there a parameter to pass to camera_ros to flip the channels?

No, and there should be no such parameter because the colour format mapping between libcamera and the ROS image encodings should work correctly. I am going to look into this. In the meantime, you can simply change the encoding from `bgr8` to `rgb8` in order to have the channels switched.

anfederman, 2022-12-29:
I did switch the channels, from bgr to rgb, but the colors still were
reversed in rqt.  Topics echo showed the format switched, but the image
didn't change.

christianrauch, 2022-12-30:
I wrote a simple python script to visualise the channels of an image:
```python3
#!/usr/bin/env python3
import rclpy
from rclpy.node import Node
from sensor_msgs.msg import Image
import numpy as np
import cv2

class FormatVisualiser(Node):
    def __init__(self):
        super().__init__('format_visualiser')
        self.sub_img = self.create_subscription(Image, "/camera/image_raw", self.on_image, 1)

    def on_image(self, msg_img):
        dtype = np.dtype(msg_img.data.typecode)
        dtype = dtype.newbyteorder('>' if msg_img.is_bigendian else '<')
        img_raw = np.asarray(msg_img.data.tolist(), dtype=dtype)\
            .reshape(msg_img.height, msg_img.width, int(msg_img.step/msg_img.width))
        print(img_raw.shape)

        cv2.imshow("raw", img_raw)
        for i in range(img_raw.shape[2]):
            cv2.imshow(f"raw channel {i}", img_raw[...,i])

        if msg_img.encoding == "bgra8":
            img_rgb = cv2.cvtColor(img_raw, cv2.COLOR_BGRA2RGB)
            cv2.imshow("rgb", img_rgb)
            for i in range(img_rgb.shape[2]):
                cv2.imshow(f"rgb channel {i}", img_rgb[...,i])

            img_bgr = cv2.cvtColor(img_raw, cv2.COLOR_BGRA2BGR)
            cv2.imshow("bgr", img_bgr)
            for i in range(img_bgr.shape[2]):
                cv2.imshow(f"bgr channel {i}", img_bgr[...,i])

        cv2.waitKey(1)

if __name__ == "__main__":
    rclpy.init()
    exporter_node = FormatVisualiser()
    rclpy.spin(exporter_node)
    exporter_node.destroy_node()
    rclpy.shutdown()
```

When I use this with the `BGRA8888` format (or `BA24` in fourcc):
```sh
ros2 run camera_ros camera_node --ros-args -p camera:=vivid -p format:=BGRA8888
```
I can clearly see that the channels in memory are `A`, `R`, `G`, `B` , i.e. in inverse order (the alpha channel is supposed to be empty). This inverse order comes from the endianness (see https://en.wikipedia.org/wiki/RGBA_color_model#Representation).

Since the ROS message contains the endianness, I thought that the client is supposed to deal with the memory representation.

christianrauch, 2022-12-30:
I (force) pushed an update to slect the target ROS encoding via the machine endianness. This should resolve your issue. Can you check?

anfederman, 2022-12-30:
Thank you, it works!  Issue is resolved.

christianrauch, 2022-12-30:
Nice. Then I will go ahead and close this issue.

closed 2022-12-30 by christianrauch