christianrauch/camera_ros #11: Configure frames per second

billmania, 2023-08-14:
How does one configure the frames per second published by the camera_ros node? Which specific ROS2 parameters can be used to accomplish this? For performance and CPU load reasons, I need to limit the frames per second. I understand how to adjust the size of the frames, using the height and width parameters, but do not know how to set a maximum frames per second.

christianrauch, 2023-08-23:
There are a couple of parameters affecting the frame duration. You can get an overview of the parameters and the set values via `ros2 param dump /camera`. The parameters that affect the frame duration are: `AeEnable`, `ExposureTime` and `FrameDurationLimits`. Then you can check the actual frame rate with `ros2 topic hz /camera/image_raw`.

christianrauch, 2023-08-23:
Since the `FrameDurationLimits` is given in microseconds, you should be able to fix the framerate to e.g. 5Hz (0.2s per frame) by setting the minimum and maximum frame duration to 200000µs:
```
ros2 run camera_ros camera_node --ros-args -p width:=320 -p height:=180 -p FrameDurationLimits:="[200000,200000]"
```

closed 2023-09-13 by christianrauch