christianrauch/camera_ros #12: Could not find a package configuration file provided by "cv_bridge"

BurmazDaniel, 2023-09-13:
Hello! I am tringing to build your node with colcon build but i get this error:
`
By not providing "Findcv_bridge.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "cv_bridge", but CMake did not find one.

  Could not find a package configuration file provided by "cv_bridge" with
  any of the following names:

    cv_bridgeConfig.cmake
    cv_bridge-config.cmake

  Add the installation prefix of "cv_bridge" to CMAKE_PREFIX_PATH or set
  "cv_bridge_DIR" to a directory containing one of the above files.  If
  "cv_bridge" provides a separate development package or SDK, be sure it has
  been installed.
`
I tried to install cv bridge using `sudo apt-get install ros-humble-cv-bridge` but it was unable to locate the package. 
How can i solve this? Thank you!

christianrauch, 2023-09-13:
If `sudo apt install ros-humble-cv-bridge` does not work for you, you should check if you have set up the ROS 2 repos correctly.

If you just want to use the node without developing on it, I recommend installing the binary package `ros-humble-camera-ros` that will also install the dependencies.

BurmazDaniel, 2023-09-13:
> If `sudo apt install ros-humble-cv-bridge` does not work for you, you should check if you have set up the ROS 2 repos correctly.
> 
> If you just want to use the node without developing on it, I recommend installing the binary package `ros-humble-camera-ros` that will also install the dependencies.

Thank you for your message!

Unfortunately ros-humble-cv-bridge is not avalible using `deb [arch=arm64 signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu bullseye main.` I am running ros2 on a raspberry pi with Raspberry Pi OS. I tried to use jammy main but some packages are not avaliable. :(
[image]

christianrauch, 2023-09-13:
> Unfortunately ros-humble-cv-bridge is not avalible using `deb [arch=arm64 signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu bullseye main.` I am running ros2 [...]

The ROS 2 repo does not provide binary packages for Debian. Only Ubuntu and RHEL are supported with binary packages: https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027. Not sure what you expect me to do here. If you want the ROS 2 packages to be released for Debian too, you have to ask the maintainers to build and release binary packages for Debian. If you want to stick to Raspberry Pi OS, you will have to compile ROS 2 from source.

I recommend using Ubuntu with the binary packages on supported Raspberry Pi boards.

BurmazDaniel, 2023-09-13:
> The ROS 2 repo does not provide binary packages for Debian. Only Ubuntu and RHEL are supported with binary packages: https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027. Not [...]

Unfortunately, libcamera and picamera2 do not work on ubuntu. Did u manage to find a workaround?

christianrauch, 2023-09-14:
> Unfortunately, libcamera and picamera2 do not work on ubuntu.

I guess this cannot be entirely true. What makes you say this? I am using the `ros-humble-camera-ros` package on Ubuntu 22.04 on a Raspberry Pi 2 Zero W. Others are using the `camera_ros` package on a Raspberry Pi with Ubuntu too.

> Did you manage to find a workaround?

As I said before, you can either compile everything from source on the Raspberry Pi OS or use the binary releases on Ubuntu 22.04.

BurmazDaniel, 2023-09-14:
> I guess this cannot be entirely true. What makes you say this? I am using the `ros-humble-camera-ros` package on Ubuntu 22.04 on a Raspberry Pi 2 Zero W. Others are using the `camera_ros` package on [...]

[https://github.com/raspberrypi/picamera2/issues/383](url)
What camera do you use for your Pi2 zero? I have the Raspberry Pi Camera Module 2 (with the Sony IMX219) and i couldnt get it to work. I followed [https://www.raspberrypi.com/documentation/computers/camera_software.html](url) and built libcamera form source on ubuntu and when i run libcamera-hello it says that there is no camera connected. On raspberry pi os everyting works just fine. I compiled from source Ros2, colocon work just fine but only the cvbridge i could not manage to install.
Do you use legacy mode (the old V4L2) instead of libcamera on your pi2 zero?
Thank you!

christianrauch, 2023-09-14:
> https://github.com/raspberrypi/picamera2/issues/383

That's a Python binding for libcamera and the issue has been closed anyway. I don't see how this is related to `camera_ros`.

> What camera do you use for your Pi2 zero? I have the Raspberry Pi Camera Module 2 (with the Sony IMX219) and i couldnt get it to work. I followed https://www.raspberrypi.com/documentation/computers/camera_software.html and built libcamera from source on ubuntu and when i run libcamera-hello it says that there is no camera connected. On raspberry pi os everyting works just fine. I compiled from source Ros2, colocon work just fine but only the cvbridge i could not manage to install. Do you use legacy mode (the old V4L2) instead of libcamera on your pi2 zero? Thank you!

These are all libcamera issues. If you cannot get libcamera running, then there is nothing `camera_ros` can do here. I gave you the option to either build ROS 2 on Raspberry Pi (that includes all required dependencies, such as `cv_bridge`), which I haven't tested, or install the package via the binary distribution. If you choose to ignore these hints then you are on your own. This repo is not for helping out with generic ROS or libcamera issues. Please ask those questions in their respective forums and git repos.

closed 2023-09-15 by BurmazDaniel