christianrauch/camera_ros #13: 'AfWindows' parameter occurs error.

Woojin-Crive, 2023-10-28:
## Error
```
terminate called after throwing an instance of 'std::length_error'
  what():  cannot create std::vector larger than max_size()
```

## Debug
```
"error " << id->type() << ", " << id->name() << ", " << info.def().toString() << ", " << info.min().toString() << ", " << info.max().toString());
error 7, AfWindows, (0, 0)/0x0, (0, 0)/0x0, (65535, 65535)/65535x65535
```

Error is caused when the below code is executed.
```
const rclcpp::ParameterValue value =
      cv_to_pv(clamp(info.def(), info.min(), info.max()), extent);
```

christianrauch, 2023-10-28:
Can you please provide the details (libcamera version, camera hardware, etc.) and steps to reproduce this?

Woojin-Crive, 2023-10-29:
i am using
libcamera v0.1.0+103-02eea04
raspberry camera module v3 (imx708)
latest raspberry pi os (Debian GNU/Linux 12 (bookworm))

just running the given code can reproduce the error.

christianrauch, 2023-10-29:
Can you provide a proper gdb backtrace to see where exactly the exception is thrown and what the content of the variables is?

Can you show the content of `extent` and also show what [`std::vector<T,Allocator>::max_size`](https://en.cppreference.com/w/cpp/container/vector/max_size) is? I assume the exception comes from https://github.com/christianrauch/camera_ros/blob/adfce05acb769a50f73fd9afed0c534e2927995b/src/cv_to_pv.cpp#L91 because `extent` is larger than `std::vector<T>::max_size` or negative.

Woojin-Crive, 2023-10-29:
Sorry for the dirty pull request and commits. I am not familiar with using git..
I closed the pull request.

If you give me some time. I will try to get the backtrace information.

Woojin-Crive, 2023-10-29:
extent => 18446744073709551615
max size of std::vector<T> => 576460752303423487

As you expected, `extent` is larger than `std::vector<T>::max_size `.

christianrauch, 2023-10-29:
I see the issue now. `AfWindows` is defined as `Control<Span<const Rectangle>>`, i.e. it takes a dynamic list of "windows", and happens to be the only control that has no fixed extent (length) specified. The default value for the `extent` is `libcamera::dynamic_extent`which is simply the maximum possible value the type `std::size_t` can take.

Can you check if https://github.com/christianrauch/camera_ros/pull/15 fixes the issue for you?

Woojin-Crive, 2023-10-31:
I will check it today, thank you!

Woojin-Crive, 2023-11-02:
`camera_node: ../include/libcamera/controls.h:167: T libcamera::ControlValue::get() const [with T = libcamera::Span<const libcamera::Rectangle>; typename std::enable_if<(libcamera::details::is_span<U>::value || std::is_same<std::__cxx11::basic_string<char>, typename std::remove_cv< <template-parameter-1-1> >::type>::value), std::nullptr_t>::type <anonymous> = nullptr]: Assertion `isArray_' failed.`

This is not working. Detailed information will be added later.

Asd2gogo, 2024-05-03:
I have a similar hw setup as mentioned in https://github.com/christianrauch/camera_ros/issues/13#issuecomment-1784148338 

- libcamera v0.2.0+120-eb00c13d
- raspberry camera module v3 (imx708)
- Ubuntu 22.04

I ran into the same 'cannot create std::vector..' issue.
I've tried #15 and got the same result as mentioned in https://github.com/christianrauch/camera_ros/issues/13#issuecomment-1789955933

Any further idea on how to fix this?

Woojin-Crive, 2024-05-03:
You can just skip the parameter
https://github.com/christianrauch/camera_ros/pull/14#discussion_r1375233421

christianrauch, 2024-05-05:
I only have a "Camera Module v1" (OV5647) and therefore cannot test the node with any other Camera Module. I tried to mock the `AfWindows` control to reproduce this and fixed this in https://github.com/christianrauch/camera_ros/pull/15. But in https://github.com/christianrauch/camera_ros/issues/13#issuecomment-1789955933, it was reported that this still causes issues with the real "Camera Module 3" (IMX708).

@Asd2gogo You can check if https://github.com/christianrauch/camera_ros/pull/15 fixes your issue?

Asd2gogo, 2024-05-05:
Yea, i have tried https://github.com/christianrauch/camera_ros/pull/15, but got the same error that was reported in https://github.com/christianrauch/camera_ros/issues/13#issuecomment-1789955933.

christianrauch, 2024-05-05:
A general comment on hardware support:
I only have a Camera Module v1 to test the node with and I currently do not need another version of the Raspberry Pi Camera Module myself. Therefore, I am not willing to purchase additional modules. If someone needs support for a Camera Module other than v1, I essentially rely on financial and material support from the community.

Please get in contact with me if you want to sponsor a "Camera Module 3" in order to have this module tested with the node and potentially have this issue resolved properly.

christianrauch, 2024-05-05:
> Yea, i have tried #15, but got the same error that was reported in [#13 (comment)](https://github.com/christianrauch/camera_ros/issues/13#issuecomment-1789955933).

Sorry. I overlooked that you already tried the PR wit the same result.

Can you, or @Woojin-Crive, provide a gdb backtrace with debug symbols for this assertion?

Woojin-Crive, 2024-05-05:
Oh, i am now just using this with a docker.
They are just working fine.. I do not have the hardware now.! Sorry for that..
Below is the context of the dockefile.

## Complie the libcamera

WORKDIR /root/libcamera
RUN git clone https://github.com/raspberrypi/libcamera.git . && \
    git checkout 563cd78e1c9858769f7e4cc2628e2515836fd6e7 && \
    meson setup build && \
    ninja -C build install && \
    ldconfig

## I am now using my own fork.

git clone https://github.com/Woojin-Crive/camera_ros.git

## Use the launch file to start a node.

ros2 launch camera_ros camera_launch.py

PR #15 (2024-08-17): handle dynamic extents
Dynamic Spans have a maximum extent for the `Control<Span<Type>>` but a variable number of elements for the `ControlValue` in the `ControlInfo`. For dynamic Spans with a single element, this previously caused the exception `std::length_error` since a vector with "maximum extent" (`dynamic_extent`) elements was constructed. Additionally, libcamera allows controls to have default and min/max values of different types than the control.

This PR thus changes the behaviour of how libcamera controls are mapped and converted to ROS parameters in the following way:
- ignore the  [... cut]
closed 2024-08-17 by christianrauch