christianrauch/camera_ros #17: error specifying default camera

anfederman, 2023-11-08:
I am trying to get this working on a ARM64 rtos version of Ubuntu 22 on a raspberry pi 4.   I don't see how to specify which camera to use in the run command. it appears the default camera is /dev/video21 instead of /dev/video0.  

Using the latest main branch.

 I also see no documentation for this node, so I don't recall how I ever figured out how to invoke it.

ubuntu@ubuntu:~/ros2_ws$ ros2 run camera_ros camera_node

[0:34:59.130093780] [2515]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+106-34b248c8
[0:34:59.164114462] [2527]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:34:59.166427593] [2527]  WARN RPI vc4.cpp:344 Mismatch between Unicam and CamHelper for embedded data usage!
[0:34:59.168108655] [2527]  INFO RPI vc4.cpp:398 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media0 and ISP device /dev/media2
[INFO] [1699464734.248255669] [camera]: 
>> cameras:
   0: imx219 (/base/soc/i2c0mux/i2c@1/imx219@10)
[WARN] [1699464734.248642815] [camera]: no camera selected, using default: "/base/soc/i2c0mux/i2c@1/imx219@10"
[INFO] [1699464734.249644789] [camera]: 
>> stream formats:
   - Pixelformat: NV21 (64x64 - 3280x2464)
   - Pixelformat: YUV420 (64x64 - 3280x2464)
   - Pixelformat: NV12 (64x64 - 3280x2464)
   - Pixelformat: YVU420 (64x64 - 3280x2464)
   - Pixelformat: XBGR8888 (64x64 - 3280x2464)
   - Pixelformat: BGR888 (64x64 - 3280x2464)
   - Pixelformat: RGB888 (64x64 - 3280x2464)
   - Pixelformat: XRGB8888 (64x64 - 3280x2464)
   - Pixelformat: RGB565 (64x64 - 3280x2464)
   - Pixelformat: YVYU (64x64 - 3280x2464)
   - Pixelformat: YUYV (64x64 - 3280x2464)
   - Pixelformat: VYUY (64x64 - 3280x2464)
   - Pixelformat: UYVY (64x64 - 3280x2464)
[WARN] [1699464734.249895676] [camera]: no pixel format selected, using default: "XBGR8888"
[INFO] [1699464734.250061934] [camera]: 
>> XBGR8888 format sizes:
   - 160x120
   - 240x160
   - 320x240
   - 400x240
   - 480x320
   - 640x360
   - 640x480
   - 720x480
   - 768x480
   - 854x480
   - 720x576
   - 800x600
   - 960x540
   - 1024x576
   - 960x640
   - 1024x600
   - 1024x768
   - 1280x720
   - 1152x864
   - 1280x800
   - 1360x768
   - 1366x768
   - 1440x900
   - 1280x1024
   - 1536x864
   - 1280x1080
   - 1600x900
   - 1400x1050
   - 1680x1050
   - 1600x1200
   - 1920x1080
   - 2048x1080
   - 1920x1200
   - 2160x1080
   - 2048x1152
   - 2560x1080
   - 2048x1536
   - 2560x1440
   - 2560x1600
   - 2960x1440
   - 2560x2048
   - 3200x1800
   - 3200x2048
   - 3200x2400
[WARN] [1699464734.250311303] [camera]: no dimensions selected, auto-selecting: "3200x2400"
[WARN] [1699464734.250561745] [camera]: stream configuration adjusted from "3200x2400-XBGR8888" to "3200x2400-XBGR8888"
[0:34:59.170919765] [2515]  INFO Camera camera.cpp:1183 configuring streams: (0) 3200x2400-XBGR8888
[0:34:59.171740685] [2527]  INFO RPI vc4.cpp:556 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
[INFO] [1699464734.253322059] [camera]: camera "/base/soc/i2c0mux/i2c@1/imx219@10" configured with 3200x2400-XBGR8888 stream
[WARN] [1699464734.256726368] [camera]: control AeFlickerPeriod (10) not handled
[WARN] [1699464734.257336789] [camera]: control NoiseReductionMode (44) not handled
[WARN] [1699464734.258103062] [camera]: control AeFlickerMode (9) not handled
[0:35:00.196569822] [2527] ERROR V4L2 v4l2_videodevice.cpp:1248 /dev/video21[27:cap]: Not enough buffers provided by V4L2VideoDevice
[0:35:00.225450892] [2527] ERROR V4L2 v4l2_videodevice.cpp:1241 /dev/video21[27:cap]: Unable to request 1 buffers: Cannot allocate memory
[0:35:00.225539799] [2527] ERROR RPI pipeline_base.cpp:662 Failed to allocate buffers
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to start camera
[ros2run]: Aborted

christianrauch, 2023-11-08:
The available cameras are listed under `>> cameras:`. You can request a specific camera ID via the parameter `camera`. By default, the first camera (ID 0) will be selected. Since only a single camera has been detected (`imx219`).

I think the issue you are facing is related to memory: `Unable to request 1 buffers: Cannot allocate memory`.

anfederman, 2023-11-08:
how can I specify a smaller format size, it is defaulting to 3200x2400?

christianrauch, 2023-11-08:
> how can I specify a smaller format size, it is defaulting to 3200x2400?

That is handled via the parameters `width` and `height`:
```
ros2 run camera_ros camera_node --ros-args -p width:=640 -p height:=480
```

I guess I should start documenting these parameters :-)

anfederman, 2023-11-08:
Thank you. I appreciate your support. I keep asking Dave Crawley to send
you a Magni, I know you would have Humble running on it in a week.

Some old guys like me still need the docs. I wonder if there needs to be a
camera.yaml file somewhere to specify parameters.

anfederman, 2023-11-08:
It is working now. Even compressed image transport. Will post notes in
Ubiquity GitHub.

closed 2023-11-09 by christianrauch