christianrauch/camera_ros #18: No cameras available on Ubuntu 22.04 with ROS humble

ubsyj, 2023-11-09:
Hi, I would like to use your camera_ros package for my ROS humble project on Raspberry Pi 4B with Ubuntu 22.04 (Jammy Jellyfish) and Camera Module 3. I could build my ros workspace but when I try to run
`ros2 run camera_ros camera_node` 
libcamera cannot find any cameras available. Did I miss something in my setup?

christianrauch, 2023-11-14:
Do the libcamera example executables (without ROS) work? Do you see the same issue with the binary packages from the repo?

rpapallas, 2023-11-20:
I have the same issue with binary packages. The following Python script can access the camera and save an image to a file:

```python
import cv2
cap = cv2.VideoCapture('/dev/video0', cv2.CAP_V4L)
cap.set(cv2.CAP_PROP_FRAME_WIDTH, 2560)
cap.set(cv2.CAP_PROP_FRAME_HEIGHT, 1440)
ret, frame = cap.read()
cv2.imwrite('image.jpg', frame)
cap.release()
```

Therefore, I don't think is permission issues.

christianrauch, 2024-03-01:
I am going to close this as there is no new information on this issue. Please check with the libcamera examples first to verify that the library can detect the camera. You can reopen this issue if the connection problem persists and is indeed not reproducible with the libcamera examples.

closed 2024-03-01 by christianrauch