christianrauch/camera_ros #22: Error "what():  cannot create std::vector larger than max_size() [ros2run]: Aborted" when run camera_node

thuhangkhuat, 2024-02-20:
Hi, i try to run camera_node in Raspberry Pi 4 by command: ros2 run camera_ros camera_node --ros-args -p width:=640 -p height:=480
And I have an error:

> [1:16:46.295450271] [14117]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+31-5e4dc46a
> [1:16:46.406360344] [14128]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
> [1:16:46.420912211] [14128]  INFO RPI vc4.cpp:401 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media0
> [INFO] [1708395316.185715280] [camera]: 
> >> cameras:
>    0: imx708 (/base/soc/i2c0mux/i2c@1/imx708@1a)
> [WARN] [1708395316.186056480] [camera]: no camera selected, using default: "/base/soc/i2c0mux/i2c@1/imx708@1a"
> [INFO] [1708395316.188579951] [camera]: 
> >> stream formats:
>    - Pixelformat: NV21 (64x64 - 4608x2592)
>    - Pixelformat: YUV420 (64x64 - 4608x2592)
>    - Pixelformat: NV12 (64x64 - 4608x2592)
>    - Pixelformat: YVU420 (64x64 - 4608x2592)
>    - Pixelformat: XBGR8888 (64x64 - 4608x2592)
>    - Pixelformat: BGR888 (64x64 - 4608x2592)
>    - Pixelformat: RGB888 (64x64 - 4608x2592)
>    - Pixelformat: XRGB8888 (64x64 - 4608x2592)
>    - Pixelformat: RGB565 (64x64 - 4608x2592)
>    - Pixelformat: YVYU (64x64 - 4608x2592)
>    - Pixelformat: YUYV (64x64 - 4608x2592)
>    - Pixelformat: VYUY (64x64 - 4608x2592)
>    - Pixelformat: UYVY (64x64 - 4608x2592)
> [WARN] [1708395316.189108444] [camera]: no pixel format selected, using default: "XBGR8888"
> [WARN] [1708395316.189741122] [camera]: stream configuration adjusted from "640x480-XBGR8888" to "640x480-XBGR8888"
> [1:16:46.426388739] [14117]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888
> [1:16:46.428186181] [14128]  INFO RPI vc4.cpp:559 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
> [INFO] [1708395316.197436143] [camera]: camera "/base/soc/i2c0mux/i2c@1/imx708@1a" configured with 640x480-XBGR8888 stream
> terminate called after throwing an instance of 'std::length_error'
>   what():  cannot create std::vector larger than max_size()
> [ros2run]: Aborted
Is it necessary to have a camera.yaml file somewhere to specify parameters? Thank you.

christianrauch, 2024-02-20:
Which camera are you using? Is this the same issue as https://github.com/christianrauch/camera_ros/issues/13?

thuhangkhuat, 2024-02-21:
Hi, I am using Raspberry camera module v3 (imx708). when I run on a USB camera (imx335), package is still working. So, I will follow the issue as 13. Thank you.

thuhangkhuat, 2024-02-21:
Hi, I follow the issue and still have error
`ros2 run camera_ros camera_node 
[0:12:03.917104703] [3684]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+31-5e4dc46a
[0:12:04.224253640] [3724]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:12:04.238788018] [3724]  INFO RPI vc4.cpp:401 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media0 and ISP device /dev/media2
[INFO] [1708496415.278274255] [camera]: 
>> cameras:
   0: imx708 (/base/soc/i2c0mux/i2c@1/imx708@1a)
[WARN] [1708496415.278599456] [camera]: no camera selected, using default: "/base/soc/i2c0mux/i2c@1/imx708@1a"
[INFO] [1708496415.281035395] [camera]: 
>> stream formats:
   - Pixelformat: NV21 (64x64 - 4608x2592)
   - Pixelformat: YUV420 (64x64 - 4608x2592)
   - Pixelformat: NV12 (64x64 - 4608x2592)
   - Pixelformat: YVU420 (64x64 - 4608x2592)
   - Pixelformat: XBGR8888 (64x64 - 4608x2592)
   - Pixelformat: BGR888 (64x64 - 4608x2592)
   - Pixelformat: RGB888 (64x64 - 4608x2592)
   - Pixelformat: XRGB8888 (64x64 - 4608x2592)
   - Pixelformat: RGB565 (64x64 - 4608x2592)
   - Pixelformat: YVYU (64x64 - 4608x2592)
   - Pixelformat: YUYV (64x64 - 4608x2592)
   - Pixelformat: VYUY (64x64 - 4608x2592)
   - Pixelformat: UYVY (64x64 - 4608x2592)
[WARN] [1708496415.281251763] [camera]: no pixel format selected, using default: "XBGR8888"
[INFO] [1708496415.281536168] [camera]: 
>> XBGR8888 format sizes:
   - 160x120
   - 240x160
   - 320x240
   - 400x240
   - 480x320
   - 640x360
   - 640x480
   - 720x480
   - 768x480
   - 854x480
   - 720x576
   - 800x600
   - 960x540
   - 1024x576
   - 960x640
   - 1024x600
   - 1024x768
   - 1280x720
   - 1152x864
   - 1280x800
   - 1360x768
   - 1366x768
   - 1440x900
   - 1280x1024
   - 1536x864
   - 1280x1080
   - 1600x900
   - 1400x1050
   - 1680x1050
   - 1600x1200
   - 1920x1080
   - 2048x1080
   - 1920x1200
   - 2160x1080
   - 2048x1152
   - 2560x1080
   - 2048x1536
   - 2560x1440
   - 2560x1600
   - 3840x1080
   - 2960x1440
   - 3440x1440
   - 2560x2048
   - 3200x1800
   - 3840x1600
   - 3200x2048
   - 3200x2400
   - 3840x2160
   - 4096x2160
   - 3840x2400
[WARN] [1708496415.281866683] [camera]: no dimensions selected, auto-selecting: "3840x2400"
[WARN] [1708496415.282442585] [camera]: stream configuration adjusted from "3840x2400-XBGR8888" to "3840x2400-XBGR8888"
[0:12:04.244417389] [3684]  INFO Camera camera.cpp:1183 configuring streams: (0) 3840x2400-XBGR8888
[0:12:04.246177428] [3724]  INFO RPI vc4.cpp:559 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
[INFO] [1708496415.291036521] [camera]: camera "/base/soc/i2c0mux/i2c@1/imx708@1a" configured with 3840x2400-XBGR8888 stream
[WARN] [1708496415.293331295] [camera]: control AeFlickerPeriod (10) not handled
[WARN] [1708496415.293506923] [camera]: control HdrMode (41) not handled
[WARN] [1708496415.294163787] [camera]: control AeFlickerMode (9) not handled
[WARN] [1708496415.294823614] [camera]: control StatsOutputEnable (20001) not handled
[WARN] [1708496415.295502996] [camera]: control NoiseReductionMode (10002) not handled
[INFO] [1708496416.365674638] [camera]: using default calibration URL
[INFO] [1708496416.365890506] [camera]: camera calibration URL: file:///root/.ros/camera_info/imx708__base_soc_i2c0mux_i2c_1_imx708_1a_3840x2400.yaml
[ERROR] [1708496416.367063180] [camera_calibration_parsers]: Unable to open camera calibration file [/root/.ros/camera_info/imx708__base_soc_i2c0mux_i2c_1_imx708_1a_3840x2400.yaml]
[WARN] [1708496416.367205567] [camera]: Camera calibration file /root/.ros/camera_info/imx708__base_soc_i2c0mux_i2c_1_imx708_1a_3840x2400.yaml not found
camera_node: ../include/libcamera/controls.h:167: T libcamera::ControlValue::get() const [with T = libcamera::Span<const libcamera::Rectangle>; typename std::enable_if<(libcamera::details::is_span<U>::value || std::is_same<std::__cxx11::basic_string<char>, typename std::remove_cv< <template-parameter-1-1> >::type>::value), std::nullptr_t>::type <anonymous> = nullptr]: Assertion `isArray_' failed.
[ros2run]: Aborted`

christianrauch, 2024-02-21:
You are using the same camera as @Woojin-Crive in the original issue. I am closing this as a duplicate.

Duplicate of #13 .

christianrauch, 2024-02-21:
Duplicate of #13 .

closed 2024-02-21 by christianrauch