christianrauch/camera_ros #26: Build errors when remote building ROS packages

trackerpal, 2024-04-01:
Hi, I am getting build errors when I try to build this package remotely from my VM. I'm trying to build the ROS packages remotely and send the executables to my RPi which runs them. I've tried installing the libcamera-dev onto my VM but the error still persists. 

```
CMake Error at /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:603 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  CMakeLists.txt:20 (pkg_check_modules) 
```

christianrauch, 2024-04-01:
The CMake project is using pkg-config to search for libcamera. You can run `pkg-config --modversion libcamera` in the shell to check which libcamera version you have installed.

Which Linux distribution and version are you using to compile? And what is your target distribution and version on the Raspberry Pi? Are you trying to cross-compile or to compile on a virtualised arm machine? On Ubuntu 22.04, there is a binary package in the official ROS repository that you can install with `sudo apt install ros-humble-libcamera`.

trackerpal, 2024-04-02:
Hi, im not sure what I did different but after i did the meson setup build and ninja build steps again, it worked for me. thank you!

closed 2024-04-02 by trackerpal