christianrauch/camera_ros #30: Invalid configuration

SchlackBoles, 2024-04-03:
Hi. 

I tried to launch your file but get the following error (ros2 run camera_ros camera_node  )

[WARN] [1712148202.003170381] [camera]: stream configuration adjusted from "1400x1050-R8" to "1456x1088-SBGGR16"
[6:29:38.453767396] [82] ERROR Camera camera.cpp:1171 Can't configure camera with invalid configuration
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to configure streams
[ros2run]: Aborted

Tried to add a camera_params.yaml with 

camera_node:
  ros__parameters:
    format: "SBGGR16"
    width: 1280
    height: 720
    role: "video"

but it does not fix the issue. Any pointers? I am using the shutter camera on a raspberry pi5.

SchlackBoles, 2024-04-03:
Should probably mention that even though I list the width and the height, the configuration is adjusted:

ros2 run camera_ros camera_node --ros-args -p width:=800 -p height:=600 

[WARN] [1712148909.694909200] [camera]: stream configuration adjusted from "800x600-R8" to "1456x1088-SBGGR16"
[6:41:26.145506196] [145] ERROR Camera camera.cpp:1171 Can't configure camera with invalid configuration
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to configure streams
[ros2run]: Aborted

SchlackBoles, 2024-04-03:
Ok, so with the code snippet you provided, i.e 
```
/ros2 run camera_ros camera_node --ros-args -p width:=160 -p height:=120
```
And this is the output:
```
/[14:46:41.029211451] [996] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_pisp.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so is signed
[14:46:41.029308303] [996] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so'
[14:46:41.029621620] [996] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so is signed
[14:46:41.029730010] [996] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
[14:46:41.032943737] [996]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[14:46:41.033175275] [1006] DEBUG Camera camera_manager.cpp:69 Starting camera manager
[14:46:41.035877369] [1006] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
[14:46:41.035987740] [1006] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 10 missing dependencies
[14:46:41.036776449] [1006] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[14:46:41.036796023] [1006] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
[14:46:41.036870320] [1006] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media3
[14:46:41.036885876] [1006] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 1 missing dependencies
[14:46:41.036978654] [1006] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[14:46:41.036992210] [1006] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: rpivid
[14:46:41.037066470] [1006] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media1
[14:46:41.037081785] [1006] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 9 missing dependencies
[14:46:41.037162822] [1006] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media2
[14:46:41.037177563] [1006] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 9 missing dependencies
[14:46:41.037822234] [1006] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[14:46:41.037835975] [1006] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: pispbe
[14:46:41.038465498] [1006] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[14:46:41.038480832] [1006] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: pispbe
[14:46:41.038692889] [1006] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerVc4'
[14:46:41.038733667] [1006] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[14:46:41.038743908] [1006] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[14:46:41.038756908] [1006] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerPiSP'
[14:46:41.038772130] [1006] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "rp1-cfe"
[14:46:41.038795001] [1006] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "pispbe"
[14:46:41.038873686] [1006]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 01-04-2024 (16:43:42)
[14:46:41.039570228] [1006] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Exposure (0x00980911)
[14:46:41.039611766] [1006] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Camera Orientation (0x009a0922)
[14:46:41.039633081] [1006] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Camera Sensor Rotation (0x009a0923)
[14:46:41.039645951] [1006] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Vertical Blanking (0x009e0901)
[14:46:41.039655914] [1006] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Horizontal Blanking (0x009e0902)
[14:46:41.039666414] [1006] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Analogue Gain (0x009e0903)
[14:46:41.039677396] [1006] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Pixel Rate (0x009f0902)
[14:46:41.039687063] [1006] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Test Pattern (0x009f0903)
[14:46:41.039745563] [1006] DEBUG CameraSensor camera_sensor.cpp:267 'imx296 6-001a': Camera sensor does not support horizontal/vertical flip
[14:46:41.040051510] [1006] DEBUG CameraSensor camera_sensor.cpp:394 'imx296 6-001a': No static test pattern map for 'imx296'
[14:46:41.040814126] [1006] DEBUG IPAManager ipa_manager.cpp:316 IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so signature is valid
[14:46:41.040862793] [1006] DEBUG IPAProxy raspberrypi_ipa_proxy.cpp:45 initializing raspberrypi proxy: loading IPA from /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so
[14:46:41.046506389] [1006] DEBUG RPiBlackLevel black_level.cpp:41  Read black levels red 3840 green 3840 blue 3840
[14:46:41.046729835] [1006] DEBUG RPiAgc agc.cpp:49 Read AGC channel
[... cut ...]
   - 1280x1024
   - 1280x1080
   - 1400x1050
[WARN] [1712178756.079257800] [camera]: stream configuration adjusted from "1400x1050-R8" to "1456x1088-SBGGR16"
[14:58:52.529855296] [1010] DEBUG RPI pipeline_base.cpp:997 Format: 1456x1088 fmt SBGGR10 Score: 3000 (best 3000)
[14:58:52.529888852] [1010] ERROR Camera camera.cpp:1171 Can't configure camera with invalid configuration
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to configure streams
[ros2run]: Aborted
```

SchlackBoles, 2024-04-03:
Thanks for helping me btw. When I run this

```
libcamera-still --list-cameras
```
I get this output (in case this is needed for you)
```
Available cameras
-----------------
0 : imx296 [1456x1088 10-bit] (/base/axi/pcie@120000/rp1/i2c@88000/imx296@1a)
    Modes: 'SBGGR10_CSI2P' : 1456x1088 [60.38 fps - (0, 0)/1456x1088 crop]

```

SchlackBoles, 2024-04-03:
But it does not help to change the height and width. I tried changing the command like this
```
ros2 run camera_ros camera_node --ros-args -p width:=1456 -p height:=1088
```
but I still run into an error
```
[15:22:30.780256841] [1375] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_pisp.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so is signed
[15:22:30.780510602] [1375] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so'
[15:22:30.780716010] [1375] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so is signed
[15:22:30.780785844] [1375] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
[15:22:30.783471081] [1375]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[15:22:30.783651582] [1385] DEBUG Camera camera_manager.cpp:69 Starting camera manager
[15:22:30.786486374] [1385] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
[15:22:30.786632838] [1385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 10 missing dependencies
[15:22:30.787475287] [1385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[15:22:30.787489565] [1385] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
[15:22:30.787556028] [1385] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media3
[15:22:30.787563158] [1385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 1 missing dependencies
[15:22:30.787701751] [1385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[15:22:30.787713381] [1385] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: rpivid
[15:22:30.787821178] [1385] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media1
[15:22:30.787847715] [1385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 9 missing dependencies
[15:22:30.787932567] [1385] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media2
[15:22:30.787940826] [1385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 9 missing dependencies
[15:22:30.788579941] [1385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[15:22:30.788586997] [1385] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: pispbe
[15:22:30.789202759] [1385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[15:22:30.789209333] [1385] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: pispbe
[15:22:30.789406723] [1385] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerVc4'
[15:22:30.789437964] [1385] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[15:22:30.789442779] [1385] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[15:22:30.789451038] [1385] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerPiSP'
[15:22:30.789461464] [1385] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "rp1-cfe"
[15:22:30.789476335] [1385] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "pispbe"
[15:22:30.789540742] [1385]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 01-04-2024 (16:43:42)
[15:22:30.790254783] [1385] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Exposure (0x00980911)
[15:22:30.790322154] [1385] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Camera Orientation (0x009a0922)
[15:22:30.790370950] [1385] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Camera Sensor Rotation (0x009a0923)
[15:22:30.790406728] [1385] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Vertical Blanking (0x009e0901)
[15:22:30.790422006] [1385] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Horizontal Blanking (0x009e0902)
[15:22:30.790455210] [1385] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Analogue Gain (0x009e0903)
[15:22:30.790467229] [1385] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Pixel Rate (0x009f0902)
[15:22:30.790487877] [1385] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Test Pattern (0x009f0903)
[15:22:30.790550470] [1385] DEBUG CameraSensor camera_sensor.cpp:267 'imx296 6-001a': Camera sensor does not support horizontal/vertical flip
[15:22:30.790858397] [1385] DEBUG CameraSensor camera_sensor.cpp:394 'imx296 6-001a': No static test pattern map for 'imx296'
[15:22:30.791630531] [1385] DEBUG IPAManager ipa_manager.cpp:316 IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so signature is valid
[15:22:30.791688994] [1385] DEBUG IPAProxy raspberrypi_ipa_proxy.cpp:45 initializing raspberrypi proxy: loading IPA from /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so
[15:22:30.797313061] [1385] DEBUG RPiBlackLevel black_level.cpp:41  Read black levels red 3840 green 3840 blue 3840
[... cut ...]
   - Pixelformat: UYVY (32x32 - 1456x1088)
[WARN] [1712180174.352889134] [camera]: stream configuration adjusted from "1456x1088-R8" to "1456x1088-SBGGR16"
[15:22:30.803477946] [1375] DEBUG RPI pipeline_base.cpp:997 Format: 1456x1088 fmt SBGGR10 Score: 3000 (best 3000)
[15:22:30.803495724] [1375] ERROR Camera camera.cpp:1171 Can't configure camera with invalid configuration
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to configure streams
[ros2run]: Aborted

```

christianrauch, 2024-04-04:
The log says
```
WARN Formats formats.cpp:997 Unsupported pixel format 0x32525942
```

So I would assume that an unsupported format is requested.

Have you tried to set a supported format explicitly?

SchlackBoles, 2024-04-04:
Sorry if I was being unclear. My question is:

How do I set your script to SBGGR10? If I understand it correctly, this is what my camera provides, but I do not understand how I can give that variable to the Ros script. Sorry if this is considered very simple - I am new to this

SchlackBoles, 2024-04-04:
What I have tried is this

```
root@devfull:/home/pi/robot_ws# export LIBCAMERA_LOG_LEVELS=*:DEBUG
root@devfull:/home/pi/robot_ws# ros2 run camera_ros camera_node --ros-args -p width:=1456 -p height:=1088 -p pixel_format:=SBGGR10_CSI2P
```
which gives this output:
```
[0:27:55.623990413] [580] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_pisp.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so is signed
[0:27:55.624285785] [580] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so'
[0:27:55.624618694] [580] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so is signed
[0:27:55.624667139] [580] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
[0:27:55.627796749] [580]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[0:27:55.628053176] [590] DEBUG Camera camera_manager.cpp:69 Starting camera manager
[0:27:55.630781710] [590] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
[0:27:55.630874655] [590] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 10 missing dependencies
[0:27:55.631692975] [590] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[0:27:55.631707456] [590] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
[0:27:55.631775716] [590] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media3
[0:27:55.631783457] [590] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 1 missing dependencies
[0:27:55.631867383] [590] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[0:27:55.631873309] [590] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: rpivid
[0:27:55.631936143] [590] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media1
[0:27:55.631943291] [590] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 9 missing dependencies
[0:27:55.632019625] [590] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media2
[0:27:55.632139718] [590] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 9 missing dependencies
[0:27:55.632869241] [590] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[0:27:55.632888667] [590] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: pispbe
[0:27:55.633581282] [590] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[0:27:55.633599763] [590] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: pispbe
[0:27:55.633867431] [590] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerVc4'
[0:27:55.633959636] [590] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[0:27:55.633975673] [590] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[0:27:55.633989210] [590] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerPiSP'
[0:27:55.634003562] [590] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "rp1-cfe"
[0:27:55.634026636] [590] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "pispbe"
[0:27:55.634176266] [590]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 01-04-2024 (16:43:42)
[0:27:55.634886419] [590] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Exposure (0x00980911)
[0:27:55.634980012] [590] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Camera Orientation (0x009a0922)
[0:27:55.635016382] [590] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Camera Sensor Rotation (0x009a0923)
[0:27:55.635076253] [590] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Vertical Blanking (0x009e0901)
[0:27:55.635087438] [590] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Horizontal Blanking (0x009e0902)
[0:27:55.635145346] [590] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Analogue Gain (0x009e0903)
[0:27:55.635200754] [590] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Pixel Rate (0x009f0902)
[0:27:55.635252717] [590] DEBUG V4L2 v4l2_device.cpp:636 'imx296 6-001a': Control: Test Pattern (0x009f0903)
[0:27:55.635320217] [590] DEBUG CameraSensor camera_sensor.cpp:267 'imx296 6-001a': Camera sensor does not support horizontal/vertical flip
[0:27:55.635626645] [590] DEBUG CameraSensor camera_sensor.cpp:394 'imx296 6-001a': No static test pattern map for 'imx296'
[0:27:55.636410353] [590] DEBUG IPAManager ipa_manager.cpp:316 IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so signature is valid
[0:27:55.636464742] [590] DEBUG IPAProxy raspberrypi_ipa_proxy.cpp:45 initializing raspberrypi proxy: loading IPA from /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so
[0:27:55.642187034] [590] DEBUG RPiBlackLevel black_level.cpp:41  Read black levels red 3840 green 3840 blue 3840
[0:27:55.642448850] [590] DEBUG RPiAgc agc.cpp:49 Read AGC channel
[... cut ...]
   - Pixelformat: UYVY (32x32 - 1456x1088)
[WARN] [1712251655.085742194] [camera]: stream configuration adjusted from "1456x1088-R8" to "1456x1088-SBGGR16"
[0:27:55.646979172] [580] DEBUG RPI pipeline_base.cpp:997 Format: 1456x1088 fmt SBGGR10 Score: 3000 (best 3000)
[0:27:55.647014135] [580] ERROR Camera camera.cpp:1171 Can't configure camera with invalid configuration
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to configure streams
[ros2run]: Aborted
```

christianrauch, 2024-04-04:
The format `SBGGR10_CSI2P` is not listed under `stream formats`. You can only select formats that are supported by the camera. Try format `YUYV`:
```sh
ros2 run camera_ros camera_node --ros-args -p format:=YUYV
```

SchlackBoles, 2024-04-04:
Ok, so this got us further. By using your command I got this output:
```
root@devfull:/home/pi/robot_ws/install# ros2 run camera_ros camera_node --ros-args -p format:=YUYV
[4:01:05.091537732] [76]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[4:01:05.097099973] [86]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 01-04-2024 (16:43:42)
[4:01:05.106900400] [86]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx296@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
[INFO] [1712264444.545937721] [camera]: 
>> cameras:
   0: imx296 (/base/axi/pcie@120000/rp1/i2c@88000/imx296@1a)
[WARN] [1712264444.546204725] [camera]: no camera selected, using default: "/base/axi/pcie@120000/rp1/i2c@88000/imx296@1a"
[INFO] [1712264444.546695954] [camera]: 
>> YUYV format sizes:
   - 160x120
   - 240x160
   - 320x240
   - 400x240
   - 480x320
   - 640x360
   - 640x480
   - 720x480
   - 768x480
   - 854x480
   - 720x576
   - 800x600
   - 960x540
   - 1024x576
   - 960x640
   - 1024x600
   - 1024x768
   - 1280x720
   - 1152x864
   - 1280x800
   - 1360x768
   - 1366x768
   - 1440x900
   - 1280x1024
   - 1280x1080
   - 1400x1050
[WARN] [1712264444.546798159] [camera]: no dimensions selected, auto-selecting: "1400x1050"
[4:01:05.108205510] [76]  INFO Camera camera.cpp:1183 configuring streams: (0) 1400x1050-YUYV
[4:01:05.108400050] [86]  INFO RPI pisp.cpp:1405 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx296@1a - Selected sensor format: 1456x1088-SBGGR10_1X10 - Selected CFE format: 1456x1088-PC1B
[INFO] [1712264444.547574891] [camera]: camera "/base/axi/pcie@120000/rp1/i2c@88000/imx296@1a" configured with 1400x1050-YUYV stream
[WARN] [1712264444.547971563] [camera]: control AeFlickerPeriod (10) not handled
[WARN] [1712264444.548070194] [camera]: control HdrMode (41) not handled
[WARN] [1712264444.548257289] [camera]: control StatsOutputEnable (20001) not handled
[WARN] [1712264444.548457033] [camera]: control NoiseReductionMode (10002) not handled
[WARN] [1712264444.548646387] [camera]: control AeFlickerMode (9) not handled
[INFO] [1712264445.552739598] [camera]: using default calibration URL
[INFO] [1712264445.552828544] [camera]: camera calibration URL: file:///root/.ros/camera_info/imx296__base_axi_pcie_120000_rp1_i2c_88000_imx296_1a_1400x1050.yaml
[ERROR] [1712264445.552910860] [camera_calibration_parsers]: Unable to open camera calibration file [/root/.ros/camera_info/imx296__base_axi_pcie_120000_rp1_i2c_88000_imx296_1a_1400x1050.yaml]
[WARN] [1712264445.552925768] [camera]: Camera calibration file /root/.ros/camera_info/imx296__base_axi_pcie_120000_rp1_i2c_88000_imx296_1a_1400x1050.yaml not found
```

Created a file in /root/.ros/camera_info/with the expected name
```
nano imx296__base_axi_pcie_120000_rp1_i2c_88000_imx296_1a_1400x1050.yaml
```
with this content

```
image_width: 1400
image_height: 1050
camera_name: imx296
camera_matrix:
  rows: 3
  cols: 3
  data: [1000, 0, 700, 0, 1000, 525, 0, 0, 1]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [0, 0, 0, 0, 0]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1, 0, 0, 0, 1, 0, 0, 0, 1]
projection_matrix:
  rows: 3
  cols: 4
  data: [1000, 0, 700, 0, 0, 1000, 525, 0, 0, 0, 1, 0]
```
First I get this output (which I thought was ok)
```
[WARN] [1712267106.665897212] [camera]: no dimensions selected, auto-selecting: "1400x1050"
[4:45:27.227167488] [451]  INFO Camera camera.cpp:1183 configuring streams: (0) 1400x1050-YUYV
[4:45:27.227287915] [461]  INFO RPI pisp.cpp:1405 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx296@1a - Selected sensor format: 1456x1088-SBGGR10_1X10 - Selected CFE format: 1456x1088-PC1B
[INFO] [1712267106.666473145] [camera]: camera "/base/axi/pcie@120000/rp1/i2c@88000/imx296@1a" configured with 1400x1050-YUYV stream
[WARN] [1712267106.666797890] [camera]: control AeFlickerPeriod (10) not handled
[WARN] [1712267106.666838632] [camera]: control HdrMode (41) not handled
[WARN] [1712267106.667052449] [camera]: control StatsOutputEnable (20001) not handled
[WARN] [1712267106.667195432] [camera]: control NoiseReductionMode (10002) not handled
[WARN] [1712267106.667328304] [camera]: control AeFlickerMode (9) not handled
```
but when I try to connect rqt_image_view I get this message again and again:
```
[ERROR] [1712267117.475358841] [camera]: request 'Request(94307:X:0/1:0)' cancelled
[ERROR] [1712267117.475380526] [camera]: request 'Request(94308:X:0/1:0)' cancelled
[ERROR] [1712267117.475402230] [camera]: request 'Request(94309:X:0/1:0)' cancelled
[ERROR] [1712267117.475423194] [camera]: request 'Request(94310:X:0/1:0)' cancelled
[ERROR] [1712267117.475444324] [camera]: request 'Request(94311:X:0/1:0)' cancelled
```
If I add the debugging and run it I get this:
```
[4:48:06.021619870] [476] DEBUG Request request.cpp:122 Request(40258:X:0/1:0)
[ERROR] [1712267265.460470060] [camera]: request 'Request(40258:X:0/1:0)' cancelled
[4:48:06.021646260] [476] DEBUG Request request.cpp:122 Request(40259:X:0/1:0)
[ERROR] [1712267265.460495764] [camera]: request 'Request(40259:X:0/1:0)' cancelled
[4:48:06.021671593] [476] DEBUG Request request.cpp:122 Request(40260:X:0/1:0)
[ERROR] [1712267265.460521727] [camera]: request 'Request(40260:X:0/1:0)' cancelled
```

Any pointers?

AbrahamMsy, 2024-04-24:
I got the std::runtime_error and ran this command 

export LIBCAMERA_LOG_LEVELS=*:DEBUG
ros2 run camera_ros camera_node --ros-args -p width:=160 -p height:=120

my output was as follows

ros2 run camera_ros camera_node --ros-args -p width:=160 -p height:=120
[1:48:10.698542824] [3993] DEBUG IPAModule ipa_module.cpp:334 ipa_rkisp1.so: IPA module /opt/ros/humble/lib/libcamera/ipa_rkisp1.so is signed
[1:48:10.698780128] [3993] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/opt/ros/humble/lib/libcamera/ipa_rkisp1.so'
[1:48:10.698981950] [3993] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /opt/ros/humble/lib/libcamera/ipa_rpi_vc4.so is signed
[1:48:10.699105917] [3993] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/opt/ros/humble/lib/libcamera/ipa_rpi_vc4.so'
[1:48:10.707727836] [3993]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0
[1:48:10.708166851] [4003] DEBUG Camera camera_manager.cpp:69 Starting camera manager
[1:48:10.720732569] [4003] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "bcm2835-codec" created from /dev/media2
[1:48:10.720816683] [4003] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 5 missing dependencies
[1:48:10.726055137] [4003] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[1:48:10.726110824] [4003] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: bcm2835-codec
[1:48:10.726724974] [4003] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "bcm2835-isp" created from /dev/media0
[1:48:10.726769142] [4003] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 4 missing dependencies
[1:48:10.727358866] [4003] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "bcm2835-isp" created from /dev/media1
[1:48:10.727397293] [4003] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 4 missing dependencies
[1:48:10.730778740] [4003] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[1:48:10.730819315] [4003] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: bcm2835-isp
[1:48:10.734522014] [4003] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[1:48:10.734561478] [4003] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: bcm2835-isp
[1:48:10.735200944] [4003] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerUVC'
[1:48:10.735283317] [4003] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'SimplePipelineHandler'
[1:48:10.735320096] [4003] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerVc4'
[1:48:10.735377746] [4003] DEBUG RPI vc4.cpp:179 Unable to acquire a Unicam instance
[1:48:10.735400710] [4003] DEBUG RPI vc4.cpp:179 Unable to acquire a Unicam instance
[1:48:10.735421322] [4003] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerRkISP1'
[1:48:10.735464249] [4003] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerISI'
terminate called after throwing an instance of 'std::runtime_error'
  what():  no cameras available
[ros2run]: Aborted

christianrauch, 2024-04-24:
> terminate called after throwing an instance of 'std::runtime_error' what(): no cameras available

You do not have a camera connected or it is simply not detected by libcamera. How is this related to the issue?

AbrahamMsy, 2024-04-24:
It's a pi camera v3 and it is connected. How then do I get it to be detected by libcamera?

Also, I thought the issues were related due to bringing up the same error, sorry about that.

christianrauch, 2024-04-24:
> It's a pi camera v3 and it is connected. How then do I get it to be detected by libcamera?

I would first test with the examples in libcamera to check this. If the libcamera examples do not detect the camera, you have to open an issue with them. If it does, and the ROS node does not, you can create a new issue here. But otherwise, this repo only deals with the ROS specific bindings and not libcamera in general.

christianrauch, 2024-04-28:
@SchlackBoles If you do not have more information about what happens before the first request cancellation or a way to reproduce this, I would close the issue.

closed 2024-05-24 by christianrauch

christianrauch, 2024-11-23:
I assume this is the same issue as https://github.com/christianrauch/camera_ros/issues/64. The format `SRGGB8` does not seem to be available with the default `video` stream role. You have to change this to `raw` or `viewfinder` (`-p format:="SRGGB8" -p role:=raw`).