christianrauch/camera_ros #34: Segmentation faults with IMX219

IGNNE, 2024-04-15:
Hi, 

I am using an IMX219 sensor connected to a RasPi Zero 2. The Pi is running Raspberry OS 12, and ROS humble via the official docker image. 

The issue is, sometimes it works, sometimes it crashes with a Segfault. It only crashes when something actually subscribes to the pictures. It seems to be somewhat related to resolution and frame rate, with really low resolutions and rates it sometimes runs for >10 min.

There is nothing suspicious in the console output, even with libcamera debug logging. I have increased CMA to 128M,

christianrauch, 2024-04-15:
I am using Ubuntu for this. But I recently also observed crashes inside the RPi ipa. Are you able to build libcamera 0.1 from source to see if this is a regression from 0.1 to 0.2?

IGNNE, 2024-04-16:
It looks like libcamera v0.1 works better, I haven't had any crashes so far. However, now the images come very slow (5+ s per image, regardless of parameters). 

I want to double check that it is not caused by our setup (distributed ros with a dedicated "camera pi"). But before, framerates worked fine (if they worked at all).

IGNNE, 2024-04-16:
Okay, looks like the low frame rate is somehow an issue of distributed ROS. With libcamera v0.1, I can get a stable 1920x1080@10Hz locally, for example. Thanks for the tip!

How would one report this? I guess it is a libcamera problem, right?

christianrauch, 2024-04-16:
> How would one report this? I guess it is a libcamera problem, right?

You can report libcamera issues via https://bugs.libcamera.org but you would need a minimal reproducer outside of ROS, for example the `cam` or `qcam` executables.

christianrauch, 2024-04-17:
I built arm64 Debian packages for version 0.1 and 0.2 of libcamera that contain the `cam` executable: [ros-humble-libcamera.zip](https://github.com/christianrauch/camera_ros/files/15016585/ros-humble-libcamera.zip). You can use them to check if the example executable reads data continuously without issues.

christianrauch, 2024-04-27:
I manually compiled the node with libcamera 0.1 and can confirm that it does not crash for me with the older version.

christianrauch, 2024-04-29:
I reverted the libcamera version in the ROS repos back to 0.1 and this fixes the issue for me. With the "testing repos" (https://docs.ros.org/en/humble/Installation/Testing.html) you should be able to install the older version again that does not show this crash.

@IGNNE @nbaldy Can you conform this works for you with libcamera 0.1 in the testing repos?

IGNNE, 2024-04-30:
Sorry for my late answers, there are some deadlines coming up and I am just happy that I found a setup that "works for me" (manually compiled against libcamera 0.1). I'll try that version as soon as I can, thank you for looking into this!

barulicm, 2024-05-07:
I'm hitting the same issue and was able to try the testing version tonight. I can confirm the crash does not occur in the v0.1 version.

Environment:
- RPi 4B
- Camera Module v2 (IMX219)
- Ubuntu 22.04

Node arguments:
`ros2 run camera_ros camera_node --ros-args -p width:=640 -p height:=480 -p format:=BGR888`

In all cases, the subscriber is an instance of `ros2 topic hz /camera/image_raw`.

Behavior with libcamera v0.2:
* subscribing locally shows messages coming in at 15 Hz
* subscribing from another machine causes the node on the Pi to segfault

Behavior with libcamera v0.1:
* subscribing locally shows messages coming in at 15 Hz
* subscribing from another machine shows messages coming in at 6 Hz

christianrauch, 2024-05-10:
I can reproduce the issue without ROS and have reported this as https://bugs.libcamera.org/show_bug.cgi?id=220. However, since the libcamera examples (`cam`, `qcam`) run fine on the Raspberry Pi, the issue is still somehow triggered by the way the code is interacting with libcamera.

christianrauch, 2024-06-21:
Some race condition probably caused this issue in libcamera 0.2 and it has been fixed since then. Versions 0.1 (released to `humble`) and 0.3 (released to `jazzy`) do not have this issue.

closed 2024-06-21 by christianrauch