christianrauch/camera_ros #41: No Calibration File for OV5647

PhilShaw852, 2024-05-16:
Pi Zero 2 W
Ubuntu 22.04 Server (headless)
ROS Humble

ros2 run camera_ros camera_node --ros-args -p format:=RGB888 -p width:=800 -p height:=600
[0:05:07.805516583] [1216]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0
[0:05:07.943397313] [1226]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:05:07.952685794] [1226]  INFO RPI vc4.cpp:401 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media0
[INFO] [1715833378.954512502] [camera]: 
>> cameras:
   0: ov5647 (/base/soc/i2c0mux/i2c@1/ov5647@36)
[WARN] [1715833378.955238972] [camera]: no camera selected, using default: "/base/soc/i2c0mux/i2c@1/ov5647@36"
[WARN] [1715833378.956960354] [camera]: stream configuration adjusted from "800x600-RGB888" to "800x600-RGB888"
[0:05:07.956180277] [1216]  INFO Camera camera.cpp:1183 configuring streams: (0) 800x600-RGB888
[0:05:07.957402848] [1226]  INFO RPI vc4.cpp:559 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[INFO] [1715833378.962008923] [camera]: camera "/base/soc/i2c0mux/i2c@1/ov5647@36" configured with 800x600-RGB888 stream
[WARN] [1715833378.963189359] [camera]: control HdrMode (41) not handled
[WARN] [1715833378.964147186] [camera]: control StatsOutputEnable (20001) not handled
[WARN] [1715833378.964396670] [camera]: control NoiseReductionMode (10002) not handled
[WARN] [1715833378.965042513] [camera]: control AeFlickerPeriod (10) not handled
[WARN] [1715833378.965817265] [camera]: control AeFlickerMode (9) not handled
[INFO] [1715833379.563679144] [camera]: using default calibration URL
[INFO] [1715833379.564236340] [camera]: camera calibration URL: file:///home/aukipi0/.ros/camera_info/ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_800x600.yaml
[ERROR] [1715833379.564968904] [camera_calibration_parsers]: Unable to open camera calibration file [/home/aukipi0/.ros/camera_info/ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_800x600.yaml]
[WARN] [1715833379.565251981] [camera]: Camera calibration file /home/aukipi0/.ros/camera_info/ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_800x600.yaml not found

I have a chicken and egg situation, I can't connect the camera to a node without the calibration file, but I can't calibrate without the camera node...

Any suggestions appreciated.

christianrauch, 2024-05-16:
The presence of the calibration file should not affect the starting of the camera. It is merely a storage to read and publish the intrinsics.

I don't see issues in your log. What exactly is not working?

PhilShaw852, 2024-05-16:
> The presence of the calibration file should not affect the starting of the camera. It is merely a storage to read and publish the intrinsics.
> 
> I don't see issues in your log. What exactly is not working?

Sorry, cropped a line at the bottom:

[ros2run]: Segmentation fault - At this point it returns to the prompt.

If I run with no parameters I get the "Not enough buffers provided by V4L2VideoDevice" error. The topics are created but nothing is published.

christianrauch, 2024-05-16:
Can you provide a back trace and more information about the libcamera version? Could this be the same issue as https://github.com/christianrauch/camera_ros/issues/34?

christianrauch, 2024-05-20:
If you get a segmentation fault when running `cam -l` then this is not related to the ROS node. Please report this to the libcamera project.

PhilShaw852, 2024-05-21:
> If you get a segmentation fault when running `cam -l` then this is not related to the ROS node. Please report this to the libcamera project.

No, Cam -l ran fine, I included this to show it ran and found the camera ok.

PhilShaw852, 2024-05-27:
I now started from fresh, installed Ubuntu and successfully installed libcamera v0.3.0.

I then added ROS Humble and Camera_Ros via sudo apt install.

When I ran the node, it was using v0.1.0 (I thought v0.2.0 was bundled with camera_ros) and the performance wasn't great (1 frame every minute or so).

So I removed camera_ros and installed from source but now the package is not found (Package 'camera_ros' not found).

christianrauch, 2024-05-27:
Not sure how I can help you with that. You either can install the binary packages from the ROS repo, where on `humble` the versions are currently as follows:
```
$ apt-cache policy ros-humble-libcamera
ros-humble-libcamera:
  Installiert:           (keine)
  Installationskandidat: 0.1.0-1jammy.20240516.152103
  Versionstabelle:
     0.1.0-1jammy.20240516.152103 500
        500 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
$ apt-cache policy ros-humble-camera-ros
ros-humble-camera-ros:
  Installiert:           (keine)
  Installationskandidat: 0.1.0-2jammy.20240517.194107
  Versionstabelle:
     0.1.0-2jammy.20240517.194107 500
        500 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
```
or you have to build packages from source the usual way inside a colcon workspace.

In any case, you have to provide more information about your crash. For the binary packages, you can install the "dbgsym" packages, e.g. `ros-humble-camera-ros-dbgsym` to get a better backtrace. If you build from source, then please also provide a backtrace and information about the versions you are compiling.

PhilShaw852, 2024-05-30:
Ubuntu 22.04.4
Pi Zero 2 W
camera_ros with libcamera v0.1.0
Camera - OV5467

I am no longer seeing crashes.

The problem now is that the framerate is unuseable, at 640x480 (XRGB8888) the compressed stream is 2fps. If I drop the image resolution down to 320x240 it creeps up to 11fps.

I ran the hardware with libcamera and Raspberry Pi OS (no ROS) and had 30fps.

I am in contact with the libcamera team but wanted to see if this issue had been seen before.

christianrauch, 2024-05-30:
> The problem now is that the framerate is unuseable, at 640x480 (XRGB8888) the compressed stream is 2fps. If I drop the image resolution down to 320x240 it creeps up to 11fps.

Is this local or remote?

On my Raspberry Pi Zero 2 W with the Camera Module v1 I get the following framerate locally and remotely:

```sh
# shell 1
ros2 run camera_ros camera_node --ros-args -p format:=RGB888 -p width:=640 -p height:=480
```

locally (full framerate):
```sh
# shell 2 (locally)
$ ros2 topic hz /camera/image_raw/compressed
average rate: 30.022
	min: 0.032s max: 0.036s std dev: 0.00070s window: 32
average rate: 30.128
	min: 0.032s max: 0.036s std dev: 0.00060s window: 63
average rate: 30.122
	min: 0.032s max: 0.036s std dev: 0.00055s window: 94
average rate: 30.148
	min: 0.032s max: 0.036s std dev: 0.00055s window: 125
```

remotely (drop to around 20 Hz):
```
# shell 2 (remotely)
$ ros2 topic hz /camera/image_raw/compressed
average rate: 17.417
	min: 0.016s max: 0.097s std dev: 0.02631s window: 19
average rate: 17.214
	min: 0.016s max: 0.097s std dev: 0.02469s window: 36
average rate: 18.433
	min: 0.005s max: 0.107s std dev: 0.02492s window: 58
average rate: 17.568
	min: 0.005s max: 0.221s std dev: 0.03118s window: 73
average rate: 18.357
	min: 0.005s max: 0.221s std dev: 0.03014s window: 95
average rate: 19.240
	min: 0.005s max: 0.221s std dev: 0.02894s window: 119
```

I guess this is a hardware limitation of the Raspberry Pi Zero 2 W. Do you think there is anything that the ROS node can do to improve the framerate?

Note that some formats consume more bandwidth than others. The "RGB" formats use 3 channels / planes while the "YUV" formats only use 2 channels / planes.

For the raw format `640x480-RGB888` above will consume around 30 MB/s ((640 * 480 * 3 channels * 1 byte per channel) * 30 frames) / (1024**2)
```
$ ros2 topic bw /camera/image_raw
Subscribed to [/camera/image_raw]
30.36 MB/s from 31 messages
	Message size mean: 0.92 MB min: 0.92 MB max: 0.92 MB
29.67 MB/s from 62 messages
	Message size mean: 0.92 MB min: 0.92 MB max: 0.92 MB
29.61 MB/s from 94 messages
	Message size mean: 0.92 MB min: 0.92 MB max: 0.92 MB
29.60 MB/s from 100 messages
	Message size mean: 0.92 MB min: 0.92 MB max: 0.92 MB
29.62 MB/s from 100 messages
	Message size mean: 0.92 MB min: 0.92 MB max: 0.92 MB
29.63 MB/s from 100 messages
	Message size mean: 0.92 MB min: 0.92 MB max: 0.92 MB
```

With the `YUYV` format:
```sh
ros2 run camera_ros camera_node --ros-args -p format:=YUYV -p width:=640 -p height:=480
```
this drops (locally) to 20 MB/s since only 2 instead of 3 channels are used:
```
$ ros2 topic bw /camera/image_raw
Subscribed to [/camera/image_raw]
19.84 MB/s from 30 messages
	Message size mean: 0.61 MB min: 0.61 MB max: 0.61 MB
19.76 MB/s from 62 messages
	Message size mean: 0.61 MB min: 0.61 MB max: 0.61 MB
19.72 MB/s from 94 messages
	Message size mean: 0.61 MB min: 0.61 MB max: 0.61 MB
```

For compressed images, you can tune the compression quality (see https://github.com/christianrauch/camera_ros/pull/23) but for this you have to compile from source at the moment.

PhilShaw852, 2024-06-03:
Thanks Christian, I have some interesting feedback. The main issue turned out to be the SD Card. Replacing it with a brand new one solved most of the issues. I guess it was an older card with a poor read / write rate.

I have a couple of other questions, should the node support 30fps @ 720p or 1080p? I'm getting 6fps at 720p.

And, do you see a framerate drop as the Pi Zero 2 W gets hotter (60 deg c)? I'm dropping from 30fps to a stable 22fps once warm. I am ordering some cooling to experiment with.

christianrauch, 2024-06-12:
> I have a couple of other questions, should the node support 30fps @ 720p or 1080p? I'm getting 6fps at 720p.

The node should support every mode that libcamera supports. There could be a couple of reasons why you are not hitting the expected framerate. If you put too much load on the CPU, the node will not have time to finish processing images. Also, if you are setting a long exposure time, or have auto exposure enabled in a dark environment, such that the exposure time surpasses the frametime (1 / framerate), then the framerate will drop below the target framerate. Generally, all the settings and effects that you see on the node should be reproducible on the standalone libcamera examples.

closed 2024-06-19 by christianrauch