christianrauch/camera_ros #51: ov5647 with Ubuntu 22.04.4 64bit (Jammy Jellyfish) error

Jai-GAY, 2024-06-25:
I can run **sudo** rpicam-hello but not [rpicam-hello](https://www.raspberrypi.com/documentation/computers/camera_software.html#rpicam-apps), and also I get a Could not open any dmaHeap device error with the camera_ros. What am I missing?
ros-humble-camera-ros (0.1.0-2jammy.20240524.040616)
```
$ ros2 run camera_ros camera_node 
[1:07:48.481956213] [3588]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0
[1:07:48.519219966] [3601] ERROR RPI dma_heaps.cpp:53 Could not open any dmaHeap device
[1:07:48.536688342] [3601] ERROR RPI vc4.cpp:209 Failed to register camera ov5647 10-0036: -12
terminate called after throwing an instance of 'std::runtime_error'
  what():  no cameras available
[ros2run]: Aborted
```
```
$ll /dev/video0
crw-rw-rw-+ 1 root video 81, 0 Jun 25 10:42 /dev/video0
$ll /dev/media0
crw-rw-rw-+ 1 root video 505, 0 Jun 25 10:42 /dev/media0
$cat /etc/udev/rules.d/xxx.rules
SUBSYSTEM=="dma_heap", GROUP="video", MODE="0660"
$sudo rpicam-hello --version
rpicam-apps build: 122b835fb927-intree 09-01-2024 (17:13:42)
libcamera build: v0.1.0+118-563cd78e

[0:38:57.836935566] [3136]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:38:57.912760444] [3140]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:38:57.921061837] [3140]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media1
[0:38:57.921561890] [3140]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Made X/EGL preview window
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[0:38:58.602930195] [3136]  INFO Camera camera.cpp:1183 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
[0:38:58.605066550] [3140]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
```
$sudo [libcamera-hello](https://www.raspberrypi.com/documentation/computers/camera_software.html#libcamera)
```
Available cameras
-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]
```
$ sudo [v4l2-ctl](https://www.raspberrypi.com/documentation/computers/camera_software.html#v4l2-drivers) --list-devices
```
bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video18
	/dev/video31
	/dev/media3

bcm2835-isp (platform:bcm2835-isp):
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/media1
	/dev/media2

unicam (platform:fe801000.csi):
	/dev/video0
	/dev/media0
```
```
$ pip3 show opencv-python
Name: opencv-python
Version: 4.9.0.80
Summary: Wrapper package for OpenCV python bindings.
Home-page: https://github.com/opencv/opencv-python
Author: 
Author-email: 
License: Apache 2.0
Location: /home/pi/.local/lib/python3.10/site-packages
Requires: numpy, numpy, numpy, numpy, numpy
Required-by: cv-3
```
```
$cat /boot/firmware/config.txt 
camera_auto_detect=0
start_x=1
dtoverlay=ov5647
```
[image]

Jai-GAY, 2024-06-25:
> I don't fully follow what commands work and which do not. If the "rpicam-apps" do not work, you should open an issue at https://github.com/raspberrypi/rpicam-apps. If libcamera does not detect any cameras, then there is nothing the node can do. Maybe this is just a configuration issue?
> 
> I am also testing on an ov5647 on a Raspberry Pi Zero 2 W, previously with Ubuntu 22.04 and libcamera 0.1, now with Ubuntu 24.04 with libcamera 0.3, without issues. You could try to compile a newer libcamera version from source or simply switch to Ubuntu 24.04.

So the prerequisites if rpicam-apps (without sudo) work and libcamera detected OV5647, camera_ros should work?

christianrauch, 2024-06-26:
> So the prerequisites if rpicam-apps (without sudo) work and libcamera detected OV5647, camera_ros should work?

Of course, libcamera has to recognise the camera. The ROS node is just using the library to access cameras, read image data, etc. This all should work as normal user without sudo. If you can use libcamera only with sudo, then your user is probably missing the access rights or other settings are missing or wrong.

Can you please confirm if you can or cannot use the libcamera example applications (a.k.a. `rpicam-apps`) as a regular user without sudo?

You can list the video devices with `v4l2-ctl --list-devices` as you have shown, and then check the required permissions with `ls`. E.g.:
```
$ ls -alh /dev/video0
crw-rw----+ 1 root video 81, 0 Jun 22 19:32 /dev/video0
```
shows me that the `root` user and users of the `video` group have access to that device.

I am not in the `video` group but I can still access the cameras. But maybe you need to be in the `video` group?

Jai-GAY, 2024-06-26:
> Can you please confirm if you can or cannot use the libcamera example applications (a.k.a. `rpicam-apps`) as a regular user without sudo?

I cannot use it without sudo. it could be due to raspberrypi step [recommend sudo ninja -C build install](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera) while GitHub site [recommends without sudo](https://github.com/raspberrypi/libcamera?tab=readme-ov-file#getting-started).

> check the required permissions with `ls`. E.g.:

> $ll /dev/video0
> crw-rw-rw-+ 1 root video 81, 0 Jun 25 10:42 /dev/video0
> $ll /dev/media0
> crw-rw-rw-+ 1 root video 505, 0 Jun 25 10:42 /dev/media0
```
ls -alh /dev/video0
crw-rw-rw-+ 1 root video 81, 13 Jun 26 15:59 /dev/video0
```

christianrauch, 2024-06-26:
> I cannot use it without sudo. it could be due to raspberrypi step [recommend sudo ninja -C build install](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera) [...]

And this is the same for the ROS node, i.e. you can use it with sudo but not without sudo? If this is the case, then I guess you have to take the issue with libcamera or the rpicam-apps projects since there is nothing the node can do about this.

Jai-GAY, 2024-06-26:
> And this is the same for the ROS node, i.e. you can use it with sudo but not without sudo?

if I use sudo ros2 run, sudo: ros2: command not found.

Jai-GAY, 2024-06-27:
after the following change. still on libcamera v0.1.0
Edit [/boot/config.txt, remove the line "camera_auto_detect=1", and add "start_x=1" and "gpu_mem=128"](https://forums.raspberrypi.com/viewtopic.php?t=323390#p1935591). Rebooting at this stage will reload the old V4L2 driver.

$cheese works

```
$ ros2 run camera_ros camera_node 
[0:34:34.080810852] [3100]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0
terminate called after throwing an instance of 'std::runtime_error'
  what():  no cameras available
[ros2run]: Aborted
```
```
$ sudo rpicam-hello 
[sudo] password for pi: 
ERROR: rpicam-apps currently only supports the Raspberry Pi platforms.
Contributions for other platforms are welcome at https://github.com/raspberrypi/rpicam-apps.
```

christianrauch, 2024-06-28:
> if I use sudo ros2 run,
> 
> ```
> [ros2run]: Aborted
> sudo: ros2: command not found
> ```

You can also start the node directly via its `camera_node` executable in the `install` folder.

christianrauch, 2024-06-30:
> You can also start the node directly via its `camera_node` executable in the `install` folder.

Concretely, I mean that instead of `sudo ros2 run camera_ros camera_node` you can also run the `camera_node` executable directly via `sudo ./install/camera_ros/lib/camera_ros/camera_node`.

@Jai-GAY Can you check if running the node with sudo this way allows you to access the camera? Also, can you be specific about what you expect to work and what concretely does not work? You are mixing a couple of things here and I am still not sure what issues are caused by the node and what is irrelevant to this repo.

Jai-GAY, 2024-07-01:
> sudo ./install/camera_ros/lib/camera_ros/camera_node

```
$ ll ./install
ls: cannot access './install': No such file or directory
```

> Also, can you be specific about what you expect to work and what concretely does not work?

hoping this works. ros2 run camera_ros camera_node
I think [camera_ros depends on libcamera](https://github.com/christianrauch/camera_ros?tab=readme-ov-file#ros-2-node-for-libcamera), so I understand what you mean.
I may have overwhelmed the issue without knowing how camera_ros and libcamera working relationship.

christianrauch, 2024-07-02:
> camera_ros was installed via
> 
> ```
> source /opt/ros/humble/setup.bash
> sudo apt install ros-$ROS_DISTRO-camera-ros
> ```
> 
> ```
> $ ll ./install
> ls: cannot access './install': No such file or directory
> ```

Of course, you have to pick the correct path. If you install the node via the Debian package, the executable will be somewhere under `/opt/ros/`.

> current error.
> [ERROR] [1719809358.804975838] [camera_calibration_parsers]: Unable to open camera calibration file [/home/pi/.ros/camera_info/ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_2560x1600.yaml]
> [WARN] [1719809358.805043906] [camera]: Camera calibration file /home/pi/.ros/camera_info/ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_2560x1600.yaml not found

What issues do you have with this? You should still get images from the camera.

Jai-GAY, 2024-07-03:
> You should still get images from the camera.

No, that is the thing.

christianrauch, 2024-07-03:
> > You should still get images from the camera.
> 
> No, that is the thing.

Can you please be more specific about what the observed and expected behaviour is? What did you try, what did you expect and what was the actual observed behaviour? Is this still related to your initial problem? It is not possible to help you with this sparse information. The missing calibration file does not affect the published images.

Jai-GAY, 2024-07-03:
can you provide a sceen capture when user run this command what output is expected?

christianrauch, 2024-07-03:
> can you provide a screen capture of when the user runs this command and what success output is expected?

If you are trying to use the node, there must be something that you expect it to do for you :-) Typically, you want to subscribe to an image topic to get the images from the camera.

If you say you are not receiving images, what did you try?

Jai-GAY, 2024-07-04:
how does the user verify setup and that everything is working when executing the following command, ros2 run camera_ros camera_node? Do you have a screenshot example?

christianrauch, 2024-07-04:
I won't be able to help you unless you tell me explicitly and in detail 1)what you have tried, 2) what you expect to happen and 3) what actually happened.

Your questions are generically about how to run a ROS node and use topics etc. For this, please consult the ROS tutorials and documentation at https://docs.ros.org/en/jazzy/Tutorials.html.

Jai-GAY, 2024-07-04:
I have only tried [installing](https://github.com/christianrauch/camera_ros?tab=readme-ov-file#install) and [launching](https://github.com/christianrauch/camera_ros?tab=readme-ov-file#launching-the-node) the node.

christianrauch, 2024-07-04:
But what did you try with the node in order to access the images? Did you subscribe to the image topics?

You said that you are not getting images from the camera. What makes you say that?

Jai-GAY, 2024-07-05:
> You said that you are not getting images from the camera. What makes you say that?

First, I needed to verify that my hardware and settings were okay, so I used 3rd party software module or application to verify.

So, I like to know how users verify that their subscription for images is not working or Camera_ros is not working.
Is there any printout statement users can observe to know that Camera_ros is working well in the backend?

christianrauch, 2024-07-05:
I created PR https://github.com/christianrauch/camera_ros/pull/52 to add more information about ROS topics, calibration and debugging to the documentation. Please have a look if this helps you to better understand what the node is supposed to do and let me know if anything is missing from your side there.

> So, I like to know how users verify that their subscription for images is not working or Camera_ros is not working.
> Is there any printout statement users can observe to know that Camera_ros is working well in the backend?

Not in the default info log level. You can enable the libcamera and ROS debug levels. With the libcamera debug information, you should see something like `DEBUG Request request.cpp:122 Request(1:C:0/1:0)` printed periodically on screen. To introspect the ROS topics, you can use the usual `ros2 topic list` and `ros2 topic echo` commands.

closed 2024-08-04 by christianrauch

Jai-GAY, 2024-09-26:
finally, I got [the legacy camera working with Ubuntu 22.04 over OpenCV 4.x](https://www.youtube.com/watch?v=va7o7wzhEE4).
[image]

christianrauch, 2024-09-26:
The "legacy" mode will be removed at some point since libcamera is now the official way to access the cameras: https://www.raspberrypi.com/documentation/computers/camera_software.html. If you need something future-proof, I would not invest in the legacy mode.

Jai-GAY, 2024-09-27:
> since libcamera is now the official way to access the cameras: https://www.raspberrypi.com/documentation/computers/camera_software.html.

I am unsure if anything has to do with the libcamera I recompiled and reinstalled last month.

Raspberry Pi’s implementation of libcamera supports the following cameras:
Official cameras:
OV5647 (V1)

christianrauch, 2024-09-29:
I don't get what you want to say. libcamera supports all Raspberry Pi cameras. It's the officially supported camera software stack on the Raspberry Pi and the one this ROS node is made for.

If you still want to use the legacy stack for a while, then you cannot use this node.