christianrauch/camera_ros #53: Ubuntu 24.04 -Pi5-ROS2 Jazzy - OV5647

jeref-merlin, 2024-07-09:
Hi community,

I'm currently trying to stream my camera as per configurations mentioned at title but the error shown was 
[0:48:28.601714275] [49141]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+156-644986c2
terminate called after throwing an instance of 'std::runtime_error'
  what():  no cameras available
[ros2run]: Aborted
I have followed the instruction to build from source. I am wondering if you could help me with this.

christianrauch, 2024-07-09:
Can you provide more details about your hard- and software setup? How did you install libcamera? Did you try to use any of the libcamera stand-alone examples without ROS?

jeref-merlin, 2024-07-09:
Hi Christian,

For hardware, I have OV5647-CSI connection to Raspberry pi 5 running on Ubuntu 24.04. I have installed the libcamera dependency according to your steps in (https://github.com/christianrauch/camera_ros). I am running them with ROS2 jazzy but no luck. Similarly from the issue stated (https://github.com/christianrauch/camera_ros/issues/51), I have built rpicam from source as libcam app and rpicam app aren't available in 24.04 yet I believe. Since you have mentioned that you managed to ubuntu 24.04 with libcamera 0.3 without any issues, I am wondering if our setup are different or any steps that I may have missed to achieve the same results as you.

Kind regards

christianrauch, 2024-07-10:
> Similarly from the issue stated (#51), I have built rpicam from source as libcam app and rpicam app aren't available in 24.04 yet I believe.

Did you also follow the steps in that issue and try any of the examples? What does `libcamera-hello --list-cameras` (with and without sudo) return? What does `v4l2-ctl --list-devices` show? The issue is also mentioning that you may have to set `disabling start_x=1` with a link to the Pi 5 documentation at https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/5MP-OV5647/#software-configuration that specifically mentions the ov5647 on the Pi 5.

Did you also try to use the camera with Raspberry Pi OS to exclude hardware or configuration errors? I.e. the Pi 5 has two MIPI ports. Have you connected the camera to the correct port and configured it accordingly?

jeref-merlin, 2024-08-22:
Hi CHristian,

I found out that l missed PISP pipeline during libcamera building process and that solved it. Thank you for the input!

closed 2024-08-22 by jeref-merlin