christianrauch/camera_ros #61: runtime_error: control NoiseReductionMode (10002) not handled

anderskm, 2024-09-13:
I am trying to run the `camera_node` on a Raspberry Pi 5 with the Camera Module 3 Wide camera. To run ROS, I am running a Docker container with Ubuntu 22 and ROS2 Humble. I have verified, that I can get an image stream from the camera both outside and inside the Docker container using `rpicam-hello`.
However, when I try to run the `camera_node`, I get a error explaining that the `control NoiseReductionMode` is not handled.
To be honest, I am a little unsure, what part of the pipeline (`libcamera`, `camera_ros`, `v4l2`?) is causing the error. But as I have verified, that I can access the camera in the same environment (inside Docker) using `rpicam-hello` (which also uses `libcamera`), I suspect that the problem lies in `camera_ros`.

### Setup
- Raspberry Pi 5, 8GB running Raspbian 64-bit
- Raspberry Pi Camera Module 3 Wide
- Docker container running Ubuntu 22.04.4 LTS (jammy)
  - ROS2 Humble using Dockerfile from https://github.com/osrf/docker_images
  - Manually compiled `libcamera` following https://github.com/raspberrypi/libcamera
  - Manually compiled `rpicam-apps` following https://www.raspberrypi.com/documentation/computers/camera_software.html#build-libcamera-and-rpicam-apps

### Command
```
ros2 run camera_ros camera_node --ros-args -p format:="YUYV" -p width:=1024 -p height:=768 -p FrameDurationLimits:="[90000,110000]"
```

### Output
<details><summary>LIBCAMERA_LOG_LEVELS=*:INFO</summary>

```
[2:38:32.443852285] [2061]  INFO Camera camera_manager.cpp:316 libcamera v0.3.1+50-69a894c4
[2:38:32.449800302] [2071]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 13-09-2024 (07:02:13)
[2:38:32.460247774] [2071]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media1 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[INFO] [1726218931.147531131] [camera]:
>> cameras:
   0: imx708_wide (/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a)
[WARN] [1726218931.147841262] [camera]: no camera selected, using default: "/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a"
[WARN] [1726218931.147922262] [camera]: set parameter 'camera' to silence this warning
[2:38:32.461043758] [2061]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
[2:38:32.461518944] [2061]  INFO Camera camera.cpp:1191 configuring streams: (0) 1024x768-YUYV
[2:38:32.461684778] [2071]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected CFE format: 1536x864-PC1B
[INFO] [1726218931.149175932] [camera]: camera "/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a" configured with 1024x768-YUYV stream
[WARN] [1726218931.149631544] [camera]: unsupported control 'AfWindows'
terminate called after throwing an instance of 'std::runtime_error'
  what():  control NoiseReductionMode (10002) not handled
[ros2run]: Aborted
```

</details>

<details><summary>LIBCAMERA_LOG_LEVELS=*:DEBUG</summary>

```
[2:25:32.525469412] [2038] DEBUG IPAModule ipa_module.cpp:334 ipa_rkisp1.so: IPA module /ros2_ws/install/libcamera/lib/libcamera/ipa_rkisp1.so is signed
[2:25:32.525569042] [2038] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/ros2_ws/install/libcamera/lib/libcamera/ipa_rkisp1.so'
[2:25:32.525648597] [2038] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_pisp.so: IPA module /ros2_ws/install/libcamera/lib/libcamera/ipa_rpi_pisp.so is signed
[2:25:32.525718524] [2038] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/ros2_ws/install/libcamera/lib/libcamera/ipa_rpi_pisp.so'
[2:25:32.526003969] [2038] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /ros2_ws/install/libcamera/lib/libcamera/ipa_rpi_vc4.so is signed
[2:25:32.526086765] [2038] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/ros2_ws/install/libcamera/lib/libcamera/ipa_rpi_vc4.so'
[2:25:32.526197803] [2038] DEBUG IPAModule ipa_module.cpp:334 ipa_soft_simple.so: IPA module /ros2_ws/install/libcamera/lib/libcamera/ipa_soft_simple.so is signed
[2:25:32.526237896] [2038] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/ros2_ws/install/libcamera/lib/libcamera/ipa_soft_simple.so'
[2:25:32.529164182] [2038]  INFO Camera camera_manager.cpp:316 libcamera v0.3.1+50-69a894c4
[2:25:32.529418924] [2048] DEBUG Camera camera_manager.cpp:71 Starting camera manager
[2:25:32.532172043] [2048] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media1
[2:25:32.532341544] [2048] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 10 missing dependencies
[2:25:32.533108731] [2048] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[2:25:32.533128935] [2048] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: rp1-cfe
[2:25:32.533220824] [2048] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media2
[2:25:32.533236398] [2048] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 1 missing dependencies
[2:25:32.533338602] [2048] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[2:25:32.533352269] [2048] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: rpivid
[2:25:32.533442825] [2048] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media0
[2:25:32.533483177] [2048] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 9 missing dependencies
[2:25:32.533611622] [2048] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media3
[2:25:32.533630325] [2048] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 9 missing dependencies
[2:25:32.534289661] [2048] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[2:25:32.534307346] [2048] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: pispbe
[2:25:32.534940292] [2048] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[2:25:32.534954274] [2048] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: pispbe
[... 86 lines cut ...]
[2:25:32.548358389] [2038] DEBUG RPI pipeline_base.cpp:284 Try color space Rec709
[2:25:32.548444518] [2038]  INFO Camera camera.cpp:1191 configuring streams: (0) 1024x768-YUYV
[2:25:32.548598204] [2048]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected CFE format: 1536x864-PC1B
[2:25:32.548619223] [2048] DEBUG RPI pisp.cpp:1508 Setting ISP Output1 to 1024x768-YUYV (sw downscale 1)
[2:25:32.548631889] [2048] DEBUG RPI pisp.cpp:1514 After setFormat, stride 2048
[2:25:32.548639797] [2048] DEBUG RPI pisp.cpp:1524 Stream ISP Output1 has color space Rec709
[2:25:32.548653575] [2048] DEBUG RPI pisp.cpp:2095 Output 1 (192, 0)/1152x864
[2:25:32.548685982] [2048] DEBUG RPI pisp.cpp:1605 Setting embedded data format 0x0-SENS
[2:25:32.548713982] [2048] DEBUG MediaDevice media_device.cpp:830 /dev/media1[rp1-cfe]: csi2[4] -> rp1-cfe-csi2_ch0[0]: 0
[2:25:32.548724686] [2048] DEBUG MediaDevice media_device.cpp:830 /dev/media1[rp1-cfe]: csi2[4] -> pisp-fe[0]: 1
[2:25:32.548733556] [2048] DEBUG MediaDevice media_device.cpp:830 /dev/media1[rp1-cfe]: csi2[5] -> rp1-cfe-embedded[0]: 1
[2:25:32.548742853] [2048] DEBUG MediaDevice media_device.cpp:830 /dev/media1[rp1-cfe]: rp1-cfe-fe_config[0] -> pisp-fe[1]: 1
[2:25:32.548754445] [2048] DEBUG MediaDevice media_device.cpp:830 /dev/media1[rp1-cfe]: pisp-fe[2] -> rp1-cfe-fe_image0[0]: 1
[2:25:32.548764131] [2048] DEBUG MediaDevice media_device.cpp:830 /dev/media1[rp1-cfe]: pisp-fe[3] -> rp1-cfe-fe_image1[0]: 0
[2:25:32.548773205] [2048] DEBUG MediaDevice media_device.cpp:830 /dev/media1[rp1-cfe]: pisp-fe[4] -> rp1-cfe-fe_stats[0]: 1
[2:25:32.548910983] [2048] DEBUG IPARPI ipa_base.cpp:1482 Applying AGC Exposure: 19992.82us (Shutter lines: 2171, AGC requested 20000.00us) Gain: 1 (Gain Code: 112)
[2:25:32.548948446] [2048] DEBUG RPiAf af.cpp:723 setLensPosition: 1
[INFO] [1726218151.236218284] [camera]: camera "/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a" configured with 1024x768-YUYV stream
[WARN] [1726218151.236606508] [camera]: unsupported control 'AfWindows'
terminate called after throwing an instance of 'std::runtime_error'
  what():  control NoiseReductionMode (10002) not handled
[ros2run]: Aborted
```

</details>

anderskm, 2024-09-13:
And for reference, the output from `rpicam-hello`:

<details><summary>LIBCAMERA_LOG_LEVELS=*:INFO</summary>

```
[3:16:31.999888749] [2093]  INFO Camera camera_manager.cpp:316 libcamera v0.3.1+50-69a894c4
[3:16:32.005456916] [2094]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 13-09-2024 (07:02:13)
[3:16:32.020307559] [2094]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media1 and ISP device /dev/media0 using PiSP variant BCM2712_C0
Preview window unavailable
[3:16:32.021234525] [2093]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900
Stream configuration adjusted
[3:16:32.021707138] [2093]  INFO Camera camera.cpp:1191 configuring streams: (0) 2304x1296-YUV420 (1) 2304x1296-BGGR_PISP_COMP1
[3:16:32.021870898] [2094]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
#0 (0.00 fps) exp 32680.00 ag 2.62 dg 1.04
#1 (30.01 fps) exp 32680.00 ag 2.56 dg 1.07
#2 (30.01 fps) exp 32680.00 ag 2.70 dg 1.01
#3 (30.01 fps) exp 32680.00 ag 2.70 dg 1.01
#4 (30.01 fps) exp 32680.00 ag 2.70 dg 1.01
#5 (30.01 fps) exp 32680.00 ag 2.70 dg 1.01
#6 (30.01 fps) exp 32680.00 ag 2.70 dg 1.01
#7 (30.01 fps) exp 32680.00 ag 2.70 dg 1.01
#8 (30.01 fps) exp 32680.00 ag 2.70 dg 1.01
#9 (30.01 fps) exp 32680.00 ag 2.70 dg 1.01
...
```

</details>

<details><summary>LIBCAMERA_LOG_LEVELS=*:DEBUG   (very very long!)</summary>

```
[3:06:03.407771861] [2086] DEBUG IPAModule ipa_module.cpp:334 ipa_rkisp1.so: IPA module /ros2_ws/install/libcamera/lib/libcamera/ipa_rkisp1.so is signed
[3:06:03.407893769] [2086] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/ros2_ws/install/libcamera/lib/libcamera/ipa_rkisp1.so'
[3:06:03.408015214] [2086] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_pisp.so: IPA module /ros2_ws/install/libcamera/lib/libcamera/ipa_rpi_pisp.so is signed
[3:06:03.408044473] [2086] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/ros2_ws/install/libcamera/lib/libcamera/ipa_rpi_pisp.so'
[3:06:03.408132770] [2086] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /ros2_ws/install/libcamera/lib/libcamera/ipa_rpi_vc4.so is signed
[3:06:03.408153770] [2086] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/ros2_ws/install/libcamera/lib/libcamera/ipa_rpi_vc4.so'
[3:06:03.408209104] [2086] DEBUG IPAModule ipa_module.cpp:334 ipa_soft_simple.so: IPA module /ros2_ws/install/libcamera/lib/libcamera/ipa_soft_simple.so is signed
[3:06:03.408227159] [2086] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/ros2_ws/install/libcamera/lib/libcamera/ipa_soft_simple.so'
[3:06:03.408274363] [2086]  INFO Camera camera_manager.cpp:316 libcamera v0.3.1+50-69a894c4
[3:06:03.408418845] [2087] DEBUG Camera camera_manager.cpp:71 Starting camera manager
[3:06:03.412575288] [2087] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media1
[3:06:03.412808233] [2087] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 10 missing dependencies
[3:06:03.413995589] [2087] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[3:06:03.414015386] [2087] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: rp1-cfe
[3:06:03.414107923] [2087] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media2
[3:06:03.414142831] [2087] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 1 missing dependencies
[3:06:03.414301572] [2087] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[3:06:03.414325220] [2087] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: rpivid
[3:06:03.414488832] [2087] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media0
[3:06:03.414514906] [2087] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 9 missing dependencies
[3:06:03.414667722] [2087] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media3
[3:06:03.414694018] [2087] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 9 missing dependencies
[3:06:03.415703078] [2087] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[3:06:03.415727541] [2087] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: pispbe
[3:06:03.416737396] [2087] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[3:06:03.416760304] [2087] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: pispbe
[3:06:03.417060101] [2087] DEBUG Camera camera_manager.cpp:136 Found registered pipeline handler 'imx8-isi'
[3:06:03.417121009] [2087] DEBUG Camera camera_manager.cpp:136 Found registered pipeline handler 'mali-c55'
[3:06:03.417153287] [2087] DEBUG Camera camera_manager.cpp:136 Found registered pipeline handler 'rkisp1'
[... cut ...]
[3:06:03.751544357] [2087] DEBUG V4L2 v4l2_videodevice.cpp:1677 /dev/video21[25:out]: Dequeuing buffer 1
[3:06:03.751551765] [2087] DEBUG V4L2 v4l2_videodevice.cpp:1677 /dev/video26[26:cap]: Dequeuing buffer 1
[3:06:03.751557394] [2087] DEBUG V4L2 v4l2_videodevice.cpp:1677 /dev/video22[27:out]: Dequeuing buffer 0
[3:06:03.751563005] [2087] DEBUG V4L2 v4l2_videodevice.cpp:1677 /dev/video27[28:cap]: Dequeuing buffer 0
[3:06:03.751568376] [2087] DEBUG Event event_dispatcher_poll.cpp:216 next timer 0x7ffed80286f8 expires in 0.994742184
```

</details>

anderskm, 2024-09-13:
And with `gdb` for debugging `camera_ros`:

**Command**:
```
LIBCAMERA_LOG_LEVELS=*:DEBUG ros2 run --prefix 'gdb -ex run --args' camera_ros camera_node --ros-args --log-level
debug -p format:="YUYV" -p width:=1024 -p height:=768 -p FrameDurationLimits:="[90000,110000]"
```

<details><summary>Output</summary>

```
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /ros2_ws/install/camera_ros/lib/camera_ros/camera_node...
Starting program: /ros2_ws/install/camera_ros/lib/camera_ros/camera_node --ros-args --log-level debug -p format:=YUYV -p width:=1024 -p height:=768 -p FrameDurationLimits:=\[90000,110000\]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff66de8e0 (LWP 897)]
[DEBUG] [1726223172.344157298] [rclcpp]: signal handler installed
[DEBUG] [1726223172.344347706] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '0' services
[DEBUG] [1726223172.344161446] [rclcpp]: deferred_signal_handler(): waiting for SIGINT/SIGTERM or uninstall
[DEBUG] [1726223172.344403243] [camera_node]: Load library libcamera_component.so
[DEBUG] [1726223172.953042899] [camera_node]: Instantiate class rclcpp_components::NodeFactoryTemplate<camera::CameraNode>
[DEBUG] [1726223172.953174881] [rcl]: Couldn't parse arg 0 (/ros2_ws/install/camera_ros/lib/camera_ros/camera_node) as a remap rule in its deprecated form. Error: Expected lexeme type (19) not found, search ended at index 54, at ./src/rcl/lexer_lookahead.c:239
[DEBUG] [1726223172.953228159] [rcl]: Initializing node 'camera' in namespace ''
[DEBUG] [1726223172.953262548] [rcl]: Using domain ID of '0'
[New Thread 0x7fffebdae8e0 (LWP 898)]
[New Thread 0x7fffeb5168e0 (LWP 899)]
[New Thread 0x7fffead068e0 (LWP 900)]
[New Thread 0x7fffea4f68e0 (LWP 901)]
[New Thread 0x7fffe9ce68e0 (LWP 902)]
[New Thread 0x7fffe94d68e0 (LWP 903)]
[New Thread 0x7fffe8c068e0 (LWP 904)]
[New Thread 0x7fffe3ffe8e0 (LWP 905)]
[DEBUG] [1726223172.969589278] [rcl]: Initializing publisher for topic name '/rosout'
[DEBUG] [1726223172.969669611] [rcl]: Expanded and remapped topic name '/rosout'
[DEBUG] [1726223172.983990521] [rcl]: Publisher initialized
[DEBUG] [1726223172.984031910] [rcl]: Node initialized
[DEBUG] [1726223172.984232577] [rcl]: Initializing service for service name 'camera/get_parameters'
[DEBUG] [1726223172.984243337] [rcl]: Expanded and remapped service name '/camera/get_parameters'
[DEBUG] [1726223172.991574264] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1726223172.991604487] [rmw_fastrtps_cpp]: Sub Topic rq/camera/get_parametersRequest
[DEBUG] [1726223172.991609672] [rmw_fastrtps_cpp]: Pub Topic rr/camera/get_parametersReply
[DEBUG] [1726223172.991613857] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1726223172.991686302] [rcl]: Service initialized
[DEBUG] [1726223172.991730191] [rcl]: Initializing service for service name 'camera/get_parameter_types'
[DEBUG] [1726223172.991747931] [rcl]: Expanded and remapped service name '/camera/get_parameter_types'
[DEBUG] [1726223172.992144618] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1726223172.992163692] [rmw_fastrtps_cpp]: Sub Topic rq/camera/get_parameter_typesRequest
[DEBUG] [1726223172.992187173] [rmw_fastrtps_cpp]: Pub Topic rr/camera/get_parameter_typesReply
[DEBUG] [1726223172.992196525] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1726223172.992240044] [rcl]: Service initialized
[DEBUG] [1726223172.992267137] [rcl]: Initializing service for service name 'camera/set_parameters'
[DEBUG] [1726223172.992279526] [rcl]: Expanded and remapped service name '/camera/set_parameters'
[DEBUG] [1726223172.992581767] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1726223172.992596693] [rmw_fastrtps_cpp]: Sub Topic rq/camera/set_parametersRequest
[DEBUG] [1726223172.992601175] [rmw_fastrtps_cpp]: Pub Topic rr/camera/set_parametersReply
[... 230 lines cut ...]
[DEBUG] [1726223173.059254006] [camera]: declare ColourGains with default not set
[DEBUG] [1726223173.059396192] [camera]: declare Contrast with default 1.000000
[DEBUG] [1726223173.059527285] [camera]: declare AfPause with default 0
[WARN] [1726223173.059644211] [camera]: unsupported control 'AfWindows'
[DEBUG] [1726223173.059776396] [camera]: declare AfMetering with default 0
[DEBUG] [1726223173.059905045] [camera]: declare LensPosition with default 1.000000
[DEBUG] [1726223173.060025990] [camera]: declare AfRange with default 0
[DEBUG] [1726223173.060136434] [camera]: declare HdrMode with default 0
[DEBUG] [1726223173.060271435] [camera]: declare AfMode with default 0
[DEBUG] [1726223173.060384731] [camera]: declare AfSpeed with default 0
[... cut ...]
[DEBUG] [1726223173.060977289] [camera]: declare Sharpness with default 1.000000
[DEBUG] [1726223173.061184808] [camera]: declare AnalogueGain with default not set
terminate called after throwing an instance of 'std::runtime_error'
  what():  control NoiseReductionMode (10002) not handled

Thread 1 "camera_node" received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=140737351826016, signo=signo@entry=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
44      ./nptl/pthread_kill.c: No such file or directory.
```

</details>

christianrauch, 2024-09-13:
Previously, this exception was caught since future libcamera versions may expose additional controls that have to be handled explicitly in the node. At some point, I accidentally removed that `try`/`catch`.

Can you cherry-pick d5489392a5c58b594c5746991f2637d192f9c8c0? That adds the exception handling again and should show a warning instead of crashing the node with an unhandled exception.

christianrauch, 2024-09-13:
@anderskm Can you check if https://github.com/christianrauch/camera_ros/pull/62 solves your issue?

anderskm, 2024-09-13:
> @anderskm Can you check if #62 solves your issue?

Yes, it does! Thank you for your quick reply and fix.

christianrauch, 2024-09-13:
> > @anderskm Can you check if #62 solves your issue?
> 
> Yes, it does! Thank you for your quick reply and fix.

Alright. I will merge it then. Thanks for testing.

closed 2024-09-13 by christianrauch

christianrauch, 2024-09-13:
> After some more experimentation, a solution seems to be to add `IF(draft::NoiseReductionMode)` and `IF(rpi::ScalerCrops)` to [camera_ros/blob/main/src/type_extent.cpp](https://github.com/christianrauch/camera_ros/blob/main/src/type_extent.cpp), but I need to test it further.

I did not add the [`draft` controls](https://git.libcamera.org/libcamera/libcamera.git/tree/src/libcamera/control_ids_draft.yaml) to the list since they will at least change their namespace in any case, which will break the API once they are released out of "draft" state.

For the `rpi::ScalerCrops` control, I am wondering where you found this. The [`rpi` controls namespace](https://git.libcamera.org/libcamera/libcamera.git/tree/src/libcamera/control_ids_rpi.yaml) only lists the controls `StatsOutputEnable` and `Bcm2835StatsOutput` and I can find a reference to [`ScalerCrop`](https://git.libcamera.org/libcamera/libcamera.git/tree/src/libcamera/control_ids_core.yaml#n411) only in the standard `libcamera` namespace and this is already handled in `get_extent`.

anderskm, 2024-09-13:
It makes sense to not include the draft controls, as you mention. In my previous comment, I was simply looking for a solution to my problem without fully understanding the inner workings of things.

After "handling" NoiseReductionMode with `IF(draft::NoiseReductionMode)`, `ScalarCrops` started to cause a similar error.
I found `rpi::ScalarCrops` in Raspberry Pi's clone of `libcamera`: https://github.com/raspberrypi/libcamera/blob/main/src/libcamera/control_ids_rpi.yaml#L29
I believe, they (RPi) recommend using that over the official `libcamera` for the time being until they sync up.

christianrauch, 2024-09-13:
Thanks for the hint about the `raspberrypi` fork. I always thought they were essentially in sync with the upstream version but they forked off after commit `82c5ea24b0e6120b75cd2f21075a3cdcb9d3d7a2` (see commit upstream https://git.libcamera.org/libcamera/libcamera.git/commit/?id=82c5ea24b0e6120b75cd2f21075a3cdcb9d3d7a2 and in the fork https://github.com/raspberrypi/libcamera/commit/82c5ea24b0e6120b75cd2f21075a3cdcb9d3d7a2). The `controls::rpi::ScalerCrops` was then added via commit https://github.com/raspberrypi/libcamera/commit/554b2a55e347d6b6b44ba81c625ac94d9616d0f0 to the fork.

It's good to know that they are now API incompatible. I don't know how easy it would be to support the original upstream project and the fork as you would need to check this at build time. I am only testing with the upstream libcamera version and the binary Debian package in the ROS repos (i.e. via bloom) uses the upstream version too. I may additionally support the fork in the future if someone contributes this and provides a reliable way to detect which version is used. But until then, I am not going to support the fork.

anderskm, 2024-09-14:
I think it is perfectly fair to only support the upstream version.
If you decide to support the raspberry pi fork in the future, I can think of two ways of handling it.

`libcamera` includes the git commit sha1 of each major version in the version name. It may be a way to check whether it is the upstream version or the raspberry pi fork. But it is going to be annoying to keep the look-up-table up to date as more and more versions are added and the two versions diverge.

It might be easier to just create an `raspberry-pi` branch of `camera_ros`, and then have raspberry pi users compile `camera_ros` by themselves. That would only require a small update to the build instructions:
```
git clone https://github.com/christianrauch/camera_ros.git src/camera_ros
```
would become
```
git clone -b raspberry-pi --single-branch https://github.com/christianrauch/camera_ros.git src/camera_ros
```
and
```
git clone https://git.libcamera.org/libcamera/libcamera.git src/libcamera
```
would become
```
git clone https://github.com/raspberrypi/libcamera.git src/libcamera
```

christianrauch, 2024-09-14:
Using git commit hashes will not work if you use a binary distributed version, e.g. via Debian packages, or archived source code without the `.git ` folder. Maintaining two branches is also not a nice option since you essentially have to maintain two forks for your own project. And since I am distributing the package in binary form as Debian package via bloom, it would also be unclear which version I want to distribute.

I reached out to the `raspberrypi` fork about their plans with these custom controls (https://github.com/raspberrypi/libcamera/issues/178) and they are indeed trying to merge them upstream. So I am optimistic that this problem will eventually vanish.

anderskm, 2024-09-14:
> I reached out to the `raspberrypi` fork about their plans with these custom controls ([raspberrypi/libcamera#178](https://github.com/raspberrypi/libcamera/issues/178)) and they are indeed trying to merge them upstream. So I am optimistic that this problem will eventually vanish.

Good call reaching out to them. Fingers crossed, that the merger happens sooner rather than later :-)

If you need help with any manual testing on with Raspberry Pi cameras, I will be happy to help.

askarkg12, 2024-09-21:
@anderskm Could you please share how you managed to get rpicam-hello to work in a container?
I have been building images using buildx on amd64 machine, but with platform being arm64.

@christianrauch I appreciate that specific docker related problems might not be as relevant to camera-ros, however would having camera-ros ready container be beneficial for this repo? 
I currently have Raspberry Pi 5 and Raspberry Pi Camera Module 3 (the same sensor as anderskm) and there has not been a smooth option that I could find yet.

christianrauch, 2024-09-22:
> I appreciate that specific docker related problems might not be as relevant to camera-ros, however would having camera-ros ready container be beneficial for this repo?

There are already binary (Ubuntu) packages for `libcamera` and `camera_ros` available in the ROS repos via bloom. I don't see the benefit of the Docker overhead on a low-powered system, such as the Raspberry Pi, over binary Debian packages.

But this is another topic anyway and not related to the original issue. If you encounter issues with the ROS 2 node, you should first check that you can run the ROS-independent standalone examples under the same conditions.

anderskm, 2024-09-23:
@askarkg12:
> Could you please share how you managed to get rpicam-hello to work in a container?

I build and install both `libcamera` and `rpicam-apps` from source using two bash-scripts. These scripts are called, when the docker image is built. Note, that `libcamera` is cloned from Raspberry Pi's repo rather than the official libcamera repo.
I use [this dockerfile](https://github.com/osrf/docker_images/blob/master/ros/humble/ubuntu/jammy/ros-base/Dockerfile) as base image, but I doubt that matters in this context.

<details><summary>install_libcamera.sh</summary>

```
###############
## libcamera ##
###############
# NOTE:
    # libcamera can be install via `apt`, but currently only an old version (v0.1.0) is provided. Therefore, libcamera is compiled from source instead.
    # Raspberry Pi has forked libcamera from the official libcamera repository, while they are working on (better) hardware support for the RPi cameras.
    # The RPi fork is planned to be merged into the official libcamera repo: https://github.com/raspberrypi/libcamera/issues/178

sudo apt-get update
sudo apt install -y python3-pip ninja-build
# Note: Install meson using pip (v>=1.5.1), as the dep version is too old (v0.61) compared to what is needed (v0.63)
sudo apt remove meson
sudo pip install meson # colcon-meson

sudo apt install -y libboost-dev
sudo apt install -y libgnutls28-dev openssl libtiff-dev pybind11-dev
sudo apt install -y qtbase5-dev libqt5core5a libqt5widgets5
sudo apt install -y cmake
sudo apt install -y python3-yaml python3-ply
sudo apt install -y libglib2.0-dev libgstreamer-plugins-base1.0-dev
# For libcamera 'cam'-app [use cam -l to list available cameras]
sudo apt install -y libevent-dev libdrm-dev libjpeg-dev libsdl2-dev

sudo apt install -y python3-jinja2

cd /tmp
git clone https://github.com/raspberrypi/libcamera.git
cd /tmp/libcamera
meson setup build --buildtype=release -Dpipelines=rpi/vc4,rpi/pisp -Dipas=rpi/vc4,rpi/pisp -Dv4l2=true -Dgstreamer=enabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=disabled -Ddocumentation=disabled -Dpycamera=enabled
ninja -C build install
ldconfig
cd /tmp
rm -rf /tmp/libcamera
```

</details>

<details><summary>install_rpicam_apps.sh</summary>

```
#################
## rpicam-apps ##
#################
# Source: https://www.raspberrypi.com/documentation/computers/camera_software.html#build-libcamera-and-rpicam-apps

sudo apt update
sudo apt install -y python3-pip ninja-build
# Install meson from pip as it is a significantly newer version
sudo pip install meson

sudo apt install -y libepoxy-dev libjpeg-dev libtiff5-dev libpng-dev

# To use the Qt preview window, install the following additional dependencies:
sudo apt install -y qtbase5-dev libqt5core5a libqt5gui5 libqt5widgets5
#For libav support in rpicam-vid, install the following additional dependencies:
sudo apt install -y libavcodec-dev libavdevice-dev libavformat-dev libswresample-dev

sudo apt install -y cmake libboost-program-options-dev libdrm-dev libexif-dev
sudo apt install -y libepoxy-dev
cd /tmp
git clone https://github.com/raspberrypi/rpicam-apps.git
cd /tmp/rpicam-apps

# Build rpicam-apps
meson setup build -Denable_libav=enabled -Denable_drm=enabled -Denable_egl=enabled -Denable_qt=enabled -Denable_opencv=disabled -Denable_tflite=disabled
meson compile -C build
meson install -C build
ldconfig
rpicam-still --version

# Clean-up
cd /tmp
rm -rf /tmp/rpicam-apps
```

</details>

I build the image using:

```
docker build --progress=plain -t ros_docker .
```

And I start the docker with the following command:

```
docker run -it --privileged  --net host -v /dev/:/dev/ -v /run/udev:/run/udev ros_docker:latest
```
You could probably narrow down the two mounts (`-v`).

PR #62 (2024-09-13): fix control exception
This handles the exception which is thrown when an unknown control is encountered during the parameter type conversion.

Fixes #61.