christianrauch/camera_ros #63: Throws an error (camera with name does not exists) while providing camera name instead of ID.

niktaiwade, 2024-10-20:
When providing a camera name instead of a camera ID, the software does not detects the cameras.

System Architecture: 2 USB cameras are connected, which are supposed monitor different angles and run some processing code by subscribing to the respective topics.
If camera ID is provided which changes (or swaps in this case) which changes the resulting output.

The software results in following error:

terminate called after throwing an instance of 'std::runtime_error'
[camera_node-1]   what():  camera with name /dev/v4l/by-id/usb-Generic_USB_Camera_200901010001-video-index0 does not exist 

Ros2 launch code:

Node(
            package='camera_ros',
            executable='camera_node',
            name='camera_node',
            output='screen',
            parameters=[{'camera': '/dev/v4l/by-id/usb-Generic_USB_Camera_200901010001-video-index0'},{'width': 1920},{'height':1080}]
        )

christianrauch, 2024-10-20:
`/dev/v4l/by-id/usb-Generic_USB_Camera_200901010001-video-index0` is not a camera name. It's the V4L2 device path.

When you start the node without specifying a camera, it will list all available cameras with their name in brackets:
```
>> cameras:
   0: Integrated IR Camera: Integrate (\_SB_.PCI0.XHC_.RHUB.HS05-5:1.0-5986:2123)
   1: Integrated Camera: Integrated C (\_SB_.PCI0.XHC_.RHUB.HS08-8:1.0-5986:2115)
```
Here, the camera names are `\_SB_.PCI0.XHC_.RHUB.HS05-5:1.0-5986:2123` and `\_SB_.PCI0.XHC_.RHUB.HS08-8:1.0-5986:2115`. If the camera name contains special characters, such as in this case, you also have to escape them or place them in quotes:
```sh
ros2 run camera_ros camera_node --ros-args -p camera:="\_SB_.PCI0.XHC_.RHUB.HS05-5:1.0-5986:2123"
```

niktaiwade, 2024-10-20:
Yes, Got it, thanks for the very fast update.

closed 2024-10-20 by niktaiwade