christianrauch/camera_ros #64: Can't configure camera with invalid configuration

StevenYui, 2024-10-28:
Hello everyone

My device and environment is Rasberry Pi 5 + Ubuntu 24.04 + ros2
I have built both the 'libcamera' and 'rpicam-apps' libraries, and 'rpicam-hello' is working.

`
Available cameras
-----------------
0 : imx219 [3280x2464 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@88000/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [30.00 fps - (65535, 65535)/65535x65535 crop]
                             1640x1232 [30.00 fps - (65535, 65535)/65535x65535 crop]
                             1920x1080 [30.00 fps - (65535, 65535)/65535x65535 crop]
                             3280x2464 [30.00 fps - (65535, 65535)/65535x65535 crop]
           'SRGGB8' : 640x480 [30.00 fps - (65535, 65535)/65535x65535 crop]
                      1640x1232 [30.00 fps - (65535, 65535)/65535x65535 crop]
                      1920x1080 [30.00 fps - (65535, 65535)/65535x65535 crop]
                      3280x2464 [30.00 fps - (65535, 65535)/65535x65535 crop]

1 : imx219 [3280x2464 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@80000/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [103.33 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 152)/1920x2160 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [103.33 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 152)/1920x2160 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]

`

But I can't run the node and launch file because of the error: 'Can't configure camera with invalid configuration.'

> ros2 run camera_ros camera_node --ros-args -p camera:=0 -p format:="SRGGB8" -p width:=1640 -p height:=1232

> LIBCAMERA_LOG_LEVELS=*:DEBUG ros2 run --prefix 'gdb -ex run --args' camera_ros camera_node --ros-args --log-level debug -p camera:=0 -p format:="SRGGB8" -p width:=1640 -p height:=1232

`
[0:52:32.410324601] [9195] DEBUG Camera camera_manager.cpp:159 Pipeline handler "rpi/pisp" matched
[0:52:32.410401564] [9195] DEBUG RPI pisp.cpp:865 Unable to acquire a CFE instance
[DEBUG] [1730106367.085952381] [camera]: found camera by id: 0
[0:52:32.414912879] [9122]  WARN V4L2 v4l2_pixelformat.cpp:346 Unsupported V4L2 pixel format RPBP
[0:52:32.415305083] [9122] DEBUG Camera camera.cpp:1132 streams configuration: (0) 1920x1080-YUV420
[DEBUG] [1730106367.090737640] [camera]: default stream configuration: "1920x1080-YUV420"
[DEBUG] [1730106367.091193177] [camera]: 
>> stream formats:
   - Pixelformat: R8 (32x32 - 3280x2464)
   - Pixelformat: R16 (32x32 - 3280x2464)
   - Pixelformat: MONO_PISP_COMP1 (32x32 - 3280x2464)
   - Pixelformat: SGRBG10 (32x32 - 3280x2464)
   - Pixelformat: SGBRG10 (32x32 - 3280x2464)
   - Pixelformat: SBGGR10 (32x32 - 3280x2464)
   - Pixelformat: SRGGB10 (32x32 - 3280x2464)
   - Pixelformat: NV21 (32x32 - 3280x2464)
   - Pixelformat: SBGGR8 (32x32 - 3280x2464)
   - Pixelformat: SGRBG12 (32x32 - 3280x2464)
   - Pixelformat: SGBRG12 (32x32 - 3280x2464)
   - Pixelformat: SBGGR12 (32x32 - 3280x2464)
   - Pixelformat: SRGGB12 (32x32 - 3280x2464)
   - Pixelformat: YUV420 (32x32 - 3280x2464)
   - Pixelformat: NV12 (32x32 - 3280x2464)
   - Pixelformat: YVU420 (32x32 - 3280x2464)
   - Pixelformat: SBGGR16 (32x32 - 3280x2464)
   - Pixelformat: BGGR_PISP_COMP1 (32x32 - 3280x2464)
   - Pixelformat: SGRBG14 (32x32 - 3280x2464)
   - Pixelformat: SGBRG14 (32x32 - 3280x2464)
   - Pixelformat: SBGGR14 (32x32 - 3280x2464)
   - Pixelformat: SRGGB14 (32x32 - 3280x2464)
   - Pixelformat: XBGR8888 (32x32 - 3280x2464)
   - Pixelformat: BGR888 (32x32 - 3280x2464)
   - Pixelformat: RGB888 (32x32 - 3280x2464)
   - Pixelformat: XRGB8888 (32x32 - 3280x2464)
   - Pixelformat: YUV444 (32x32 - 3280x2464)
   - Pixelformat: YVU444 (32x32 - 3280x2464)
   - Pixelformat: SGBRG16 (32x32 - 3280x2464)
   - Pixelformat: GBRG_PISP_COMP1 (32x32 - 3280x2464)
   - Pixelformat: SGRBG16 (32x32 - 3280x2464)
   - Pixelformat: GRBG_PISP_COMP1 (32x32 - 3280x2464)
   - Pixelformat: YUV422 (32x32 - 3280x2464)
   - Pixelformat: YVU422 (32x32 - 3280x2464)
   - Pixelformat: SRGGB16 (32x32 - 3280x2464)
   - Pixelformat: RGGB_PISP_COMP1 (32x32 - 3280x2464)
   - Pixelformat: BGR161616 (32x32 - 3280x2464)
   - Pixelformat: RGB161616 (32x32 - 3280x2464)
   - Pixelformat: SRGGB8 (32x32 - 3280x2464)
   - Pixelformat: SGRBG8 (32x32 - 3280x2464)
   - Pixelformat: SGBRG8 (32x32 - 3280x2464)
   - Pixelformat: YVYU (32x32 - 3280x2464)
   - Pixelformat: YUYV (32x32 - 3280x2464)
   - Pixelformat: VYUY (32x32 - 3280x2464)
   - Pixelformat: UYVY (32x32 - 3280x2464)
[0:52:32.420319638] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 640x480 fmt SRGGB10 Score: 4504.81 (best 4504.81)
[0:52:32.420434657] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 1640x1232 fmt SRGGB10 Score: 1000 (best 1000)
[0:52:32.420450583] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 1920x1080 fmt SRGGB10 Score: 1541.48 (best 1000)
[0:52:32.420459453] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 3280x2464 fmt SRGGB10 Score: 1718 (best 1000)
[0:52:32.420470897] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 640x480 fmt SRGGB8 Score: 3504.81 (best 1000)
[0:52:32.420479934] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 1640x1232 fmt SRGGB8 Score: 0 (best 0)
[0:52:32.420491879] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 1920x1080 fmt SRGGB8 Score: 541.478 (best 0)
[0:52:32.420499564] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 3280x2464 fmt SRGGB8 Score: 718 (best 0)
[INFO] [1730106367.095753603] [camera]: 
>> stream formats:
   - Pixelformat: R8 (32x32 - 3280x2464)
   - Pixelformat: NV21 (32x32 - 3280x2464)
   - Pixelformat: SBGGR8 (32x32 - 3280x2464)
   - Pixelformat: SBGGR16 (32x32 - 3280x2464)
   - Pixelformat: BGGR_PISP_COMP1 (32x32 - 3280x2464)
   - Pixelformat: XBGR8888 (32x32 - 3280x2464)
   - Pixelformat: BGR888 (32x32 - 3280x2464)
   - Pixelformat: RGB888 (32x32 - 3280x2464)
   - Pixelformat: XRGB8888 (32x32 - 3280x2464)
   - Pixelformat: SGBRG16 (32x32 - 3280x2464)
   - Pixelformat: GBRG_PISP_COMP1 (32x32 - 3280x2464)
   - Pixelformat: SGRBG16 (32x32 - 3280x2464)
   - Pixelformat: GRBG_PISP_COMP1 (32x32 - 3280x2464)
   - Pixelformat: SRGGB16 (32x32 - 3280x2464)
   - Pixelformat: RGGB_PISP_COMP1 (32x32 - 3280x2464)
   - Pixelformat: SRGGB8 (32x32 - 3280x2464)
   - Pixelformat: SGRBG8 (32x32 - 3280x2464)
   - Pixelformat: SGBRG8 (32x32 - 3280x2464)
   - Pixelformat: YUYV (32x32 - 3280x2464)
   - Pixelformat: UYVY (32x32 - 3280x2464)
[WARN] [1730106367.095948548] [camera]: stream configuration adjusted from "1640x1232-SRGGB8" to "1640x1232-SBGGR16"
[0:52:32.420987675] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 640x480 fmt SRGGB10 Score: 6504.81 (best 6504.81)
[0:52:32.421039360] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 1640x1232 fmt SRGGB10 Score: 3000 (best 3000)
[0:52:32.421050416] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 1920x1080 fmt SRGGB10 Score: 3541.48 (best 3000)
[0:52:32.421062323] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 3280x2464 fmt SRGGB10 Score: 3718 (best 3000)
[0:52:32.421071620] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 640x480 fmt SRGGB8 Score: 7504.81 (best 3000)
[0:52:32.421080027] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 1640x1232 fmt SRGGB8 Score: 4000 (best 3000)
[0:52:32.421089564] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 1920x1080 fmt SRGGB8 Score: 4541.48 (best 3000)
[0:52:32.421099675] [9122] DEBUG RPI pipeline_base.cpp:979 Format: 3280x2464 fmt SRGGB8 Score: 4718 (best 3000)
[0:52:32.421124453] [9122] ERROR Camera camera.cpp:1185 Can't configure camera with invalid configuration
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to configure streams

Thread 1 "camera_node" received signal SIGABRT, Aborted.
Download failed: Invalid argument.  Continuing without source file ./nptl/./nptl/pthread_kill.c.
__pthread_kill_implementation (threadid=281474842202144, signo=signo@entry=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
warning: 44     ./nptl/pthread_kill.c: No such file or directory
`

That's my first time posting an issue on GitHub. If there are any problems, please let me know.

christianrauch, 2024-10-28:
Does this work without specifying the format and the dimensions at the same time, i.e. if you either specify `-p format:="SRGGB8"` or `-p width:=1640 -p height:=1232`?

Can you also check if changing the `role` from its default `video` to `still` or `raw`?

StevenYui, 2024-11-04:
Sorry for the late recommendation. 
After trying the `role` for `raw` and `viewfinder`, the node runs great! 
Thank you for your help!

wad652, 2024-11-21:
Hello @StevenYui. I have been trying to get this package running with a Raspberry Pi 5, Ubuntu 24.04, ROS2-Jazzy and the Raspberry Pi global shutter camera. From your previous post I can see that you were able to get this package running in a similar configuration. How have you managed that? Have you modified anything? Can you give me a quick overview? Thank you in advance :)

StevenYui, 2024-11-29:
Hello @wad652, you need to check if you have built "libcamera" and "rpicam-apps."
After building both projects, run "rpicam-hello --list-cameras" to verify that the camera can be detected.

christianrauch, 2025-01-02:
I was able to reproduce a similar behaviour with a `imx708_wide` (Camera Module 3 / Sony IMX708) but could not resolve this due to the auto adjustment to an invalid configuration. I was only able to work around this, by selecting a supported configuration manually, that is not auto-adjusted to an invalid configuration.

This is quite cumbersome and not a nice out-of-the box experience for new users. Eventually, I decided to change the default value for `role` to `viewfinder` (https://github.com/christianrauch/camera_ros/pull/76) in order to have a valid default configuration.

This issue should now be resolved on HEAD via PR https://github.com/christianrauch/camera_ros/pull/76. Let me know here if you are still encountering this issue with the current state, and provide information on how to reproduce this.

closed 2025-01-02 by christianrauch