christianrauch/camera_ros #67: Unable to Configure and Run camera_ros After Reinstallation

codermery, 2024-11-28:
Description
I previously used the camera_ros package successfully on my Raspberry Pi 5 without any additional configuration. However, after resetting my computer, I’ve been unable to get it working despite following the same steps and reinstalling all required dependencies.

The node fails to configure the camera and consistently throws errors. I also noticed that both libcamera and camera_ros have been updated recently, and reverting to older versions did not solve the problem. Additionally, I can still capture images using QCam, so libcamera appears to be functioning correctly.

System Details
Hardware: Raspberry Pi 5 + IMX708 camera module (2 units)
Software:
OS: Ubuntu 24.04
ROS 2 Jazzy
libcamera
v4l2
camera_ros

Current Issues and Errors
Libcamera Error:
Failed stream start: Invalid argument (22)
Camera_ros Error:
Can't configure camera with invalid configuration
terminate called after throwing an instance of 'std::runtime_error'
what():  failed to configure streams
Steps to Reproduce
Installed all dependencies for camera_ros and libcamera.
Verified the camera works with QCam.
Attempted to run camera_ros using the following commands:
ros2 run camera_ros camera_node --ros-args -p camera:=0 -p format:="YUYV" -p width:=1640 -p height:=1232
or with adjusted configurations:
ros2 run camera_ros camera_node --ros-args -p camera:=0
Tried modifying the role parameter to raw and viewfinder, but the errors persisted.
Reverted both libcamera and camera_ros to previous versions.
What I Tried
Verified that QCam can still capture images, confirming that libcamera works.
Used supported formats and resolutions (e.g., 1640x1232-YUYV).
Reinstalled all dependencies.
Tried previous versions of both libcamera and camera_ros.
Expected Behavior
The node should configure the camera successfully and stream images, as it did during the initial setup without any manual adjustments.

Questions
Are there any known issues with the latest updates to libcamera or camera_ros?
How can I prevent these configuration errors?
Is there a recommended debugging or setup process to ensure compatibility between libcamera and camera_ros?
Additional Context
The current situation has become very frustrating, as the same hardware and software setup worked flawlessly during my initial installation. Any help or guidance would be greatly appreciated.

This version is formatted for GitHub Issues and includes all relevant details to make your problem clear and reproducible. You can copy-paste this directly into the Issues section of the camera_ros repository.

christianrauch, 2024-11-28:
> 1. **v4l2_camera Error:**
> 
> ```shell
> Failed stream start: Invalid argument (22)
> ```

Are you referring to https://gitlab.com/boldhearts/ros2_v4l2_camera.git? This is not using libcamera. Are you trying to use the Raspberry Pi camera with V4L2 instead of libcamera?

> 3. **Camera_ros Error:**
> 
> ```
> Can't configure camera with invalid configuration
> terminate called after throwing an instance of 'std::runtime_error'
> what():  failed to configure streams
> Abort.
> ```

Where is this message coming from? The node does not show an error "Can't configure camera with invalid configuration".

The `failed to configure streams` exception looks like a duplicate of https://github.com/christianrauch/camera_ros/issues/64. Did you try any of the workarounds there?

Did you have a look at the ["Trouble Shooting" section](https://github.com/christianrauch/camera_ros?tab=readme-ov-file#trouble-shooting)? Especially enabling debug information for libcamera and the ROS node should provide more useful information. Can you also list explicitly which commands you run? It's not yet clear to me which commands cause which error.

Since you also list "2 units", do you have 2 cameras connected to the board? Does this work with a single camera only or with a different camera connected to the board?

codermery, 2024-11-29:
### Current Issues and Observations

#### 1. V4L2 and Libcamera Usage
I am using the following resources:
- **Libcamera**: [Libcamera GitHub](https://github.com/raspberrypi/libcamera)
- **V4L2**: [V4L2 GitHub](https://github.com/tier4/ros2_v4l2_camera)

I understand that V4L2 does not utilize Libcamera, but the issue persists even when I use Libcamera exclusively.

#### 2. Error: `Can't configure camera with invalid configuration`
- The error occurs as follows:
  ```text
  [0:10:09.009613503] [3434] ERROR Camera camera.cpp:1171 Can't configure camera with invalid configuration
  terminate called after throwing an instance of 'std::runtime_error'
    what():  failed to configure streams
- I referred to Issue #64 and tried the suggested workarounds, but the issue persists.

#### 3. Pixel Format Errors in Debug Logs
- Default pixel format YUV420 is not supported:
```text
[WARN] [camera]: default pixel format (YUV420) not supported
```
- Adjustments auto-select to "1920x1080-R8", but still fail:
```text
[WARN] [camera]: stream configuration adjusted from "1920x1080-R8" to "2304x1296-SBGGR16"
```

#### 4. QSocketNotifier Error
-This occurs intermittently:
```text
QObject::connect: Cannot queue arguments of type 'QSocketNotifier*'
```

#### 5. Compilation Errors (Libcamera)
- When compiling with Libcamera, switch errors are encountered:
```text
/home/pixbot/ros2_ws/src/camera_ros/src/clamp.cpp: In function ‘libcamera::ControlValue clamp(const libcamera::ControlValue&, const libcamera::ControlValue&, const libcamera::ControlValue&)’:
error: enumeration value ‘ControlTypePoint’ not handled in switch [-Werror=switch]

```
-Similar issues occur in cv_to_pv.cpp and other files.

####  Observations
- Cameras are detected correctly in debug logs:
```text
>> cameras:
   0: imx708 (/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a)
   1: imx708 (/base/axi/pcie@120000/rp1/i2c@80000/imx708@1a)

```
- Qcam is able to provide an image feed, so the hardware seems functional.

- These issues began after re-installing the environment on a reset system.

#### Steps Tried
-Referred to [Trouble Shooting](https://github.com/raspberrypi/libcamera/wiki/Troubleshooting).
-Enabled detailed debugging using LIBCAMERA_LOG_LEVELS=*:DEBUG.
-Verified hardware by testing with Qcam.

christianrauch, 2024-11-29:
> #### 1. V4L2 and Libcamera Usage
> I am using the following resources:
> 
> * **Libcamera**: [Libcamera GitHub](https://github.com/raspberrypi/libcamera)

The `libcamera` repo at https://github.com/raspberrypi/libcamera is a fork of the upstream repo that is not supported. Use the official upstream repo or the binary packages from the ROS repo.

> #### 2. Error: `Can't configure camera with invalid configuration`
> * The error occurs as follows:
>   ```
>   [0:10:09.009613503] [3434] ERROR Camera camera.cpp:1171 Can't configure camera with invalid configuration
>   terminate called after throwing an instance of 'std::runtime_error'
>     what():  failed to configure streams
>   ```

You still have to provide the commands that you run and the debug information.

> #### 3. Pixel Format Errors in Debug Logs
> * Default pixel format YUV420 is not supported:
> 
> ```
> [WARN] [camera]: default pixel format (YUV420) not supported
> ```
> 
> * Adjustments auto-select to "1920x1080-R8", but still fail:
> 
> ```
> [WARN] [camera]: stream configuration adjusted from "1920x1080-R8" to "2304x1296-SBGGR16"
> ```

Did you try to enable debug information?

> #### 4. QSocketNotifier Error
> -This occurs intermittently:
> 
> ```
> QObject::connect: Cannot queue arguments of type 'QSocketNotifier*'
> ```

This repo is not using Qt. So this error cannot come from the ROS node.

> #### 5. Compilation Errors (Libcamera)
> * When compiling with Libcamera, switch errors are encountered:
> 
> ```
> /home/pixbot/ros2_ws/src/camera_ros/src/clamp.cpp: In function ‘libcamera::ControlValue clamp(const libcamera::ControlValue&, const libcamera::ControlValue&, const libcamera::ControlValue&)’:
> error: enumeration value ‘ControlTypePoint’ not handled in switch [-Werror=switch]
> ```
> 
> -Similar issues occur in cv_to_pv.cpp and other files.

Use the newest released version from the upstream repo or the binary package from the ROS repo.

> #### Observations
> * Cameras are detected correctly in debug logs:
> 
> ```
> >> cameras:
>    0: imx708 (/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a)
>    1: imx708 (/base/axi/pcie@120000/rp1/i2c@80000/imx708@1a)
> ```

Did you try with a single camera as suggested?

> * Qcam is able to provide an image feed, so the hardware seems functional.
> * These issues began after re-installing the environment on a reset system.

What could I conclude from this? I don't know what change and could have caused an issue.

> #### Steps Tried
> -Referred to [Trouble Shooting](https://github.com/raspberrypi/libcamera/wiki/Troubleshooting). 

This links to the non-existing website https://github.com/raspberrypi/libcamera/wiki/Troubleshooting. What should I find there?

> -Enabled detailed debugging using LIBCAMERA_LOG_LEVELS=*:DEBUG.

Where is this debug log?

> -Verified hardware by testing with Qcam.

If you want help, you have to try the suggestions and provide the information that is asked for. Since I do not have the hardware to test and reproduce this myself, I rely on the information from those that have the hardware. But if you cannot provide information to better understand the issue and see if it is different from the other `failed to configure streams` issue, I have to close those since this just consumes time without any progress.

codermery, 2024-11-30:
#### **Hardware and Software Used**

> Device: Raspberry Pi 5
> Camera: Raspberry Pi Camera Module 3
> Operating Systems: `Ubuntu 24.04` (64-bit) arm
> Libcamera Version: v0.3.0
> ROS 2 Version:` Jazzy`

#### **Findings**
Camera Not Detected on Ubuntu:
The `"Available cameras"` section is empty.
The error `"Unable to acquire a Unicam instance"` was encountered.
Camera Works Seamlessly on Raspbian:
The camera is detected, and image capture is successful.
The ROS 2 node runs without issues.

Driver Compatibility:
It was observed that camera drivers (especially Unicam) are not functioning correctly on Ubuntu.

The Raspberry Pi Camera Module 3 is not properly detected on Ubuntu, whereas it works seamlessly on Raspbian. This indicates a potential incompatibility with camera drivers or the libcamera configuration on Ubuntu.

LIBCANERA_LOG_LEVELS=*:DEBUG cam -l
```text
[1:11:25.540318062] [5901] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so is signed
[1:11:25.540429339] [5901] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
[1:11:25.540470987] [5901]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0
[1:11:25.540642060] [5904] DEBUG Camera camera_manager.cpp:69 Starting camera manager
[1:11:25.568831078] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
[1:11:25.570020219] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 10 missing dependencies
[1:11:25.573099571] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[1:11:25.573152033] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
[1:11:25.573346588] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media1
[1:11:25.573655327] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 10 missing dependencies
[1:11:25.575633537] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[1:11:25.575648370] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: rp1-cfe
[1:11:25.575769092] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media3
[1:11:25.575779444] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 1 missing dependencies
[1:11:25.576069646] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[1:11:25.576092034] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: rpivid
[1:11:25.576450699] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media2
[1:11:25.576489069] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 9 missing dependencies
[1:11:25.576858363] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media4
[1:11:25.576922789] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media4 due to 9 missing dependencies
[1:11:25.579778716] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[1:11:25.579868216] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: pispbe
[1:11:25.583751767] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media4 found
[1:11:25.583789174] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media4: pispbe
[1:11:25.584485355] [5904] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'rpi/vc4'
[1:11:25.584546762] [5904] DEBUG RPI vc4.cpp:178 Unable to acquire a Unicam instance
[1:11:25.584554836] [5904] DEBUG RPI vc4.cpp:178 Unable to acquire a Unicam instance
Available cameras:

```

~/ros2_ws$ ros2 run camera_ros camera_node 

```text
[1:07:20.248118571] [5694] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so is signed
[1:07:20.248228923] [5694] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
[1:07:20.279787433] [5694]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0
[1:07:20.280057192] [5709] DEBUG Camera camera_manager.cpp:69 Starting camera manager
[1:07:20.306533728] [5709] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
[1:07:20.306835877] [5709] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 10 missing dependencies
[1:07:20.308694918] [5709] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[1:07:20.308708622] [5709] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
[1:07:20.308849844] [5709] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media1
[1:07:20.309106900] [5709] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 10 missing dependencies
[... cut ...]
[1:07:20.311068849] [5709] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media3
[... 101 lines cut ...]
    /dev/media4
rp1-cfe (platform:1f00110000.csi):
    /dev/video8
    /dev/video9
    /dev/video10
    /dev/video11
    /dev/video12
    /dev/video13
    /dev/video14
    /dev/video15
    /dev/media0
rp1-cfe (platform:1f00128000.csi):
    /dev/video0
    /dev/video1
    /dev/video2
    /dev/video3
    /dev/video4
    /dev/video5
    /dev/video6
    /dev/video7
    /dev/media1
rpivid (platform:rpivid):
    /dev/video19
    /dev/media3
```

christianrauch, 2024-11-30:
Your issue report is inconsistent.

Previously, you reported that the cameras were detected and only failed during the configuration via`failed to configure streams`. Now you report that cameras are not detected `no cameras available`. You also report that the libcamera example `cam` does not detect the cameras. I don't see how the ROS node can help you with that. If the cameras are not detected because of configuration issues etc., then the ROS node cannot use them.

Please consider the Raspberry Pi documentation on how to set up the cameras.

Also, if `ros2 run camera_ros camera_node` already fails to detect cameras, why would the launch file `ros2 launch camera_ros camera.launch.py` help?

Please go back to the original issue of the configuration failure (`failed to configure streams`) and provide the debug information that was asked for.

PR #40 (2025-01-02): troubleshooting common issues
This cleans up the documentation a bit, adds information about the ["raspberrypi" fork of libcamera](https://github.com/raspberrypi/libcamera) and instructions for troubleshooting common issues with camera detection and buffer allocation, both specifically on the Raspberry Pi.

The documentation now mentions that the official upstream version of libcamera may not support all, and especially newer, Raspberry Pi Camera Modules. A common pitfall of Raspberry Pi users is that they use the official upstream version with newer camera modules, which then cannot be detect [... cut]
closed 2025-01-02 by christianrauch