christianrauch/camera_ros #68: build error with 0.3.2 and after

sonelu, 2024-12-04:
It looks like there were a number of additional data types in the `libcamera::ControlValue` that result in compile errors:

```
/home/mh5/mh5_ws/src/camera_ros/src/clamp.cpp: In function 'libcamera::ControlValue clamp(const libcamera::ControlValue&, const libcamera::ControlValue&, const libcamera::ControlValue&)':
/home/mh5/mh5_ws/src/camera_ros/src/clamp.cpp:128:10: error: enumeration value 'ControlTypePoint' not handled in switch [-Werror=switch]
  128 |   switch (value.type()) {
      |          ^
/home/mh5/mh5_ws/src/camera_ros/src/clamp.cpp: In function 'bool operator<(const libcamera::ControlValue&, const libcamera::ControlValue&)':
/home/mh5/mh5_ws/src/camera_ros/src/clamp.cpp:254:10: error: enumeration value 'ControlTypePoint' not handled in switch [-Werror=switch]
  254 |   switch (lhs.type()) {
      |          ^
/home/mh5/mh5_ws/src/camera_ros/src/clamp.cpp: In function 'bool operator>(const libcamera::ControlValue&, const libcamera::ControlValue&)':
/home/mh5/mh5_ws/src/camera_ros/src/clamp.cpp:275:10: error: enumeration value 'ControlTypePoint' not handled in switch [-Werror=switch]
  275 |   switch (lhs.type()) {
      |          ^
/home/mh5/mh5_ws/src/camera_ros/src/cv_to_pv.cpp: In function 'rclcpp::ParameterValue cv_to_pv(const libcamera::ControlValue&)':
/home/mh5/mh5_ws/src/camera_ros/src/cv_to_pv.cpp:96:10: error: enumeration value 'ControlTypePoint' not handled in switch [-Werror=switch]
   96 |   switch (value.type()) {
      |          ^
/home/mh5/mh5_ws/src/camera_ros/src/cv_to_pv.cpp: In function 'rclcpp::ParameterType cv_to_pv_type(const libcamera::ControlId*)':
/home/mh5/mh5_ws/src/camera_ros/src/cv_to_pv.cpp:115:12: error: enumeration value 'ControlTypePoint' not handled in switch [-Werror=switch]
  115 |     switch (id->type()) {
      |            ^
/home/mh5/mh5_ws/src/camera_ros/src/cv_to_pv.cpp:135:12: error: enumeration value 'ControlTypePoint' not handled in switch [-Werror=switch]
  135 |     switch (id->type()) {
      |            ^
```

I think it would make sense to change this (needs to be done in multiple places):

```cpp
  switch (value.type()) {
    CASE_NONE(None)
    CASE_CLAMP(Bool)
    CASE_CLAMP(Byte)
    CASE_CLAMP(Integer32)
    CASE_CLAMP(Integer64)
    CASE_CLAMP(Float)
    CASE_CLAMP(String)
    CASE_CLAMP(Rectangle)
    CASE_CLAMP(Size)
  }

  return {};
```

into this:
```cpp
  switch (value.type()) {
    CASE_NONE(None)
    CASE_CLAMP(Bool)
    CASE_CLAMP(Byte)
    CASE_CLAMP(Integer32)
    CASE_CLAMP(Integer64)
    CASE_CLAMP(Float)
    CASE_CLAMP(String)
    CASE_CLAMP(Rectangle)
    CASE_CLAMP(Size)
  default:
    return {};
  }
```

christianrauch, 2024-12-04:
I am currently only supporting tagged releases, and `ControlTypePoint` was added after the latest 0.3.2 release. I recommend using the binary `libcamera` release in the ROS repos, or if you have to build from source, use the latest tag `v0.3.2`. I will add support for new types once they are tagged & released.

sonelu, 2024-12-04:
Understand.

The confusing part is that Raspberry Pi fork is a little ahead. The version that is installed with the latest system is v0.3.2+rpt20241119 and that includes the commit with the additional ControlType:

https://github.com/raspberrypi/libcamera/blob/1230f78d2f6d38d812568fd27f7b985b9ff19e39/include/libcamera/controls.h#L28

And unfortunatelly there is no deb package for raspberry pi with the plain 0.3.2 to downgrade to:
```
mh5@mh5g:~/mh5_ws $ apt list libcamera0.3*
Listing... Done
libcamera0.3-dbgsym/stable 0.3.2+rpt20241119-1 arm64
libcamera0.3-dbgsym/stable 0.3.2+rpt20241119-1 armhf
libcamera0.3/stable,now 0.3.2+rpt20241119-1 arm64 [installed,automatic]
libcamera0.3/stable 0.3.2+rpt20241119-1 armhf
```
For the time being I have removed the `-Werror` flag in the cmake file. The warnings are there to be dealt with in the future, but build finishes.

christianrauch, 2024-12-04:
> The confusing part is that Raspberry Pi fork is a little ahead. The version that is installed with the latest system is v0.3.2+rpt20241119 and that includes the commit with the additional [...]

I only support the upstream `libcamera` project (https://git.libcamera.org/libcamera/libcamera.git). The latest tagged version `v0.3.2` (https://git.libcamera.org/libcamera/libcamera.git/tag/?h=v0.3.2) does not contain these changes. The `raspberrypi` fork contains unreleased and sometimes unreviewed changes. I don't have the capacity to support an additional fork, and I also do not want to support features that are not officially supported by libcamera upstream. To support a new upcoming feature, it has to be reviewed, merged and released. 

> And unfortunatelly there is no deb package for raspberry pi with the plain 0.3.2 to downgrade to:
>
> ```
> mh5@mh5g:~/mh5_ws $ apt list libcamera0.3*
> Listing... Done
> libcamera0.3-dbgsym/stable [...]

I am not totally clear how you use ROS with the Raspberry Pi. If you use Raspberry Pi OS, then are you compiling all ROS packages, including `camera_ros`, from source? There are bloomed libcamera packages (`apt list ros-*-libcamera`) that are derived from the official upstream repo. These are available for all platforms and architectures supported by bloom (Ubuntu, RHEL).

Tsaimingchun14, 2025-02-12:
@sonelu How did you build camera_ros from source inside the conda env? I try to use the libcamera pre-installed in raspberrypi os by setting `export PKG_CONFIG_PATH=/usr/lib/aarch64-linux-gnu/pkgconfig:$PKG_CONFIG_PATH` and it seems to find it. But later on it couldn't find the libamera/*.h files.

sonelu, 2025-02-12:
@Tsaimingchun14  

first, make sure that you installed the `libcamera-dev` on your machine. The standard system installation installs only `libcamera` libraries, not the headers:
```
sudo apt install libcamera-dev
```
at this moment this will install the 0.3.2 RPi fork (0.3.2+rpt20241119-1).
The installer will place the libraries in `/usr/lib/aarch-linux-gnu/` and the headers in `/usr/include/libcamera` (where there is another `libcamera` folder, hence the `#include libcamera/...h` in the code!).

You can now try again with the `export`.

I did it slightly different:
- copied the `pc` files from `/usr/lib/aarch-linux-gnu/pkgconfig/libcamera*` to the environment: `/home/<user>/miniforge3/envs/<env>/lib/pkgconfig/`. There should be 2 of them: `libcamera-base.pc` and `libcamera.pc`.
- then modified the files to remove the definitions at start and hardcoding the paths in clean, for instance `libcamera-base.pc` changes from:
```
prefix=/usr
includedir=${prefix}/include
libdir=${prefix}/lib/aarch64-linux-gnu

Name: libcamera-base
Description: Camera support base utility library
Version: 0.3.2
Libs: -L${libdir} -lcamera-base
Cflags: -I${includedir}/libcamera
```
into:
```
Name: libcamera-base
Description: Camera support base utility library
Version: 0.3.2
Libs: -L$/usr/lib/aarch64-linux-gnu -lcamera-base
Cflags: -I$/usr/include/libcamera
```
You should be able to compile `camera_ros` in your environment without the export.

On a side note, I was thinking to put a PR for a feedstock for libcamera, libcamera-dev as a conda-forge package so that you can install them with conda/mamba directly in the environment, instead of the `apt install route` that seems to create problems with packages in virtual environments. If there is enough interest for this, I can try to follow up with the idea.