christianrauch/camera_ros #72: Unable to discover Raspberry Pi Camera on RPi 5/Ubuntu 24.05/ROS 2 Jazzy

ARLunan, 2024-12-14:
Deploying the Camera with config.txt : camera_auto_detect=1, camera appears as rpe-cfe  after successfully installing Binary libcamera and building camera _ros, able to launch camera_node with result below: When using a USB Webcam an image is published /camera/image/raw - implying the Package works with a Usb camera.

With Raspi Camera v2.1 (IMX219) which checked out ok with raspberry-os:

I understand that the RPi5 doe not use the RPi 4 bcm2835 driver that may be an issue here, but would appreciate an assessment of the observed behavior or troubleshooting steps.

Run package: (Same behavior with ros2 launch camera_ros camera.launch.py)
ubuntu@rp5-ub24j-mb:~/camera_ws$ ros2 run camera_ros camera_node
[0:38:26.527483208] [5760]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2
terminate called after throwing an instance of 'std::runtime_error'
  what():  no cameras available
[ros2run]: Aborted

Checking the device list a RPi camera on video0 appears to be detected. 

ubuntu@rp5-ub24j-mb:~/camera_ws$ ls -l /dev/video0
crw-rw----+ 1 root video 81, 10 Dec 13 18:37 /dev/video0

ubuntu@rp5-ub24j-mb:~/camera_ws$ v4l2-ctl --list-devices
pispbe (platform:1000880000.pisp_be):
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/video24
	/dev/video25
	/dev/video26
	/dev/video27
	/dev/video28
	/dev/video29
	/dev/video30
	/dev/video31
	/dev/video32
	/dev/video33
	/dev/video34
	/dev/video35
	/dev/video36
	/dev/video37
	/dev/media1
	/dev/media3

rp1-cfe (platform:1f00110000.csi):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/media0

rpivid (platform:rpivid):
	/dev/video19
	/dev/media2

christianrauch, 2024-12-14:
Which OS are you using? Is that camera supported on the Raspberry Pi 5 with that OS?

For generic questions about camera support on specific Raspberry Pi boards or OSes, please refer to https://forums.raspberrypi.com.

yassinfrh, 2024-12-14:
I have the same issue and I'm using the Raspberry Pi 5 with Ubuntu 24.05 and the camera is Raspberrry Pi Camera Rev 1.3. The camera is supported and I tested it using rpicam-hello --qt-preview.

ARLunan, 2024-12-14:
Tnx 4 response and hope this is useful information on the failure of the ros2 camera_ros package to find the RPi Camera on /dev/video0. Machine is a Raspberry Pi 5/4G. My previous posted log from v4l2-ctl is using installed OS Canonical Ubuntu 24.04/ROS 2 Jazzy on the Raspberry Pi5/Raspberry Pi Camera v2.1. Kernal is 6.8.0-1015-raspi

ubuntu@rp5-ub24j-mb:~$ uname -a
Linux rp5-ub24j-mb 6.8.0-1015-raspi #17-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 11 14:12:16 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

To test the RPi & Camera on a booted raspberrypi-os  directly equipped with rpi-os SDCard: $ uname -a > Linux raspberrypi-os 6.6.62+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux) 

Connected Camera is a Raspberry Pi V2.1 Camera imx219 which functions on a raspberrypi-os SD Card boot displayed below.  

Note the reference to BCM2712 instead of the BCM2835 overlay

pi@raspberrypi-os:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx219 [3280x2464 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@88000/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [83.70 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]

pi@raspberrypi-os:~ $ rpicam-hello
[0:14:09.910454986] [2080]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+99-1230f78d
[0:14:09.918000761] [2083]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 29-08-2024 (16:33:32)
[0:14:09.934087666] [2083]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
Made X/EGL preview window
Mode selection for 1640:1232:12:P
    SRGGB10_CSI2P,640x480/0 - Score: 4504.81
    SRGGB10_CSI2P,1640x1232/0 - Score: 1000
    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
    SRGGB10_CSI2P,3280x2464/0 - Score: 1718
    SRGGB8,640x480/0 - Score: 5504.81
    SRGGB8,1640x1232/0 - Score: 2000
    SRGGB8,1920x1080/0 - Score: 2541.48
    SRGGB8,3280x2464/0 - Score: 2718
Stream configuration adjusted
[0:14:12.104037600] [2080]  INFO Camera camera.cpp:1197 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-BGGR_PISP_COMP1
[0:14:12.104139619] [2083]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B

christianrauch, 2024-12-14:
@ARLunan I understand that you have two different OS setups for the same Raspberry Pi + Camera combination. Can you systematically run all the commands that you are using for debugging on both OS setups for comparison?

Also, is there evidence that the Ubuntu 24.04 image on the Raspberry Pi 5 is indeed supporting the camera you want to use? Do you have a working example where the camera runs on Ubuntu without the ROS node?

ARLunan, 2024-12-16:
Following up on your question about Raspberry Pi 5 support for Raspberry Pi CSI Camera, unfortunately no. Please refer to this post by other developers and myself investigating the matter . [Raspberry Pi Camera Support](https://askubuntu.com/posts/1535607/revisions). The critical  situation appears to be BCM2711 Library which is different than the one in a Raspberry Pi 4 that uses the BCM2835 Library, where the RPI Camera correctly works. Not even the Canonical Apps work with the CSI Camera So until some means to support the RPI CSI  Camera on Canonical Ubuntu 24.04 is developed, ROS 2 Nodes are a problem.

christianrauch, 2024-12-16:
> Following up on your question about Raspberry Pi 5 support for Raspberry Pi CSI Camera, unfortunately no. Please refer to this post by other developers and myself investigating the matter .

Looking at https://askubuntu.com/questions/1535607/do-ubuntu-24-04-raspberry-pi-desktop-camera-and-cheese-apps-support-raspberry-pi, the question is rather new, and it will take some time to get an answer.

> The critical situation appears to be BCM2711 Library which is different than the one in a Raspberry Pi 4 that uses the BCM2835 Library, where the RPI Camera correctly works.

What do you mean by "BCM2711 Library which is different [...] BCM2835 Library"? These are different chip sets, not libraries. You need kernel driver support for these, of course. If the chip set is not fully supported by Ubuntu, you may need to build custom kernel modules or install a different kernel altogether.

> So until some means to support the RPI CSI Camera on Canonical Ubuntu 24.04 is developed, ROS 2 Nodes are a problem.

Not sure why "ROS 2 Nodes are a problem" here. If the kernel is missing driver support (kernel modules) for some hardware needed to run the cameras, then you simply cannot use anything relying on libcamera or other software support. There is nothing a ROS 2 node can do here.

lukmace, 2024-12-24:
I encountered the same issue, and here’s how I resolved it:

    For the camera module V2 (IMX219), use **start_x=1** in /boot/firmware/config.txt.
    For the camera module V3 (IMX708), use **camera_auto_detect=1** in /boot/firmware/config.txt.

Everything works perfectly for me now. I’m running Ubuntu 24.04 on a Raspberry Pi 5, and I’m using ROS 2 Jazzy for my robot.

christianrauch, 2024-12-28:
> I encountered the same issue, and here’s how I resolved it:
>
> ```
> For the camera module V2 (IMX219), use **start_x=1** in /boot/firmware/config.txt.
> For the camera module V3 (IMX708), use [...]

I cannot reproduce this on a Raspberry Pi 5 with the `imx708`, the Raspberry Pi OS and the upstream libcamera (https://git.libcamera.org/libcamera/libcamera.git) following the instructions at https://libcamera.org/getting-started.html.

Are you using the upstream version of libcamera or the Raspberry Pi fork?

The camera is correctly detected:
```
$ sudo vclog --msg
[...]
006547.715: dtb_file 'bcm2712-rpi-5-b.dtb'
006630.772: Found camera 'imx708' on port 0, unicam_port 0
006702.450: Loaded overlay 'imx708'
006825.687: dtparam: audio=on
006830.284: Unknown dtparam 'audio' - ignored
006886.400: Loaded overlay 'vc4-kms-v3d-pi5'
[...]
```
and `camera_auto_detect` is already set correctly:
```
$ grep camera_auto_detect /boot/firmware/config.txt 
camera_auto_detect=1
```

It works with the libcamera packages that come with Raspberry Pi OS though. So this is very likely related to which libcamera fork you use.

PR #40 (2025-01-02): troubleshooting common issues
This cleans up the documentation a bit, adds information about the ["raspberrypi" fork of libcamera](https://github.com/raspberrypi/libcamera) and instructions for troubleshooting common issues with camera detection and buffer allocation, both specifically on the Raspberry Pi.

The documentation now mentions that the official upstream version of libcamera may not support all, and especially newer, Raspberry Pi Camera Modules. A common pitfall of Raspberry Pi users is that they use the official upstream version with newer camera modules, which then cannot be detect [... cut]
closed 2025-01-02 by christianrauch