christianrauch/camera_ros #79: Place cropped image in the center

a-uhlig, 2025-01-09:
I am running the camera_ros node successfully on my Raspberry Pi. But I want to crop the image so that it is square (e.g. 1280x1280). Therefor I set the `widht` and the `height` Parameters. This crops the output image, but unfortunately the crop is at one edge.
Is it possible to place the cutout in the center?

christianrauch, 2025-01-09:
TL;DR: No, you have to crop the image yourself. But what you want to do is quite common and there probably already exist a ROS node doing this. Have a look at the [`image_proc::CropDecimateNode`](https://docs.ros.org/en/jazzy/p/image_proc/doc/components.html#image-proc-cropdecimatenode).

---

> I am running the camera_ros node successfully on my Raspberry Pi. But I want to crop the image so that it is square (e.g. 1280x1280).

The node does not have something like a cropping functionality. When you set a resolution, libcamera will try to find the closest matching camera configuration, and it is then up to the camera how to provide the data with the requested configuration. 

> Therefor I set the `widht` and the `height` Parameters. This crops the output image, but unfortunately the crop is at one edge.

I assumed you set both to 1280 and I was about to say that this configuration does not exist. At least on the Camera Module 3 (`imx708_wide`), there is no such resolution listed for the `XRGB8888` format:
```
>> XRGB8888 format sizes:
   - 160x120
   - 240x160
   - 320x240
   - 400x240
   - 480x320
   - 640x360
   - 640x480
   - 720x480
   - 768x480
   - 854x480
   - 720x576
   - 800x600
   - 960x540
   - 1024x576
   - 960x640
   - 1024x600
   - 1024x768
   - 1280x720
   - 1152x864
   - 1280x800
   - 1360x768
   - 1366x768
   - 1440x900
   - 1280x1024
   - 1536x864
   - 1280x1080
   - 1600x900
   - 1400x1050
   - 1680x1050
   - 1600x1200
   - 1920x1080
   - 2048x1080
   - 1920x1200
   - 2160x1080
   - 2048x1152
   - 2560x1080
   - 2048x1536
   - 2560x1440
   - 2560x1600
   - 3840x1080
   - 2960x1440
   - 3440x1440
   - 2560x2048
   - 3200x1800
   - 3840x1600
   - 3200x2048
   - 3200x2400
   - 3840x2160
   - 4096x2160
   - 3840x2400
```

But the camera is indeed correctly configured with `1280x1280-XRGB8888` when I start it with `--ros-args -p width:=1280 -p height:=1280`. I don't know what is going on here in the background.

> Is it possible to place the cutout in the center?

Unless the camera has such a dedicated control: No. The parameters `width` and `height` set the desired resolution of the stream configuration and do not actively crop anything. It is up to the camera how to do this. If you need the `1280x1280` configuration to behave differently, you can open an issue with the Raspberry Pi team for them to send patches to libcamera. The camera_ros node just forwards the data provided by libcamera. If you need anything on top of this, you have to do this in a different node.

a-uhlig, 2025-01-10:
Thank you very much, the image_proc::CropDecimateNode works in general. Although I'm still struggling a bit with my configuration.

With an older version of ROS2 (and a different camera node) it was easier. There I could specify any size and get a centered crop.

But that's probably not up to you. I guess it has to do with the change of the driver stack from v4l to libcamera on the Raspberry.

closed 2025-01-10 by a-uhlig