christianrauch/camera_ros #80: [Build Error] "enumeration value ‘ControlTypePoint’ not handled in switch" while building together with raspberrypi/libcamera

horizon105457, 2025-01-10:
### Abstract:
Error `enumeration value ‘ControlTypePoint’ not handled in switch` while building together with raspberrypi/libcamera.
Latest "raspberrypi/libcamera" contains new attributes of libcamera 0.4 but has a version number of 0.3.2. The patch for official v0.4 [implement support for new control types](https://github.com/christianrauch/camera_ros/commit/2e974e067a7c59694b6e472db40e1ea25cb9c97f) does not take effect.

## Problem:

Hi,
I'm trying to build this package from source for a docker container(humble-percetion-jammy) on raspberry pi 4B. I had built a "raspberrypi/libcamera" instead of official libcamera as the dependency of "rpicam-apps" and "camera_ros".
When I compiled "camera_ros", it reported errors with following message:

## Error Message:

```bash
$ colcon build --event-handlers=console_direct+
Starting >>> camera_ros
Consolidate compiler generated dependencies of target utils
[  8%] Building CXX object CMakeFiles/utils.dir/src/clamp.cpp.o
/drivers_ws/src/camera_ros/src/clamp.cpp: In function ‘libcamera::ControlValue clamp(const libcamera::ControlValue&, const libcamera::ControlValue&, const libcamera::ControlValue&)’:
/drivers_ws/src/camera_ros/src/clamp.cpp:146:10: error: enumeration value ‘ControlTypePoint’ not handled in switch [-Werror=switch]
  146 |   switch (value.type()) {
      |          ^
/drivers_ws/src/camera_ros/src/clamp.cpp: In function ‘bool operator<(const libcamera::ControlValue&, const libcamera::ControlValue&)’:
/drivers_ws/src/camera_ros/src/clamp.cpp:299:10: error: enumeration value ‘ControlTypePoint’ not handled in switch [-Werror=switch]
  299 |   switch (lhs.type()) {
      |          ^
/drivers_ws/src/camera_ros/src/clamp.cpp: In function ‘bool operator>(const libcamera::ControlValue&, const libcamera::ControlValue&)’:
/drivers_ws/src/camera_ros/src/clamp.cpp:325:10: error: enumeration value ‘ControlTypePoint’ not handled in switch [-Werror=switch]
  325 |   switch (lhs.type()) {
      |          ^
cc1plus: all warnings being treated as errors     
gmake[2]: *** [CMakeFiles/utils.dir/build.make:76: CMakeFiles/utils.dir/src/clamp.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/utils.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
--- stderr: camera_ros                            
/drivers_ws/src/camera_ros/src/clamp.cpp: In function ‘libcamera::ControlValue clamp(const libcamera::ControlValue&, const libcamera::ControlValue&, const libcamera::ControlValue&)’:
/drivers_ws/src/camera_ros/src/clamp.cpp:146:10: error: enumeration value ‘ControlTypePoint’ not handled in switch [-Werror=switch]
  146 |   switch (value.type()) {
      |          ^
/drivers_ws/src/camera_ros/src/clamp.cpp: In function ‘bool operator<(const libcamera::ControlValue&, const libcamera::ControlValue&)’:
/drivers_ws/src/camera_ros/src/clamp.cpp:299:10: error: enumeration value ‘ControlTypePoint’ not handled in switch [-Werror=switch]
  299 |   switch (lhs.type()) {
      |          ^
/drivers_ws/src/camera_ros/src/clamp.cpp: In function ‘bool operator>(const libcamera::ControlValue&, const libcamera::ControlValue&)’:
/drivers_ws/src/camera_ros/src/clamp.cpp:325:10: error: enumeration value ‘ControlTypePoint’ not handled in switch [-Werror=switch]
  325 |   switch (lhs.type()) {
      |          ^
cc1plus: all warnings being treated as errors
gmake[2]: *** [CMakeFiles/utils.dir/build.make:76: CMakeFiles/utils.dir/src/clamp.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/utils.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< camera_ros [5.69s, exited with code 2]

Summary: 0 packages finished [6.86s]
  1 package failed: camera_ros
  1 package had stderr output: camera_ros
```
## Error Tracing
I noticed that the error log points to the variable type **ControlTypePoint** which was newly added at libcamera 0.4 (https://patchwork.libcamera.org/patch/21442/) .Recent commits fixed it by adding support code while `LIBCAMERA_VER_GE(0, 4, 0) == true`, but in my case, that seems to have been ignored. 
Perhaps it's because "raspberrypi/libcamera" has a different version order. In the latest v0.3.2, "raspberrypi/libcamera" already has these new control types. 

I am doing further tests including override the version compare and the results remain to be verified.  If you need more information I can provide it.

horizon105457, 2025-01-10:
This issue is out of date. please refer to https://github.com/christianrauch/camera_ros/issues/68#issuecomment-2518158539

closed 2025-01-10 by horizon105457

christianrauch, 2025-01-11:
Support for the new control types is there. It is just, as you recognised, that this only works with the 0.4 version and the raspberrypi fork has not released a version 0.4 yet. Either create an issue at https://github.com/raspberrypi/libcamera that they are out of sync with libcamera if you want to use the `camera_ros` upstream code, or manually patch a `add_compile_options(-Wno-error=switch)` into the CMake to turn those errors into warnings again.