christianrauch/camera_ros #83: Enabling R8/Mono8 format for OV9281

bobzwik, 2025-02-01:
Hello! 

I've got my camera_ros node working properly, the [OV9281 camera](https://www.waveshare.com/ov9281-110-camera.htm) I have is streaming and I can view `/camera/image_raw` and `/camera/image_raw/compressed` on Rviz2.

However, I can't help but notice that the stream gets configured to used a color pixel format, even though the OV9281 is a mono sensor. 

```
root@rpi4:/ros2_ws# ros2 run camera_ros camera_node --ros-args -p width:=640 -p height:=400 -p FrameDurationLimits:="[50000,50000]"
[0:56:03.999564967] [307]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0
[0:56:04.012420947] [317]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:56:04.014413963] [317]  INFO RPI vc4.cpp:401 Registered camera /base/soc/i2c0mux/i2c@1/ov9281@60 to Unicam device /dev/media3 and ISP device /dev/media2
[INFO] [1738428715.375194600] [camera]: 
>> cameras:
   0: ov9281 (/base/soc/i2c0mux/i2c@1/ov9281@60)
[WARN] [1738428715.375599762] [camera]: no camera selected, using default: "/base/soc/i2c0mux/i2c@1/ov9281@60"
[WARN] [1738428715.375727353] [camera]: set parameter 'camera' to silence this warning
[INFO] [1738428715.376720010] [camera]: 
>> stream formats:
   - NV21 (64x64 - 1280x800)
   - XBGR8888 (64x64 - 1280x800)
   - BGR888 (64x64 - 1280x800)
   - RGB888 (64x64 - 1280x800)
   - XRGB8888 (64x64 - 1280x800)
   - YUYV (64x64 - 1280x800)
   - UYVY (64x64 - 1280x800)
[WARN] [1738428715.377004248] [camera]: no pixel format selected, auto-selecting: "XRGB8888"
[WARN] [1738428715.377091191] [camera]: set parameter 'format' to silence this warning
[INFO] [1738428715.377223727] [camera]: 
>> XRGB8888 format sizes:
   - 160x120
   - 240x160
   - 320x240
   - 400x240
   - 480x320
   - 640x360
   - 640x480
   - 720x480
   - 768x480
   - 854x480
   - 720x576
   - 800x600
   - 960x540
   - 1024x576
   - 960x640
   - 1024x600
   - 1024x768
   - 1280x720
   - 1280x800
[0:56:04.017446655] [307]  INFO Camera camera.cpp:1202 configuring streams: (0) 640x400-XRGB8888
[0:56:04.020198960] [317]  INFO RPI vc4.cpp:570 Sensor: /base/soc/i2c0mux/i2c@1/ov9281@60 - Selected sensor format: 640x400-Y10_1X10 - Selected unicam format: 640x400-Y10P
[INFO] [1738428715.389532601] [camera]: camera "/base/soc/i2c0mux/i2c@1/ov9281@60" configured with 640x400-XRGB8888 stream
[WARN] [1738428715.394509514] [camera]: unknown control: NoiseReductionMode (10002)
[0:56:04.051682972] [323]  WARN IPARPI ipa_base.cpp:1087 Could not set SHARPNESS - no sharpen algorithm
[0:56:04.052263225] [323]  WARN IPARPI ipa_base.cpp:1224 No HDR algorithm available
[INFO] [1738428715.837292178] [camera]: using default calibration URL
[INFO] [1738428715.837504436] [camera]: camera calibration URL: file:///root/.ros/camera_info/ov9281__base_soc_i2c0mux_i2c_1_ov9281_60_640x400.yaml
[ERROR] [1738428715.837771174] [camera_calibration_parsers]: Unable to open camera calibration file [/root/.ros/camera_info/ov9281__base_soc_i2c0mux_i2c_1_ov9281_60_640x400.yaml]
[WARN] [1738428715.837873784] [camera]: Camera calibration file /root/.ros/camera_info/ov9281__base_soc_i2c0mux_i2c_1_ov9281_60_640x400.yaml not found
[0:56:04.668536100] [323]  WARN IPARPI ipa_base.cpp:1087 Could not set SHARPNESS - no sharpen algorithm
[0:56:04.668727247] [323]  WARN IPARPI ipa_base.cpp:1224 No HDR algorithm available
```
All the suggested pixel formats are color formats. If I understand what I'm reading correctly, libcamera has the sensor streaming with a Y10 format (10 bits per pixel) (`Selected sensor format: 640x400-Y10_1X10 - Selected unicam format: 640x400-Y10P`), so `camera_ros` suggests formats that are at least 10 bits per pixel from [this list](https://github.com/christianrauch/camera_ros/blob/63e23e7e553a361ce6b6e14ff882240c1ac2b461/src/format_mapping.cpp#L23) to stream as ROS messages.

According to [libcamera](https://git.libcamera.org/libcamera/libcamera.git/tree/src/libcamera/formats.cpp)
NV21: 12 bits per pixel
YUYV / yuv422_yuy2: 16 bits per pixel
UYVY / yuv422: 16 bits per pixel
BGR888 / rgb8 = 24 bits per pixel
XBGR8888 / rgba8 = 32 bits per pixel

I don't mind using YUYV and 16 bits per pixel, but I was wondering if I could set the camera to stream in `R8` pixel format (8 bits per pixel) and see if that reduces the computational load. Sure, the node, when using YUYV, only uses 6-7% of a single core of my Raspberry Pi 4, but I'd like to have to option to have 8 bit per pixel. **But I am unable to figure out how to set the stream to R8**. 

Note: While the node works when setting `-p format:=NV21`, Rviz2 is unable to decode it, nor is 'camera_node' able to compress it.

Here are some libcamera-hello commands that demonstrates that the R8 mode function properly.

```
bobzwik@rpi4:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : ov9281 [1280x800 10-bit MONO] (/base/soc/i2c0mux/i2c@1/ov9281@60)
    Modes: 'R8' : 640x400 [325.52 fps - (0, 0)/1280x800 crop]
                  1280x720 [180.51 fps - (0, 0)/1280x720 crop]
                  1280x800 [150.97 fps - (0, 0)/1280x800 crop]
           'R10_CSI2P' : 640x400 [260.42 fps - (0, 0)/1280x800 crop]
                         1280x720 [144.40 fps - (0, 0)/1280x720 crop]
                         1280x800 [120.76 fps - (0, 0)/1280x800 crop]
```

```
bobzwik@rpi4:~ $ libcamera-hello --viewfinder-mode 640:400:8 --framerate 30
[0:02:12.407198962] [2103]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+50-83cb8101
[0:02:12.446040745] [2106]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:02:12.447065978] [2106]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov9281@60 to Unicam device /dev/media3 and ISP device /dev/media1
[0:02:12.447142307] [2106]  INFO RPI pipeline_base.cpp:1121 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 640:400:8:P(30)
    R8,640x400/325.521 - Score: 0
    R8,1280x720/180.505 - Score: 306.667
    R8,1280x800/150.966 - Score: 260
    R10_CSI2P,640x400/260.417 - Score: 1000
    R10_CSI2P,1280x720/144.404 - Score: 1306.67
    R10_CSI2P,1280x800/120.758 - Score: 1260
Stream configuration adjusted
[0:02:12.454371471] [2103]  INFO Camera camera.cpp:1202 configuring streams: (0) 640x400-YUV420 (1) 640x400-R8
[0:02:12.454642325] [2106]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov9281@60 - Selected sensor format: 640x400-Y8_1X8 - Selected unicam format: 640x400-GREY
[0:02:12.457796482] [2109]  WARN IPARPI ipa_base.cpp:1109 Could not set SHARPNESS - no sharpen algorithm
#0 (0.00 fps) exp 33145.00 ag 8.00 dg 1.00
#1 (28.80 fps) exp 33145.00 ag 8.00 dg 1.00
#2 (28.80 fps) exp 33145.00 ag 8.00 dg 1.00
#3 (28.80 fps) exp 33145.00 ag 8.00 dg 1.00
```

But I obtain this error when passing `-p format:=R8` on the `camera_ros camera_node`
```
[INFO] [1738425788.111140928] [camera]: 
>> stream formats:
   - NV21 (64x64 - 1280x800)
   - XBGR8888 (64x64 - 1280x800)
   - BGR888 (64x64 - 1280x800)
   - RGB888 (64x64 - 1280x800)
   - XRGB8888 (64x64 - 1280x800)
   - YUYV (64x64 - 1280x800)
   - UYVY (64x64 - 1280x800)
terminate called after throwing an instance of 'std::runtime_error'
  what():  unsupported pixel format "R8"
[ros2run]: Aborted
```

Details on my setup:
- Raspberry Pi 4
- Raspberry Pi OS Bookworm 64bit
- Docker "FROM ros:humble-ros-base"
- Cloned this repo and raspberrypi/libcamera, tried tag v0.3.2 and v0.4.0

I am pretty new to ROS and libcamera, hence why I require guidance for simply configuring the pixel format to R8.

Thanks!

christianrauch, 2025-02-01:
Can you try out other stream roles (https://github.com/christianrauch/camera_ros?tab=readme-ov-file#static-camera-stream-configuration) and also provide the debug information (https://github.com/christianrauch/camera_ros?tab=readme-ov-file#trouble-shooting)?

Can you also list the stream information provided by the "cam" example from libcamera (`cam --camera 1 --info`)?

The camera lists the supported formats, which depends on the stream role and if the format can be mapped 1:1 to a ROS message format. You cannot select a format that is not supported.

bobzwik, 2025-02-02:
Setting to `role:=raw` did the trick! `R8` practically halved the CPU load, even when compressing. Thanks!

`R8` is the only available mode for `role:=raw`, I guess if I want 10bit I'm return to the default role.

christianrauch, 2025-02-02:
> `R8` is the only available mode for `role:=raw`, I guess if I want 10bit I'm return to the default role.

Which 10 bit mode are you talking about? The ROS messages only define `mono8` and `mono16` formats. As suggested, you can list all formats for the selected stream role using the debug flag. This may list a 10 bit monochrome format, but since there is no 1:1 mapping to any ROS format, it will be filtered out.

Can you simply run the node with the debug flag to list all formats and then paste the output here?

bobzwik, 2025-02-02:
The OV9281 only supports 8 bit monochrome (`R8`) or 10 bit monochrome (`R10`/`Y10`).

When running the "non-debug" node, `vc4.cpp` selects the pixel format `Y10` and the node selects by default a XRGB8888 stream (but I can switch it to RGB888 or YUYV) as seen in the first command output of my first post.

Here is the debug output:
```
Reading symbols from /ros2_ws/install/camera_ros/lib/camera_ros/camera_node...
Starting program: /ros2_ws/install/camera_ros/lib/camera_ros/camera_node --ros-args --log-level debug -p width:=640 -p height:=480
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ff66fe8e0 (LWP 90)]
[DEBUG] [1738505470.875624865] [rclcpp]: signal handler installed
[DEBUG] [1738505470.875626346] [rclcpp]: deferred_signal_handler(): waiting for SIGINT/SIGTERM or uninstall
[DEBUG] [1738505470.880932259] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '0' services
[DEBUG] [1738505470.884522701] [camera_node]: Load library libcamera_component.so
[DEBUG] [1738505473.827948755] [camera_node]: Instantiate class rclcpp_components::NodeFactoryTemplate<camera::CameraNode>
[DEBUG] [1738505473.830536780] [rcl]: Couldn't parse arg 0 (/ros2_ws/install/camera_ros/lib/camera_ros/camera_node) as a remap rule in its deprecated form. Error: Expected lexeme type (19) not found, search ended at index 54, at ./src/rcl/lexer_lookahead.c:239
[DEBUG] [1738505473.830654681] [rcl]: Initializing node 'camera' in namespace ''
[DEBUG] [1738505473.830710937] [rcl]: Using domain ID of '0'
[New Thread 0x7fec0ee8e0 (LWP 91)]
[New Thread 0x7feb8578e0 (LWP 92)]
[New Thread 0x7feb0478e0 (LWP 93)]
[New Thread 0x7fea8378e0 (LWP 94)]
[New Thread 0x7fea0278e0 (LWP 95)]
[New Thread 0x7fe98178e0 (LWP 96)]
[New Thread 0x7fe8f478e0 (LWP 97)]
[New Thread 0x7fe3ffe8e0 (LWP 98)]
[DEBUG] [1738505473.934279113] [rcl]: Initializing publisher for topic name '/rosout'
[DEBUG] [1738505473.934501008] [rcl]: Expanded and remapped topic name '/rosout'
[DEBUG] [1738505473.982161841] [rcl]: Publisher initialized
[DEBUG] [1738505473.982348405] [rcl]: Node initialized
[DEBUG] [1738505473.986331298] [rcl]: Initializing service for service name 'camera/get_parameters'
[DEBUG] [1738505473.986516010] [rcl]: Expanded and remapped service name '/camera/get_parameters'
[DEBUG] [1738505474.011506607] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1738505474.011714244] [rmw_fastrtps_cpp]: Sub Topic rq/camera/get_parametersRequest
[DEBUG] [1738505474.011850644] [rmw_fastrtps_cpp]: Pub Topic rr/camera/get_parametersReply
[DEBUG] [1738505474.011910789] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1738505474.012210791] [rcl]: Service initialized
[DEBUG] [1738505474.012388282] [rcl]: Initializing service for service name 'camera/get_parameter_types'
[DEBUG] [1738505474.012475203] [rcl]: Expanded and remapped service name '/camera/get_parameter_types'
[DEBUG] [1738505474.013832352] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1738505474.013978807] [rmw_fastrtps_cpp]: Sub Topic rq/camera/get_parameter_typesRequest
[DEBUG] [1738505474.014028341] [rmw_fastrtps_cpp]: Pub Topic rr/camera/get_parameter_typesReply
[DEBUG] [1738505474.014072413] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1738505474.014295382] [rcl]: Service initialized
[DEBUG] [1738505474.014428208] [rcl]: Initializing service for service name 'camera/set_parameters'
[DEBUG] [1738505474.014511945] [rcl]: Expanded and remapped service name '/camera/set_parameters'
[DEBUG] [1738505474.015941571] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1738505474.016084563] [rmw_fastrtps_cpp]: Sub Topic rq/camera/set_parametersRequest
[DEBUG] [1738505474.016128116] [rmw_fastrtps_cpp]: Pub Topic rr/camera/set_parametersReply
[DEBUG] [1738505474.016143782] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1738505474.016291052] [rcl]: Service initialized
[DEBUG] [1738505474.016507596] [rcl]: Initializing service for service name 'camera/set_parameters_atomically'
[DEBUG] [1738505474.016557741] [rcl]: Expanded and remapped service name '/camera/set_parameters_atomically'
[DEBUG] [1738505474.018304128] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1738505474.018479804] [rmw_fastrtps_cpp]: Sub Topic rq/camera/set_parameters_atomicallyRequest
[DEBUG] [1738505474.018533783] [rmw_fastrtps_cpp]: Pub Topic rr/camera/set_parameters_atomicallyReply
[DEBUG] [1738505474.018580261] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1738505474.018823082] [rcl]: Service initialized
[DEBUG] [1738505474.018926298] [rcl]: Initializing service for service name 'camera/describe_parameters'
[DEBUG] [1738505474.018976351] [rcl]: Expanded and remapped service name '/camera/describe_parameters'
[DEBUG] [1738505474.020880452] [rmw_fastrtps_cpp]: ************ Service Details *********
[DEBUG] [1738505474.021007593] [rmw_fastrtps_cpp]: Sub Topic rq/camera/describe_parametersRequest
[DEBUG] [1738505474.021041388] [rmw_fastrtps_cpp]: Pub Topic rr/camera/describe_parametersReply
[DEBUG] [1738505474.021227711] [rmw_fastrtps_cpp]: ***********
[DEBUG] [1738505474.021487900] [rcl]: Service initialized
[DEBUG] [1738505474.021666483] [rcl]: Initializing service for service name 'camera/list_parameters'
[DEBUG] [1738505474.021728295] [rcl]: Expanded and remapped service name '/camera/list_parameters'
[... cut ...]
[0:17:44.712336706] [124] DEBUG V4L2 v4l2_videodevice.cpp:1578 /dev/video20[25:out]: Prepared to import 32 buffers
[0:17:44.713021987] [124] DEBUG RPI vc4.cpp:286 Preparing 1 buffers for stream ISP Output0
[0:17:44.714880866] [124] DEBUG V4L2 v4l2_videodevice.cpp:1311 /dev/video21[26:cap]: 1 buffers requested.
```

Which does output `default pixel format (R10_CSI2P) not supported`.

So what is the viewfinder role actually doing? Is it re-encoding the `R10` format into 8 bit color channels, and loosing 2 bits of precision?

christianrauch, 2025-02-02:
Thanks for providing the detailed output. The listed formats are expected for the selected roles. There are three different stream roles (plus `raw`) with different purposes. See the [libcamera documentation](https://www.libcamera.org/api-html/classlibcamera_1_1StreamUsage.html#aab99a06a62416abbbb9d6c44c491c61f) for details.

The default `viewfinder` role is meant for showing the image on a screen. Therefore, it makes sense to provide colour formats. For processing the image, you are usually would use either `video` and `still` as a stream role. The `viewfinder` role is more meant as a preview of the image. The full libcamera API allows you to capture multiple streams simultaneously.

The `raw` role shows `R8` and `R10_CSI2P`:
```
[DEBUG] [1738506277.805445768] [camera]: 
>> stream formats:
   - R8 (640x400 - 1280x800)
   - R10_CSI2P (640x400 - 1280x800)
```
Where `R10_CSI2P` is indeed some hardware-specific format.

What formats are shown for the `video` and `still` roles?

bobzwik, 2025-02-02:
`role:=still` :

```
>> cameras:
   0: ov9281 (/base/soc/i2c0mux/i2c@1/ov9281@60)
[WARN] [1738514062.174790980] [camera]: no camera selected, using default: "/base/soc/i2c0mux/i2c@1/ov9281@60"
[WARN] [1738514062.175018071] [camera]: set parameter 'camera' to silence this warning
[0:03:45.631258997] [89] DEBUG Camera camera.cpp:1137 streams configuration: (0) 1280x800-YUV420
[DEBUG] [1738514062.176045026] [camera]: default StillCapture stream configuration: "1280x800-YUV420"
[DEBUG] [1738514062.176671077] [camera]: 
>> stream formats:
   - NV21 (64x64 - 1280x800)
   - YUV420 (64x64 - 1280x800)
   - NV12 (64x64 - 1280x800)
   - YVU420 (64x64 - 1280x800)
   - XBGR8888 (64x64 - 1280x800)
   - BGR888 (64x64 - 1280x800)
   - RGB888 (64x64 - 1280x800)
   - XRGB8888 (64x64 - 1280x800)
   - RGB565 (64x64 - 1280x800)
   - YVYU (64x64 - 1280x800)
   - YUYV (64x64 - 1280x800)
   - VYUY (64x64 - 1280x800)
   - UYVY (64x64 - 1280x800)
[INFO] [1738514062.177266702] [camera]: 
>> stream formats:
   - NV21 (64x64 - 1280x800)
   - XBGR8888 (64x64 - 1280x800)
   - BGR888 (64x64 - 1280x800)
   - RGB888 (64x64 - 1280x800)
   - XRGB8888 (64x64 - 1280x800)
   - YUYV (64x64 - 1280x800)
   - UYVY (64x64 - 1280x800)
[WARN] [1738514062.177360257] [camera]: default pixel format (YUV420) not supported
[WARN] [1738514062.177398979] [camera]: no pixel format selected, auto-selecting: "NV21"
```

`role:=video` :
```
>> cameras:
   0: ov9281 (/base/soc/i2c0mux/i2c@1/ov9281@60)
[WARN] [1738514178.300955111] [camera]: no camera selected, using default: "/base/soc/i2c0mux/i2c@1/ov9281@60"
[WARN] [1738514178.301039331] [camera]: set parameter 'camera' to silence this warning
[0:05:41.757442840] [111] DEBUG Camera camera.cpp:1137 streams configuration: (0) 1920x1080-YUV420
[DEBUG] [1738514178.302534103] [camera]: default VideoRecording stream configuration: "1920x1080-YUV420"
[DEBUG] [1738514178.303156314] [camera]: 
>> stream formats:
   - NV21 (64x64 - 1280x800)
   - YUV420 (64x64 - 1280x800)
   - NV12 (64x64 - 1280x800)
   - YVU420 (64x64 - 1280x800)
   - XBGR8888 (64x64 - 1280x800)
   - BGR888 (64x64 - 1280x800)
   - RGB888 (64x64 - 1280x800)
   - XRGB8888 (64x64 - 1280x800)
   - RGB565 (64x64 - 1280x800)
   - YVYU (64x64 - 1280x800)
   - YUYV (64x64 - 1280x800)
   - VYUY (64x64 - 1280x800)
   - UYVY (64x64 - 1280x800)
[INFO] [1738514178.304461237] [camera]: 
>> stream formats:
   - NV21 (64x64 - 1280x800)
   - XBGR8888 (64x64 - 1280x800)
   - BGR888 (64x64 - 1280x800)
   - RGB888 (64x64 - 1280x800)
   - XRGB8888 (64x64 - 1280x800)
   - YUYV (64x64 - 1280x800)
   - UYVY (64x64 - 1280x800)
[WARN] [1738514178.307348355] [camera]: default pixel format (YUV420) not supported
[WARN] [1738514178.307459538] [camera]: no pixel format selected, auto-selecting: "NV21"
```

So, same as `viewfinder`, but defaults to `NV21` instead of `XRGB8888`.

So, is `raw` okay to use for feature tracking for Visual Odometry?

One odd thing I noticed, is if i set the camera_ros node to 30 fps and either viewfinder/still/video, I will get a steady 30 fps (`topic hz /camera/image_raw`) but with `raw`, I get 28.8 fps. Which is the same FPS I get doing `libcamera-hello --viewfinder-mode 640:400:8 --framerate 30` (as seen in my first post). Is this normal?

christianrauch, 2025-02-03:
> So, same as `viewfinder`, but defaults to `NV21` instead of `XRGB8888`.

Interesting that the other roles also do not list monochrome pixel formats.

> So, is `raw` okay to use for feature tracking for Visual Odometry?

Sure. There is nothing you can gain from mapping your grey scale data to colour.

In terms of 8 bit vs. 10 bit, I would have expected that the camera lists the normal "unpacked" `R10` instead of the packed Bayer format `R10_CSI2P` in parallel to `R8`.

It also looks like you are using the official upstream libcamera version. Can you also try the "raspberrypi" fork at https://github.com/raspberrypi/libcamera? But overall, this is out-of-scope for the ROS node, as it simply relies on libcamera to provide the interface to the camera. If you have questions related to support for the OV9281, please refer to the respective projects.

> One odd thing I noticed, is if i set the camera_ros node to 30 fps and either viewfinder/still/video, I will get a steady 30 fps (`topic hz /camera/image_raw`) but with `raw`, I get a steady 28.8 fps. Which is the same FPS I get doing `libcamera-hello --viewfinder-mode 640:400:8 --framerate 30` (as seen in my first post). Is this normal?

I don't know. In the end, it is up to the stream role and the camera driver which configurations can be set. And then there are other configuration parameters that impact the time for a frame. Does the `FrameDurationLimits` change with the different stream role and exposure settings?

bobzwik, 2025-02-04:
> It also looks like you are using the official upstream libcamera version.

Hmm, that odd, why so? 
I'm running ROS in Docker, and my Dockerfile clones the `raspberrypi/libcamera` repo and switches to the `0.4.0` tag. Outside of the container, I've got `libcamera build: v0.4.0+50-83cb8101` installed on my RPi4 (running Pi OS (Bookworm) 64-bit), and `83cb810` seems to be the latest (Jan 23) commit on `raspberrypi/libcamera`.

`FrameDurationLimits` does not set if I do not set it, for all roles. But in good lighting, the FPS defaults to a stable 30 fps for `viewfinder`, `still` and `video`, while 28.7-28.8 fps in `raw` (measured with `ros2 topic hz /camera/image_raw/compressed` ). In bad lighting, the FPS lowers to about 8 since it seems auto exposure is on automatically, and `raw` still produces maybe 0.5 fps slower than the other roles.

But for my application, the FPS isn't a tremendous issue. Worst comes to worst, I can set `FrameDurationLimits` to 48000 to get a perfect 20 Hz.

But I do agree, these issues do not stem from the ROS node. I might eventually bring up this issue with rpi/libcamera.

bobzwik, 2025-02-04:
Well....

I would have love to leave it at that, but there seems to be another pretty big difference between `raw` and the others. 

The exposure is pretty different. Even if I manually set the exposure, `viewfinder` is quite brighter. I imagine this comes from a libcamera or hardware issue, but maybe you have some experience with this? 

```
ros2 run camera_ros camera_node --ros-args -p width:=1280 -p height:=800 -p role:=raw -p FrameDurationLimits:="[50000,50000]" -p AeEnable:=false -p ExposureTime:=32000 -p AnalogueGain:=2.0
````

`raw` (`R8`):

[image]

`viewfinder` (`XRGB8888`):

[image]

And the parameters are practically identical:
`raw` (`R8`):
```
root@rpi4:/ros2_ws# ros2 param dump /camera
/camera:
  ros__parameters:
    AeConstraintMode: 0
    AeEnable: false
    AeExposureMode: 0
    AeFlickerMode: 0
    AeFlickerPeriod: null
    AeMeteringMode: 0
    AnalogueGain: 2.0
    Brightness: 0.0
    Contrast: 1.0
    ExposureTime: 32000
    ExposureValue: 0.0
    FrameDurationLimits:
    - 50000
    - 50000
    HdrMode: 0
    ScalerCrop:
    - 0
    - 0
    - 1066
    - 800
    Sharpness: 1.0
    StatsOutputEnable: false
    camera: null
    format: ''
    height: 800
    jpeg_quality: 95
    qos_overrides:
      /parameter_events:
        publisher:
          depth: 1000
          durability: volatile
          history: keep_last
          reliability: reliable
    role: raw
    use_sim_time: false
    width: 1280
```

`viewfinder` (`XRGB8888`):
```
root@rpi4:/ros2_ws# ros2 param dump /camera
/camera:
  ros__parameters:
    AeConstraintMode: 0
    AeEnable: false
    AeExposureMode: 0
    AeFlickerMode: 0
    AeFlickerPeriod: null
    AeMeteringMode: 0
    AnalogueGain: 2.0
    Brightness: 0.0
    Contrast: 1.0
    ExposureTime: 32000
    ExposureValue: 0.0
    FrameDurationLimits:
    - 50000
    - 50000
    HdrMode: 0
    ScalerCrop:
    - 0
    - 0
    - 1280
    - 800
    Sharpness: 1.0
    StatsOutputEnable: false
    camera: null
    format: ''
    height: 800
    jpeg_quality: 95
    qos_overrides:
      /parameter_events:
        publisher:
          depth: 1000
          durability: volatile
          history: keep_last
          reliability: reliable
    role: viewfinder
    use_sim_time: false
    width: 1280
```

(For some reason, `raw` sets `ScaleCrop` to [0 0 1066 800], but it doesn't affect the picture, as you can see in the comparison between both images. Manually setting to [0 0 1280 800] does not affect the picture either.)

I had to pack up for the evening, but I'll test with libcamera-still this week to test if it is purely a difference between `R8` and `R10`.

bobzwik, 2025-02-06:
> I would be really curious to see if this is indeed supported by libcamera on the Raspberry Pi.

Setting 640x400, 180 fps (`role:=raw -p format:=R8`) uses 27-28% of a core.
`ros2 topic hz /camera/image_raw` yields 162 Hz (and increases CPU usage to 41% of a core).
`ros2 topic hz /camera/image_raw/compressed` yields  135 Hz (and increases CPU usage to 56% of a core).
So I wouldn't be surprised if 180 is actually being achieved. `ros2 topic hz` is still a python node, one could write an equivalent c++ node and maybe be able to measure fps more accurately.

If I let the default role and format (`viewfinder` and `XRGB8888`), `ros2 topic hz /camera/image_raw/compressed` does show 180 fps, but the node uses 85% of a core, without compression and when not running `ros2 topic hz` (and uses 180% when running `ros2 topic hz /camera/image_raw/compressed`).

Setting 1280x800, 120 fps uses (`role:=raw -p format:=R8`) uses 34-38% of a core.
`ros2 topic hz /camera/image_raw` yields 67Hz (and increases CPU usage to 50% of a core). 
`ros2 topic hz /camera/image_raw/compressed` yields  57 Hz (and increases CPU usage to 73% of a core).

This time, the default role and format do not improve the fps measurement. `camera_node` uses 240%, without running `ros2 topic hz`.

christianrauch, 2025-02-06:
The node does "lazy" publishing and on-demand de-/compression. If you set a raw uncompressed format, such as `R8`, and subscribe to the `image_raw/compressed` topic, only then will the raw image data be compressed. And vice versa: If you chose `MJPEG` as format and subscribe to `image_raw`, the data will be decompressed. This takes away resources from the CPU and the node will probably run slower. You can see this from your results. Hence, to get the optimal performance, only subscribe to `image_raw` when using any of the raw uncompressed formats.

I tried to implement the reading and publishing of image data as efficient as possible, e.g. by copying the data directly into the messages instead of doing a costly conversion. Without proper profiling, it is hard to say how much the image data processing and publishing in this node and how much the rest of the ROS framework contribute to the reduced performance here.

bobzwik, 2025-02-12:
> The exposure is pretty different. Even if I manually set the exposure, `viewfinder` is quite brighter. I imagine this comes from a libcamera or hardware issue, but maybe you have some experience with this?

So, I've come to the conclusion that the reencoding to `XRGB8888` or `YUYV` when using `viewfinder` applies some sort of post-processing that increases the brightness and contrast of the raw image. With `raw`, this obviously doesn't occur.

I had a little fun and modified `rpicam-apps` ([rpicam-ros2](https://github.com/bobzwik/rpicam-apps/blob/main/apps/rpicam_ros2.cpp) based off `rpicam-raw` and [rpicam-ros2-vf](https://github.com/bobzwik/rpicam-apps/blob/main/apps/rpicam_ros2_vf.cpp) based on `rpicam-hello`) to essentially publish the image to ROS2 topics, using `image_transport`. I call the apps like I would call a normal rpicam-app:
```
rpicam-ros2 --mode 640:400:8 --framerate 20
``` 
```
rpicam-ros2-vf --viewfinder-mode 640:400:8 --framerate 20
```
and the footage is viewable in `rviz2` or `rqt_image_view` and automatically compressed with `image_transport_plugins`. With very similar CPU usage as `camera_ros`.

I obtain the same images as I would with `camera_ros`:
- `rpicam-ros2` produces the same image as `camera_ros` with the `raw` role
- `rpicam-ros2-vf` produces the same image as `camera_ros` with the `viewfinder` role 

So I believe the postprocessing occurs somewhere in `libcamera`, depending on the selected stream mode/role.

So, I am content with being able to use `camera_ros` with the `raw` role to obtain my unprocessed 8-bit footage. 

Thanks!

christianrauch, 2025-02-12:
The ROS node essentially copies the data coming of the image planes provided by libcamera without any further processing. I don't have the insight to what the IPAs or pipelines are doing. I would assume that you get the same results when exporting the images with the libcamera or [rpicam](https://github.com/raspberrypi/rpicam-apps) examples.

christianrauch, 2025-02-13:
Alright. I think everything is then clear now.

I added the `R16` / `MONO16` format recently (#84), so the node will publish 16 bit greyscale images if a camera supports this pixel format. But for everything else, the node relies on libcamera. Especially if you want to publish the 10 bit grey scale images, you would need to add the encoding string to [`image_encodings.hpp`](https://github.com/ros2/common_interfaces/blob/rolling/sensor_msgs/include/sensor_msgs/image_encodings.hpp) first and then make sure that you are indeed using the `R10` format and not the raw `R10_CSI2P`.

closed 2025-02-13 by christianrauch