christianrauch/camera_ros #85: Compiling issue on RPi5/BookWorm

vipermn, 2025-02-08:
I am using a RPI 5 with 8Gb.  I have the latest and updated Bookworm installed.
I downloaded ROS2 Jazzy source and compiled.  ROS2 is working as expected.

```
user@pi5-pios-ros2:~/camera_ws $ cam -l
[26:54:20.263672267] [6611]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+50-83cb8101
[26:54:20.274844848] [6614]  INFO RPI pisp.cpp:720 libpisp version v1.1.0 e7974a156008 27-01-2025 (21:50:51)
[26:54:20.290681967] [6614]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx296@1a to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
Available cameras:
1: 'imx296' (/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a)

```
The imx296 is the Pi Global Shutter camera.

Then I downloaded the Camera_Ros source (main) and attempted to compile.  I am getting the following:

```
user@pi5-pios-ros2:~/camera_ws $ rosdep install -y --from-paths src --ignore-src --rosdistro jazzy --skip-keys=libcamera
#All required rosdeps installed successfully
user@pi5-pios-ros2:~/camera_ws $ colcon build --event-handlers=console_direct+
Starting >>> camera_ros
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1")
-- Found ament_cmake: 2.5.3 (/home/tom/ros2_jazzy/install/ament_cmake/share/ament_cmake/cmake)
-- Found Python3: /usr/bin/python3 (found version "3.11.2") found components: Interpreter
-- Found rclcpp: 28.1.6 (/home/tom/ros2_jazzy/install/rclcpp/share/rclcpp/cmake)
-- Found rosidl_generator_c: 4.6.5 (/home/tom/ros2_jazzy/install/rosidl_generator_c/share/rosidl_generator_c/cmake)
-- Found rosidl_generator_cpp: 4.6.5 (/home/tom/ros2_jazzy/install/rosidl_generator_cpp/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_introspection_c;rosidl_typesupport_fastrtps_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_introspection_cpp;rosidl_typesupport_fastrtps_cpp
-- Found rmw_implementation_cmake: 7.3.1 (/home/tom/ros2_jazzy/install/rmw_implementation_cmake/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 8.4.1 (/home/tom/ros2_jazzy/install/rmw_fastrtps_cpp/share/rmw_fastrtps_cpp/cmake)
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found version "3.0.15")
-- Found FastRTPS: /home/tom/ros2_jazzy/install/fastrtps/include (Required is at least version "2.13")
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rclcpp_components: 28.1.6 (/home/tom/ros2_jazzy/install/rclcpp_components/share/rclcpp_components/cmake)
-- Found sensor_msgs: 5.3.5 (/home/tom/ros2_jazzy/install/sensor_msgs/share/sensor_msgs/cmake)
-- Found camera_info_manager: 5.1.5 (/home/tom/ros2_jazzy/install/camera_info_manager/share/camera_info_manager/cmake)
-- Setting yaml-cpp_DIR to: '/home/tom/ros2_jazzy/install/yaml_cpp_vendor/share/yaml_cpp_vendor/cmake/../../../opt/yaml_cpp_vendor/share/cmake/yaml-cpp'
-- Checking for module 'libcamera>=0.1'
--   Found libcamera, version 0.4.0
-- Found ament_lint_auto: 0.17.1 (/home/tom/ros2_jazzy/install/ament_lint_auto/share/ament_lint_auto/cmake)
-- Added test 'clang_format' to check C / C++ code style
-- Added test 'cppcheck' to perform static code analysis on C / C++ code
-- Configured cppcheck include dirs:
-- Configured cppcheck exclude dirs and/or files:
-- Added test 'flake8' to check Python code syntax and style conventions
-- Configured 'flake8' exclude dirs and/or files:
-- Added test 'lint_cmake' to check CMake code style
-- Added test 'mypy' to statically type check Python code.
-- Added test 'pep257' to check Python code against some of the docstring style conventions in PEP 257
-- Added test 'pyflakes' to perform static code analysis on Python code
-- Added test 'xmllint' to check XML markup files
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tom/camera_ws/build/camera_ros
[ 33%] Building CXX object CMakeFiles/utils.dir/src/format_mapping.cpp.o
[ 33%] Building CXX object CMakeFiles/utils.dir/src/cv_to_pv.cpp.o
[ 33%] Building CXX object CMakeFiles/utils.dir/src/clamp.cpp.o
[ 33%] Building CXX object CMakeFiles/camera_node.dir/rclcpp_components/node_main_camera_node.cpp.o
[ 41%] Building CXX object CMakeFiles/utils.dir/src/parameter_conflict_check.cpp.o
[ 50%] Building CXX object CMakeFiles/utils.dir/src/pretty_print.cpp.o
[ 58%] Building CXX object CMakeFiles/utils.dir/src/pv_to_cv.cpp.o
[ 66%] Building CXX object CMakeFiles/utils.dir/src/types.cpp.o
[ 75%] Building CXX object CMakeFiles/utils.dir/src/type_extent.cpp.o
[ 75%] Built target utils
[ 83%] Building CXX object CMakeFiles/camera_component.dir/src/CameraNode.cpp.o
[ 91%] Linking CXX executable camera_node
[ 91%] Built target camera_node
In file included from /home/tom/camera_ws/src/camera_ros/src/CameraNode.cpp:55:
/home/tom/ros2_jazzy/install/sensor_msgs/include/sensor_msgs/sensor_msgs/image_encodings.hpp:44:12: error: conflicting declaration ‘const char sensor_msgs::image_encodings::RGB8 []’
   44 | const char RGB8[] = "rgb8";
      |            ^~~~
In file included from /usr/include/cv_bridge/cv_bridge.h:41,
                 from /home/tom/camera_ws/src/camera_ros/src/CameraNode.cpp:19:
/usr/include/sensor_msgs/image_encodings.h:47:23: note: previous declaration as ‘const std::string sensor_msgs::image_encodings::RGB8’
   47 |     const std::string RGB8 = "rgb8";
      |                       ^~~~

```
This is the first of many similar errors. Then the ending.

```
gmake[2]: *** [CMakeFiles/camera_component.dir/build.make:76: CMakeFiles/camera_component.dir/src/CameraNode.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:167: CMakeFiles/camera_component.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< camera_ros [22.3s, exited with code 2]

Summary: 0 packages finished [22.5s]
  1 package failed: camera_ros
  1 package had stderr output: camera_ros

```

OpenCV version is:

```
user@pi5-pios-ros2:~/camera_ws $ opencv_version
4.6.0

```
Any thoughts on the next step to pursue?  It looks to me like this is a version mismatch somewhere along the line, but I don't know what to do next.

Thank you!

christianrauch, 2025-02-08:
Well, it looks like the `sensor_msgs` headers you installed (`/home/tom/ros2_jazzy/install/sensor_msgs/include/sensor_msgs/sensor_msgs/image_encodings.hpp`) are interfering with headers from system packages `sensor_msgs` (`/usr/include/sensor_msgs/image_encodings.h`). According to https://packages.debian.org/bookworm/arm64/libsensor-msgs-dev/filelist, you installed the [`libsensor-msgs-dev`](https://packages.debian.org/bookworm/libsensor-msgs-dev) package.

If you really want to compile ROS 2 yourself, don't mix it up with the system packages. Or just use Ubuntu with the official ROS 2 packages on the Raspberry Pi.

closed 2025-02-12 by vipermn

0x62, 2025-02-22:
Sharing the below as I had the same problem - I have to use Bookworm as I'm also trying to use this with the Raspberry Pi AI Hat+, which doesn't have Ubuntu drivers.

I was able to get it to build by adding [vision_opencv](https://github.com/ros-perception/vision_opencv/tree/rolling) to the ROS workspace, rebuilding ROS, then sourcing `local_setup.bash` and skipping the `rosdep install` before building `camera_ros`.

christianrauch, 2025-02-22:
Since the node depends on `cv_bridge` from `vision_opencv`, and this dependency is listed in the [`package.xml`](https://github.com/christianrauch/camera_ros/blob/0.3.0/package.xml#L17), you can still use rosdep to figure out which package should be installed/build.

But since there are no official binary packages distributed for the "Raspberry Pi OS", you have to build them from source. Do not mix the sources with the binary packages. To avoid this, I simply recommend removing any Debian package that is produced from the ROS sources.