christianrauch/camera_ros #91: Array issue on RPI5/Ubuntu 24.01/Jazzy

vipermn, 2025-02-12:
I have a RPI Global Shutter camera (IMX296).
The Pi is a Pi 5 with 8Gb.

I am running:
```
tom@pi5-u24:~/camera_ws$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04.1 LTS
Release:        24.04
Codename:       noble

```

ROS2 Jazzy successfully installed from binaries according to ROS2 documentation.

Followed your steps for compiling libcamera and camera_ros using the "raspberrypi" fork.  Compiling was successful.  I don't know if it makes a difference but I did not compile with gstreamer since that was optional.

```
tom@pi5-u24:~/camera_ws$ ros2 run camera_ros camera_node
[27:05:38.119467852] [17174]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+51-ca36c77f
[27:05:38.139082673] [17189]  INFO RPI pisp.cpp:720 libpisp version v1.0.7 28196ed6edcf 11-02-2025 (20:25:46)
[27:05:38.152628019] [17189]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx296@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[INFO] [1739327428.807878559] [camera]:
>> cameras:
   0: imx296 (/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a)
[WARN] [1739327428.808025504] [camera]: no camera selected, using default: "/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a"
[WARN] [1739327428.808048949] [camera]: set parameter 'camera' to silence this warning
[27:05:38.153298800] [17174]  WARN V4L2 v4l2_pixelformat.cpp:346 Unsupported V4L2 pixel format RPBP
[INFO] [1739327428.808642026] [camera]:
>> stream formats:
   - R8 (32x32 - 1456x1088)
   - R16 (32x32 - 1456x1088)
   - MONO_PISP_COMP1 (32x32 - 1456x1088)
   - NV21 (32x32 - 1456x1088)
   - SBGGR8 (32x32 - 1456x1088)
   - SBGGR16 (32x32 - 1456x1088)
   - BGGR_PISP_COMP1 (32x32 - 1456x1088)
   - XBGR8888 (32x32 - 1456x1088)
   - BGR888 (32x32 - 1456x1088)
   - RGB888 (32x32 - 1456x1088)
   - XRGB8888 (32x32 - 1456x1088)
   - SGBRG16 (32x32 - 1456x1088)
   - GBRG_PISP_COMP1 (32x32 - 1456x1088)
   - SGRBG16 (32x32 - 1456x1088)
   - GRBG_PISP_COMP1 (32x32 - 1456x1088)
   - SRGGB16 (32x32 - 1456x1088)
   - RGGB_PISP_COMP1 (32x32 - 1456x1088)
   - BGR161616 (32x32 - 1456x1088)
   - RGB161616 (32x32 - 1456x1088)
   - SRGGB8 (32x32 - 1456x1088)
   - SGRBG8 (32x32 - 1456x1088)
   - SGBRG8 (32x32 - 1456x1088)
   - YUYV (32x32 - 1456x1088)
   - UYVY (32x32 - 1456x1088)
[WARN] [1739327428.808725026] [camera]: no pixel format selected, auto-selecting: "XRGB8888"
[WARN] [1739327428.808743267] [camera]: set parameter 'format' to silence this warning
[INFO] [1739327428.808781823] [camera]:
>> XRGB8888 format sizes:
   - 160x120
   - 240x160
   - 320x240
   - 400x240
   - 480x320
   - 640x360
   - 640x480
   - 720x480
   - 768x480
   - 854x480
   - 720x576
   - 800x600
   - 960x540
   - 1024x576
   - 960x640
   - 1024x600
   - 1024x768
   - 1280x720
   - 1152x864
   - 1280x800
   - 1360x768
   - 1366x768
   - 1440x900
   - 1280x1024
   - 1280x1080
   - 1400x1050
[WARN] [1739327428.808846619] [camera]: no dimensions selected, auto-selecting: "800x600"
[WARN] [1739327428.808862675] [camera]: set parameter 'width' or 'height' to silence this warning
[WARN] [1739327428.808994824] [camera]: stream configuration adjusted from "800x600-XRGB8888" to "800x600-XRGB8888"
[27:05:38.153940470] [17174]  INFO Camera camera.cpp:1202 configuring streams: (0) 800x600-XRGB8888
[27:05:38.154062971] [17189]  INFO RPI pisp.cpp:1484 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx296@1a - Selected sensor format: 1456x1088-SBGGR10_1X10 - Selected CFE format: 1456x1088-PC1B
[INFO] [1739327428.809545641] [camera]: camera "/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a" configured with 800x600-XRGB8888 stream
[WARN] [1739327428.810040903] [camera]: unknown control: SyncFrames (20014)
[WARN] [1739327428.810284312] [camera]: unknown control: SyncMode (20011)
[WARN] [1739327428.810541165] [camera]: unknown control: CnnEnableInputTensor (20007)
[WARN] [1739327428.810797166] [camera]: unknown control: NoiseReductionMode (10002)
camera_node: ../../src/libcamera/include/libcamera/controls.h:204: T libcamera::ControlValue::get() const [with T = libcamera::Span<const long int>; typename std::enable_if<(libcamera::details::is_span<U>::value || std::is_same<std::__cxx11::basic_string<char>, typename std::remove_cv< <template-parameter-1-1> >::type>::value), std::nullptr_t>::type <anonymous> = nullptr]: Assertion `isArray_' failed.
[ros2run]: Aborted
```

```
tom@pi5-u24:~/camera_ws$ cam -l
[27:13:14.160895787] [17236]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+51-ca36c77f
[27:13:14.181145831] [17239]  INFO RPI pisp.cpp:720 libpisp version v1.0.7 28196ed6edcf 11-02-2025 (20:25:46)
[27:13:14.193825355] [17239]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx296@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
Available cameras:
1: 'imx296' (/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a)

```

Your thoughts on the next steps?

GeoGk, 2025-02-12:
I run into the same issue yesterday. I built the camera workspace in debug mode as suggested in the README and tracked down the issue in the libcamera package. 
In `controls.h` in the following get function
```
template<typename T, std::enable_if_t<details::is_span<T>::value ||
					      std::is_same<std::string, std::remove_cv_t<T>>::value,
					      std::nullptr_t> = nullptr>
#else
	template<typename T>
#endif
	T get() const
	{
		assert(type_ == details::control_type<std::remove_cv_t<T>>::value);
		assert(isArray_);

		using V = typename T::value_type;
		const V *value = reinterpret_cast<const V *>(data().data());
		return T{ value, numElements_ };
	}
```
the `assert(isArray_)` fails when called by `void IpaBase::applyControls(const ControlList &controls)`  when it reaches the case for `FRAME_DURATION_LIMITS` below (present in `ipa_base.cpp`)
```
case controls::FRAME_DURATION_LIMITS: {
			auto frameDurations = ctrl.second.get<Span<const int64_t>>();
			applyFrameDurations(frameDurations[0] * 1.0us, frameDurations[1] * 1.0us);
			break;
		}
```
Apparently the get function expects to be able to return an array of two `int64_t`s  however the key of the map (from where the values are retrieved) holds a single `int64_t`. I altered the code of the `case controls::FRAME_DURATION_LIMITS` as you can see below in order to use another `T get()` function (that is, the one that handles 'non-array-like' data) and get around the assertion issue.
```
case controls::FRAME_DURATION_LIMITS: {
			auto frameDurations = ctrl.second.get<const int64_t>();
			applyFrameDurations(frameDurations * 1.0us, frameDurations * 1.0us);
			break;
		}
```
After this change the node runs normally, however I haven't looked deeper into why this happens and I assume that my 'fix' is not necessarily 'sane' as far as the configuration of the camera is concerned.
I mainly care to have a stream of images in ROS2 at the moment and I do not care much about their quality, however if I find something more concrete I will come back with an update.

christianrauch, 2025-02-12:
Does https://github.com/christianrauch/camera_ros/pull/89 fix your issue? I ran into issues with `FRAME_DURATION_LIMITS` in the @raspberrypi fork and this fixed it.

vipermn, 2025-02-12:
#89 does fix my issue!  I have pulled the latest, recompiled and now camera_ros is working.

Thank you for the quick fix and also for developing and maintaining camera_ros!

closed 2025-02-12 by vipermn