christianrauch/camera_ros #94: Raspberry Pi 5 + Raspberry Pi Camera Module, ROS2 Docker arm64v8, camera_ros - working example

se1exin, 2025-02-12:
Hi there,

I recently bought a RPi 5 and wanted to use some of my old v1 Raspberry Pi Camera Modules with ROS via `camera_ros` - but I like to do everything ROS related on the raspberry pi in Docker containers and ended up having a terrible time getting the cameras to work inside docker.

Long story short, I finally figured it out and have the cameras detected by `camera_ros` inside my docker container, running on Rasbian Bookworm 64bit Lite.

I have created a GitHub repo with a Dockerfile and working example that anyone with the same hardware should be able to reproduce: https://github.com/se1exin/camera_ros-in-docker-rpi5

I just thought I'd share it for anyone that comes searching here trying to achieve a similar setup that I am using 😄  If the maintainers of `camera_ros` want to link to it in the readme I'm fine with that as well 😄

Note my Dockerfile works for the following cameras, but I don't see why it shouldn't work for all Raspberry Pi Camera Modules:

-  Raspberry Pi Cam rev1.3 (ov5647)
-  Raspberry Pi NoIR Cam (ov5647)

christianrauch, 2025-02-12:
> I like to do everything ROS related on the raspberry pi in Docker containers and ended up having a terrible time getting the cameras to work inside docker.

Was there a particular issue with the libcamera code or was this just related to Docker configuration?

> I have created a GitHub repo with a Dockerfile and working example that anyone with the same hardware should be able to reproduce: https://github.com/se1exin/camera_ros-in-docker-rpi5

Nice. The `Dockerfile` looks straight forward. Does this work with the `CSI` connection out of the box?

Unfortunately, the support for newer Raspberry Pi Camera module hasn't been merged yet into the upstream libcamera project. Once this is done, the support will be officially provided by the bloomed packages in the ROS repos and then you only have to install the Debian packages in the Docker image.

se1exin, 2025-02-12:
> Was there a particular issue with the libcamera code or was this just related to Docker configuration?

I think it was related to libcamera, when trying to run `camera_ros` in a custom container based on `arm64v8/ros:jazzy`, using the deb packages, I kept getting the error `what(): no cameras available`.

To isolate the source of the issue I tested accessing the cameras directly on the host (e.g. via libcamera-still), and accessing the cameras in a third party docker container (https://github.com/hyzhak/pi-camera-in-docker) with *success*. Which at least meant that my docker hardware config was correct.

With this info I concluded that the issue was most likely with the version of `libcamera` that `camera_ros` was pulling in, but also potentially because of something unknown happening in the docker container I was basing off  (`arm64v8/ros:jazzy`). I did see in the following in the readme for this project, which kind summarizes my experience above:

>> [!NOTE]
>> This also installs the package [`libcamera`](https://index.ros.org/r/libcamera/) as dependency. This is the bloomed version of the official upstream repo at https://git.libcamera.org/libcamera/libcamera.git and may not contain full support for all Raspberry Pi camera modules. If you need full camera module support on Raspberry Pi, you have to build the "raspberrypi" fork from https://github.com/raspberrypi/libcamera manually.

...but to be honest the build instructions in the readme are not 100% correct for building the raspberrypi fork the libcamera. It took me several attempts to get the dependencies and build setup right, which resulted in my example Dockerfile. The difference in build instructions may be because of my base container though (arm64v8/ros:jazzy).

Anyhow, that was pretty long winded but I hope it answers your question and gives some more context to my solution.

> Does this work with the CSI connection out of the box?

Yes it does - both cameras tested were connected using a CSI ribbon cable, with no modification to the raspberry pi `config.txt` file 😄

christianrauch, 2025-02-13:
Alright. I am going to close the issue, but it will stay "searchable" so others interested in a Docker image can find your repo.

closed 2025-02-13 by christianrauch