christianrauch/camera_ros #97: No supported formats with raw role for RPI Camera Module 3

0x62, 2025-02-25:
When I select role `raw` with the Camera Module 3 (IMX708), I get an error that there are no supported formats. I built `camera_ros` with the `libcamera` fork from Raspberry Pi.

I've tried a few different format options with role `video`, the following is the typical performance I've managed to get - as measured by `ros2 topic hz` on `/camera/image_raw` on the local node:

* 420p ~10 hz
* 720p ~8 hz
* 1080p ~6 hz

Some formats give as low as 1-3 hz. I've seen as high as 12hz briefly, but I've never managed to get close to 30FPS even with low resolutions selected in `video` mode.

ROS Distro: Jazzy
OS: Bookworm
Hardware: Raspberry Pi 5 (8GB)

christianrauch, 2025-02-25:
> When I select role `raw` with the Camera Module 3 (IMX708), I get an error that there are no supported formats. I built `camera_ros` with the `libcamera` fork from Raspberry Pi.

Which commands did you run? The node should select a reasonable format by default. There should only be very rare cases where the camera does not provide a pixel format which is supported by the ROS message.

I also have a `imx708` and with the defaults (just a `os2 run camera_ros camera_node`) I get:
```
camera "/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a" configured with 800x600-XRGB8888 stream
```
and with `ros2 topic hz /camera/image_raw`:
```
average rate: 14.918
```

There are many factors that impact the effective frame rate.

First, there are a couple of camera settings that decide how long it takes to take a frame. E.g. a high exposure time, or auto exposure in low-light conditions, will limit how many frames can be generated per second. Also see https://github.com/christianrauch/camera_ros?tab=readme-ov-file#dynamic-frame-configuration-controls for constraining the min/max frame duration.

Secondly, if you transmit raw images over a network, you will be limited by the bandwidth. You can check with `ros2 topic bw /camera/image_raw`. If this maxes out at your network bandwidth, you have to reduce the resolution or pixel format, or use compressed transport.

And on top of this all, the `ros2 topic` tools are actually not optimal to measure all this since this includes deserialisation and the default RMW is known to not perform well at this.

christianrauch, 2025-02-25:
I saw that with the most recent libcamera version from @raspberrypi, and the new handling of scalar default values for span controls, the `FrameDurationLimits` control was effectively disabled. I fixed this by ignoring the default value instead of the entire control, and the `FrameDurationLimits` control should be available again.

The `FrameDurationLimits` lets you set lower and upper bounds for the frame duration. Setting both to the same value essentially fixes the frame rate. For example, if you set the frame duration bounds to 20 ms:
```sh
ros2 run camera_ros camera_node --ros-args -p FrameDurationLimits:="[20000, 20000]"
```
you will get a frame rate of 50 Hz (1 / 20 ms):
```
$ ros2 topic hz /camera/image_raw
average rate: 50.108
	min: 0.017s max: 0.023s std dev: 0.00180s window: 52
average rate: 50.112
	min: 0.017s max: 0.023s std dev: 0.00174s window: 103
```

Feel free to also play around with the other exposure and duration controls to see how they affect the timing.

0x62, 2025-02-26:
Thanks for the pointers. I upgraded to the latest version, and the FrameDurationLimits is now working and I'm seeing much better frame rates.

Regarding the original title of the issue, running with role `raw` gives me:

```
ben@roarm:~ $ ros2 run camera_ros camera_node --ros-args -p role:=raw
[0:24:37.325181559] [6041]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
[0:24:37.333388839] [6056]  INFO RPI pisp.cpp:720 libpisp version v1.0.7 28196ed6edcf 22-02-2025 (01:26:46)
[0:24:37.345831471] [6056]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media1 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[INFO] [1740605222.193733750] [camera]:
>> cameras:
   0: imx708 (/base/axi/pcie@120000/rp1/i2c@80000/imx708@1a)
[WARN] [1740605222.194037082] [camera]: no camera selected, using default: "/base/axi/pcie@120000/rp1/i2c@80000/imx708@1a"
[WARN] [1740605222.194113544] [camera]: set parameter 'camera' to silence this warning
[INFO] [1740605222.194407395] [camera]:
>> stream formats:
terminate called after throwing an instance of 'std::runtime_error'
  what():  camera does not provide any of the supported pixel formats
[ros2run]: Aborted
```

christianrauch, 2025-02-26:
Running this with debug info (`--log-level camera:=debug`) will also show you the pixel formats that are not supported by the ROS image message, e.g. for the `imx708_wide`:
```
>> stream formats:
   - SRGGB10_CSI2P (1536x864 - 4608x2592)
```

But `SRGGB10_CSI2P` is not a supported `encoding` for the `Image` message.

closed 2025-02-27 by 0x62