christianrauch/camera_ros #102: Calibrating two cameras with camera ros and image pipeline camera calibration

maiabriggs, 2025-03-10:
Hi there! 

Hope you are well :)

I see that recently you added a parameter for setting the camera info URL. However, I've been having some trouble loading the calibration file after calibrating, as well as setting it using the set_camera_info service whilst running the camera_calibrator from the Image Pipeline package. I was wondering if this might be due to my calibrating the two cameras as a stereo camera, or if I am just doing something wrong when launching the nodes. 

Initially, I was launching the camera using my stereo_camera.launch.py file and then using ros2 run camera_calibration cameracalibrator, with all the relevant parameters. I then tried setting the camera info parameter in the launch file, but this didn't seem to work. Now I am trying to run:

ros2 run camera_ros camera_node --ros-args -p camera_info_url:=/home/maia/camera_calibration_info/left.yaml

Which does attempt to load a camera calibration file, but fails to load it:

[INFO] [1741607347.188546284] [camera]: camera calibration URL: /home/maia/camera_calibration_info/left.yaml
[ERROR] [1741607347.188648192] [camera]: Invalid camera calibration URL: /home/maia/camera_calibration_info/left.yaml
[WARN] [1741607347.188689656] [camera]: failed to load camera calibration info from provided URL, using default URL
[INFO] [1741607347.188722915] [camera]: using default calibration URL

Do you have any idea what I am doing wrong?

Thank you so much for your help!
Kind regards,
Maia :)

christianrauch, 2025-03-10:
Well, there are multiple ways to provide the camera info. Once of them is to set it via the ROS service, which will also write it to the default location in `~/.ros/camera_info/`, or you can now manually provide the URL to the calibration file. If you get an error, then this usually means the file does not exist or is not in the expected format. But it is hard to say without more detailed information. Before trying to debug this in a more complex setup with multiple cameras, you should probably reduce the complexity in the setup and try this with valid fake data on a single camera.

Did you check the [CameraInfoManager documentation](https://docs.ros.org/en/jazzy/p/camera_info_manager/generated/classcamera__info__manager_1_1CameraInfoManager.html) or google the issue? The documentation lists the valid URL formats, and using file paths without the `file://` does not appear to be a valid value. If you run the node without setting the calibration, you probably see something like `camera calibration URL: file:///home/[...]`. I am quite sure that you have to provide a URL that points to a file instead of just the local file path (`/home/[...]`).

@samyarsadat implemented this in https://github.com/christianrauch/camera_ros/pull/90. Do you see what is wrong here? Can you document how the new parameter is to be used with a PR if you have time?

samyarsadat, 2025-03-12:
> [@samyarsadat](https://github.com/samyarsadat) implemented this in [#90](https://github.com/christianrauch/camera_ros/pull/90). Do you see what is wrong here? Can you document how the new parameter is to be used with a PR if you have time?

I believe, as you mentioned, the issue is that a local file path is being provided instead of a URL. I've tested this before myself, and indeed, normal file paths do not work. Setting `camera_info_url` to `file:///home/maia/camera_calibration_info/left.yaml` should fix the issue, assuming the camera calibration file is valid.

I've been a little busy these past few weeks, but I will create a PR to add some usage documentation for the parameter as soon as possible.

julianhfmn, 2025-03-14:
Even when giving a file url which is valid, I have a similar issue: 

`julian@aiina-pc:~/aiina$ ros2 run camera_ros camera_node --ros-args -p camera_info_url:=file:///home/julian/.ros/camera_info/test.yaml
[8:03:48.664358117] [51103]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0
[INFO] [1741946715.181685034] [camera]: 
>> cameras:
   0: Arducam USB Camera: Arducam USB (\_SB_.PCI0.GP19.XHC2.RHUB.PRT1-1:1.0-0c45:6366)
[WARN] [1741946715.181757251] [camera]: no camera selected, using default: "\_SB_.PCI0.GP19.XHC2.RHUB.PRT1-1:1.0-0c45:6366"
[WARN] [1741946715.181768152] [camera]: set parameter 'camera' to silence this warning
[INFO] [1741946715.199840982] [camera]: 
>> stream formats:
   - MJPEG (176x144 - 1920x1080)
   - YUYV (176x144 - 1920x1080)
[WARN] [1741946715.199862824] [camera]: no pixel format selected, auto-selecting: "MJPEG"
[WARN] [1741946715.199867783] [camera]: set parameter 'format' to silence this warning
[INFO] [1741946715.199875428] [camera]: 
>> MJPEG format sizes:
   - 176x144
   - 288x160
   - 320x180
   - 320x240
   - 352x288
   - 432x240
   - 640x320
   - 640x360
   - 640x480
   - 848x480
   - 800x600
   - 1024x576
   - 960x720
   - 1280x720
   - 1920x1080
[WARN] [1741946715.199888973] [camera]: no dimensions selected, auto-selecting: "1920x1080"
[WARN] [1741946715.199893432] [camera]: set parameter 'width' or 'height' to silence this warning
[8:03:48.864650769] [51103]  INFO Camera camera.cpp:1202 configuring streams: (0) 1920x1080-MJPEG
[INFO] [1741946715.253810289] [camera]: camera "\_SB_.PCI0.GP19.XHC2.RHUB.PRT1-1:1.0-0c45:6366" configured with 1920x1080-MJPEG stream
[INFO] [1741946716.044899170] [camera]: using default calibration URL
[INFO] [1741946716.044941711] [camera]: camera calibration URL: file:///home/julian/.ros/camera_info/ArducamUSBCamera_ArducamUSB___SB__PCI0_GP19_XHC2_RHUB_PRT1_1_1_0_0c45_6366_1920x1080.yaml
^C[INFO] [1741946718.029502078] [rclcpp]: signal_handler(signum=2)
`

My solution was when running `ros2 run camera_ros camera_node` to use the set_camera_info rosservice, first parsing the yaml with my config.

The file url is valid, but is ignored.

Furthermore, I have the issue that I cannot set the frame_id: 
`ros2 run camera_ros camera_node --ros-args -p frame_id:=camera_frame`

Does not change the standard frame_id, "camera" which is set in [https://github.com/christianrauch/camera_ros/blob/d54147c18bd433d83ee0b884e4297ba38a06a513/src/CameraNode.cpp#L541](CameraNode.cpp)

christianrauch, 2025-03-14:
> The file url is valid, but is ignored.

Can you attach the file here?

> Furthermore, I have the issue that I cannot set the frame_id: `ros2 run camera_ros camera_node --ros-args -p frame_id:=camera_frame`
> 
> This does not change the standard frame_id, "camera" which is set in [https://github.com/christianrauch/camera_ros/blob/d54147c18bd433d83ee0b884e4297ba38a06a513/src/CameraNode.cpp#L541](CameraNode.cpp)

Well, that parameter does not exist. The `frame_id` in the header is currently hard coded:
https://github.com/christianrauch/camera_ros/blob/d54147c18bd433d83ee0b884e4297ba38a06a513/src/CameraNode.cpp#L541

maiabriggs, 2025-03-14:
Sorry for the late response! Haven't had the chance to test this until today, adding file:// to the start of my path fixed the problem!

Thank you so much for all your help :)

julianhfmn, 2025-03-14:
Here is the content of the file: 

```
image_width: 1920
image_height: 1080
camera_name: ArducamUSBCamera_ArducamUSB___SB__PCI0_GP19_XHC2_RHUB_PRT1_1_1_0_0c45_6366_1920x1080
camera_matrix:
  rows: 3
  cols: 3
  data: [715.62855, 0, 960, 0, 718.19196, 540, 0, 0, 1]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.37254850636327208, 0.16932111307918721, -0.0023668421974224027, -0.0016968691894848353, -0.043420118688005153]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1, 0, 0, 0, 1, 0, 0, 0, 1]
projection_matrix:
  rows: 3
  cols: 4
  data: [715.62855, 0, 960, 0, 0, 718.19196, 540, 0, 0, 0, 1, 0]
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false
```

It is mainly the same as the standard url. 

For the frame_id: Is there another way to set it? If not, I assume I would need to republish an adjusted topic with the correct frame?

samyarsadat, 2025-03-14:
> The file url is valid, but is ignored.

Are you sure that your `camera_ros` package is up-to-date? Based on the log output you've provided, it seems to me that you're using a version prior to the addition of the `camera_info_url` parameter.

christianrauch, 2025-03-15:
> Here is the content of the file:

Passing that file to `camera_info_url`:
```sh
ros2 run camera_ros camera_node --ros-args -p camera_info_url:=file:///home/[...]/calib.yaml
```
works for me. Passing a wrong URL or an invalid file should show a warning `failed to load camera calibration info from provided URL, using default URL`. If this is not the case, you are probably using an outdated version.

> For the frame_id: Is there another way to set it? If not, I assume I would need to republish an adjusted topic with the correct frame?

Or, you just implement the missing parameter and send a PR :-)