christianrauch/camera_ros #103: libcamera source meson setup build error

ARLunan, 2025-03-11:
On a Raspberry Pi 4/4 GB Ubuntu 22.04 following the Source steps to install libcamera. Installed colcon-meson with $ sudo apt install -y python3-colcon-meson. $ which meson showed installed in /usr/bin/meson. After cloning the libcamera repo to home directory, running -$ meson setup build ...  

Error : /home/ubuntu/.local/bin/meson: No such file or directory. Seem its expecting to find meson in the root ?

christianrauch, 2025-03-11:
It looks like you are mixing system-wide and user colcon installations. Just remove one of them and also the build folder from your workspace.

ARLunan, 2025-03-13:
May I ask another question: I realized I was NOT following your instructions, attempting to compile the libcamera from source separately from camera_ros. So I followed ur steps precisely which resulted in I believe a successful libcamera build, but failed : camera_ros

**ubuntu@rp4-ub22h-mt:~/camera_ws$** colcon build --event-handlers=console_direct+
Starting >>> libcamera
The Meson build system    
.
**The Meson Setup completed:**

Found ninja-1.10.1 at /usr/bin/ninja
INFO: autodetecting backend as ninja            
INFO: calculating backend command to run: /usr/bin/ninja
.
[303/303] Linking target src/py/libcamera/_libcamera.so
ninja: Entering directory `/home/ubuntu/camera_ws/build/libcamera'
 .                                                             
[0/22] Generating src/libcamera/version.cpp with a custom command
[1/22] Generating src/libcamera/version.cpp with a custom command
                         
Starting >>> camera_ros
.
**But** 
-- Generating done                                     
-- Build files have been written to: /home/ubuntu/camera_ws/build/camera_ros
[  7%] Building CXX object CMakeFiles/utils.dir/src/pretty_print.cpp.o
[ 23%] Building CXX object CMakeFiles/param.dir/src/clamp.cpp.o
[ 23%] Building CXX object CMakeFiles/utils.dir/src/format_mapping.cpp.o
[ 30%] Building CXX object CMakeFiles/camera_node.dir/rclcpp_components/node_main_camera_node.cpp.o
[ 38%] Building CXX object CMakeFiles/param.dir/src/cv_to_pv.cpp.o
[ 38%] Built target utils                                  
[ 46%] Building CXX object CMakeFiles/param.dir/src/pv_to_cv.cpp.o
[ 53%] Building CXX object CMakeFiles/param.dir/src/types.cpp.o
[ 61%] Building CXX object CMakeFiles/param.dir/src/type_extent.cpp.o
[ 69%] Building CXX object CMakeFiles/param.dir/src/ParameterHandler.cpp.o
[ 76%] Building CXX object CMakeFiles/param.dir/src/ParameterConflictHandler.cpp.o
[Processing: camera_ros]                                   
/home/ubuntu/camera_ws/src/camera_ros/src/type_extent.cpp: In function ‘std::size_t get_extent(const libcamera::ControlId*)’:
/home/ubuntu/camera_ws/src/camera_ros/src/type_extent.cpp:42:6: error: ‘AeLocked’ is not a member of ‘libcamera::controls’; did you mean ‘AwbLocked’?
   42 |   IF(AeLocked)
      |      ^~~~~~~~
/home/ubuntu/camera_ws/src/camera_ros/src/type_extent.cpp:42:6: error: ‘AeLocked’ is not a member of ‘libcamera::controls’; did you mean ‘AwbLocked’?
   42 |   IF(AeLocked)
      |      ^~~~~~~~
.
/home/ubuntu/camera_ws/src/camera_ros/src/type_extent.cpp:34:44: note: in definition of macro ‘IF’
   34 |     return get_extent(libcamera::controls::T);
      |                                            ^
gmake[2]: *** [CMakeFiles/param.dir/build.make:132: CMakeFiles/param.dir/src/type_extent.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:169: CMakeFiles/param.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< camera_ros [43.5s, exited with code 2]
                                       
Summary: 1 package finished [11min 14s]
  1 package failed: camera_ros
  1 package had stderr output: camera_ros
ubuntu@rp4-ub22h-mt:~/camera_ws$

Testing my V2.1 camera, that is listed on 
$ v4l2-ctl --list-devices 
unicam (platform:fe801000.csi):
	/dev/video0
	/dev/video1
	/dev/media0

**ubuntu@rp4-ub22h-mt:~/camera_ws$** cam -l
Segmentation fault (core dumped)

Obvious fix, or should I follow through asking this on robotics.stackexchange.com ?

christianrauch, 2025-03-13:
Without proper formating, this is a bit hard to read. What is the actual problem you are facing? If you have libcamera compilation errors in the @raspberrypi fork, please report them to https://github.com/raspberrypi/libcamera.

ARLunan, 2025-03-14:
My action was to delete  my trial workspaces, and rebuild the Meson & Colcon-meson dependencies following the libcamera repos instructions. I did a clean Build of the libcamera repos which generated a “version” ERROR. I posted a relevant  question on Robotics’s.stackexchange.com

closed 2025-03-14 by ARLunan