christianrauch/camera_ros #109: Regression on ROS 2 Buildfarm for Rolling

Yadunund, 2025-04-14:
`camera_ros 0.3.0` is failing to build on the buildfarm for Rolling. This is blocking the [announced Rolling sync](https://discourse.ros.org/t/preparing-for-rolling-sync-2025-04-11/43166). We will proceed with the sync which will remove these binaries from Rolling but they will be added once the build is fixed and a new release is bloomed.

```bash
23:13:40 /tmp/binarydeb/ros-rolling-camera-ros-0.3.0/src/type_extent.cpp: In function ‘std::size_t get_extent(const libcamera::ControlId*)’:
23:13:40 /tmp/binarydeb/ros-rolling-camera-ros-0.3.0/src/type_extent.cpp:41:6: error: ‘AeLocked’ is not a member of ‘libcamera::controls’; did you mean ‘AwbLocked’?
23:13:40    41 |   IF(AeLocked)
23:13:40       |      ^~~~~~~~
23:13:40 /tmp/binarydeb/ros-rolling-camera-ros-0.3.0/src/type_extent.cpp:32:40: note: in definition of macro ‘IF’
23:13:40    32 |   if (id->id() == libcamera::controls::T.id()) \
23:13:40       |                                        ^
23:13:40 /tmp/binarydeb/ros-rolling-camera-ros-0.3.0/src/type_extent.cpp:41:6: error: ‘AeLocked’ is not a member of ‘libcamera::controls’; did you mean ‘AwbLocked’?
23:13:40    41 |   IF(AeLocked)
23:13:40       |      ^~~~~~~~
23:13:40 /tmp/binarydeb/ros-rolling-camera-ros-0.3.0/src/type_extent.cpp:33:44: note: in definition of macro ‘IF’
23:13:40    33 |     return get_extent(libcamera::controls::T);

```

See: https://build.ros2.org/view/Rbin_uN64/job/Rbin_uN64__camera_ros__ubuntu_noble_amd64__binary/36/console

Yadunund, 2025-04-14:
> Is it not possible to keep the currently released versions (libcamera 0.4 and camera_ros 0.3) in the apt repo instead of removing them entirely and then only update once new versions build?

It is certainly possible. You can open a PR to downgrade the `libcamera` version to 0.4.0 in rosdistro if you think that will allow camera_ros 0.3.0 to build. Are you able to do that asap?

christianrauch, 2025-04-15:
Well, if all PRs to rolling are held back, then a re-release of 0.4 as a new PR would certainly also be held back?

What I actually meant by "Is it not possible to keep the currently released versions" is to keep the current libcamera 0.4 package in the "ros2" repo and do not move the libcamera 0.5 from "ros2-testing" over.

The fix for the issue above is on current master. I will create a new release once the libcamera 0.5 PR is merged.

christianrauch, 2025-04-15:
@Yadunund `camera_ros` release 0.4 (https://github.com/ros/rosdistro/pull/45181) should fix the build issue with `libcamera` 0.5.

christianrauch, 2025-04-15:
With the 0.4 release, the rolling build is back to normal: https://build.ros2.org/view/Rbin_uN64/job/Rbin_uN64__camera_ros__ubuntu_noble_amd64__binary/38/.

closed 2025-04-15 by christianrauch