christianrauch/camera_ros #113: Running 2 cameras in same container failed

loicvigneron, 2025-04-18:
Hello,

I'am trying to run two cameras, each in a composable node:

`from launch import LaunchDescription
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    return LaunchDescription([
        ComposableNodeContainer(
            name='container',
            namespace='stereo',
            package='rclcpp_components',
            executable='component_container_mt',
            composable_node_descriptions=[
                ComposableNode(
                    package='camera_ros',
                    plugin='camera::CameraNode',
                    name='left_camera_node',
                    parameters=[{'width': 640, 'height': 480, 'format': 'BGR888', 'camera': 1, }],
                    remappings=[
                        ('camera/image_raw', 'left/image_rect'),
                        ('camera/camera_info', 'left/camera_info'),
                        ('camera/image_raw/compressed', 'left/image_rect'),
                    ]
                ),
                ComposableNode(
                    package='camera_ros',
                    plugin='camera::CameraNode',
                    name='right_camera_node',
                    parameters=[{'width': 640, 'height': 480, 'format': 'BGR888', 'camera': 0, }],
                    remappings=[
                        ('camera/image_raw', 'right/image_rect'),
                        ('camera/camera_info', 'right/camera_info'),
                        ('camera/image_raw/compressed', 'right/image_rect'),
                    ]
                ),
            ],
            output='screen',
        )
    ])
`
The first one is starting but not the second, here is the output:

`ros2 launch perception perception_launch.py
[INFO] [launch]: All log files can be found below /home/pi/.ros/log/2025-04-19-00-10-44-256605-ovcs-ros2-6023
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_mt-1]: process started with pid [6038]
[component_container_mt-1] [INFO] [1745014244.820045882] [stereo.perception_container]: Load Library: /home/pi/Development/camera_ws/install/camera_ros/lib/libcamera_component.so
[component_container_mt-1] [INFO] [1745014244.924920649] [stereo.perception_container]: Found class: rclcpp_components::NodeFactoryTemplate<camera::CameraNode>
[component_container_mt-1] [INFO] [1745014244.925027390] [stereo.perception_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<camera::CameraNode>
[component_container_mt-1] [0:59:47.462490326] [6038]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
[component_container_mt-1] [0:59:47.491571411] [6067]  INFO RPI pisp.cpp:720 libpisp version v1.0.7 28196ed6edcf 17-04-2025 (19:50:16)
[component_container_mt-1] [0:59:47.504784007] [6067]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx477@1a to CFE device /dev/media1 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[component_container_mt-1] [0:59:47.504896156] [6067]  INFO RPI pisp.cpp:720 libpisp version v1.0.7 28196ed6edcf 17-04-2025 (19:50:16)
[component_container_mt-1] [0:59:47.523672557] [6067]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx477@1a to CFE device /dev/media2 and ISP device /dev/media4 using PiSP variant BCM2712_C0
[component_container_mt-1] [0:59:47.524425540] [6038]  WARN V4L2 v4l2_pixelformat.cpp:346 Unsupported V4L2 pixel format RPBP
[component_container_mt-1] [INFO] [1745014245.001998684] [left_camera_node]: 
[component_container_mt-1] >> stream formats:
[component_container_mt-1]    - R8 (32x32 - 4056x3040)
[component_container_mt-1]    - R16 (32x32 - 4056x3040)
[component_container_mt-1]    - MONO_PISP_COMP1 (32x32 - 4056x3040)
[component_container_mt-1]    - NV21 (32x32 - 4056x3040)
[component_container_mt-1]    - SBGGR8 (32x32 - 4056x3040)
[component_container_mt-1]    - SBGGR16 (32x32 - 4056x3040)
[component_container_mt-1]    - BGGR_PISP_COMP1 (32x32 - 4056x3040)
[component_container_mt-1]    - XBGR8888 (32x32 - 4056x3040)
[component_container_mt-1]    - BGR888 (32x32 - 4056x3040)
[component_container_mt-1]    - RGB888 (32x32 - 4056x3040)
[... 75 lines cut ...]
[component_container_mt-1] [WARN] [1745014245.005023726] [left_camera_node]: FrameDurationLimits: cannot set default scalar value '33333' on span control (extend: 2), default will be ignored
[component_container_mt-1] [WARN] [1745014245.005532209] [left_camera_node]: unknown control: NoiseReductionMode (10002)
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/left_camera_node' in container '/stereo/perception_container'
[component_container_mt-1] [INFO] [1745014245.529032374] [stereo.perception_container]: Found class: rclcpp_components::NodeFactoryTemplate<camera::CameraNode>
[component_container_mt-1] [INFO] [1745014245.529091615] [stereo.perception_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<camera::CameraNode>
[component_container_mt-1] [0:59:48.060467004] [6053] FATAL Process process.cpp:117 Multiple ProcessManager objects are not allowed
[component_container_mt-1] Backtrace:
[component_container_mt-1] libcamera::ProcessManager::ProcessManager()+0x94 (/home/pi/Development/camera_ws/install/libcamera/lib/libcamera.so.0.4.0 [0x0000ffffa5118cb4])
[component_container_mt-1] libcamera::CameraManager::Private::Private()+0x80 (/home/pi/Development/camera_ws/install/libcamera/lib/libcamera.so.0.4.0 [0x0000ffffa50ddec0])
[component_container_mt-1] libcamera::CameraManager::CameraManager()+0x40 (/home/pi/Development/camera_ws/install/libcamera/lib/libcamera.so.0.4.0 [0x0000ffffa50ddff0])
[component_container_mt-1] camera::CameraNode::CameraNode(rclcpp::NodeOptions const&)+0xb4 (/home/pi/Development/camera_ws/install/camera_ros/lib/libcamera_component.so [0x0000ffffa56178e8])
[component_container_mt-1] void std::_Construct<camera::CameraNode, rclcpp::NodeOptions const&>(camera::CameraNode*, rclcpp::NodeOptions const&)+0x70 (/home/pi/Development/camera_ws/install/camera_ros/lib/libcamera_component.so [0x0000ffffa56c3000])
[component_container_mt-1] std::_Sp_counted_ptr_inplace<camera::CameraNode, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<rclcpp::NodeOptions const&>(std::allocator<void>, rclcpp::NodeOptions const&)+0x74 (/home/pi/Development/camera_ws/install/camera_ros/lib/libcamera_component.so [0x0000ffffa56c29ec])
[component_container_mt-1] std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<camera::CameraNode, std::allocator<void>, rclcpp::NodeOptions const&>(camera::CameraNode*&, std::_Sp_alloc_shared_tag<std::allocator<void> >, rclcpp::NodeOptions const&)+0x9c (/home/pi/Development/camera_ws/install/camera_ros/lib/libcamera_component.so [0x0000ffffa56c2458])
[component_container_mt-1] std::__shared_ptr<camera::CameraNode, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<void>, rclcpp::NodeOptions const&>(std::_Sp_alloc_shared_tag<std::allocator<void> >, rclcpp::NodeOptions const&)+0x48 (/home/pi/Development/camera_ws/install/camera_ros/lib/libcamera_component.so [0x0000ffffa56c20f4])
[component_container_mt-1] std::shared_ptr<camera::CameraNode>::shared_ptr<std::allocator<void>, rclcpp::NodeOptions const&>(std::_Sp_alloc_shared_tag<std::allocator<void> >, rclcpp::NodeOptions const&)+0x34 (/home/pi/Development/camera_ws/install/camera_ros/lib/libcamera_component.so [0x0000ffffa56c1e34])
[component_container_mt-1] std::shared_ptr<camera::CameraNode> std::make_shared<camera::CameraNode, rclcpp::NodeOptions const&>(rclcpp::NodeOptions const&)+0x4c (/home/pi/Development/camera_ws/install/camera_ros/lib/libcamera_component.so [0x0000ffffa56c1c7c])
[component_container_mt-1] rclcpp_components::NodeFactoryTemplate<camera::CameraNode>::create_node_instance(rclcpp::NodeOptions const&)+0x40 (/home/pi/Development/camera_ws/install/camera_ros/lib/libcamera_component.so [0x0000ffffa56c1b04])
[component_container_mt-1] rclcpp_components::ComponentManager::on_load_node(std::shared_ptr<rmw_request_id_s>, std::shared_ptr<composition_interfaces::srv::LoadNode_Request_<std::allocator<void> > >, std::shared_ptr<composition_interfaces::srv::LoadNode_Response_<std::allocator<void> > >)+0x230 (/opt/ros/jazzy/lib/libcomponent_manager.so [0x0000ffffb95c4e90])
[component_container_mt-1] rclcpp_components::ComponentManager::on_load_node(std::shared_ptr<rmw_request_id_s>, std::shared_ptr<composition_interfaces::srv::LoadNode_Request_<std::allocator<void> > >, std::shared_ptr<composition_interfaces::srv::LoadNode_Response_<std::allocator<void> > >)+0x230 (/opt/ros/jazzy/lib/libcomponent_manager.so [0x0000ffffb95c4e90])`

Any idea why? :D I'am a bit lost :/

Thank you,
Loic

christianrauch, 2025-04-18:
The error is `FATAL Process process.cpp:117 Multiple ProcessManager objects are not allowed` and comes from libcamera here: https://gitlab.freedesktop.org/camera/libcamera/-/blame/master/src/libcamera/process.cpp?ref_type=heads#L116-118

I guess you are supposed to instantiate the `libcamera::ProcessManager` only once per process. You should check with the libcamera devs why this is the case and if there is a way around this. Because you are instantiating the camera node twice as composable node inside the container, the `ProcessManager` is also instantiated twice, hence the error message. The only way around this is to run all camera nodes as separate processes, e.g. by running them in separate containers.

Why do you want to run both camera nodes in the same container? They are not exchanging data with each other, so there is no benefit regarding zero-copy communication.

The other way around this is to support multiple cameras in the same node. You can only have one `libcamera::CameraManager`, but you can access multiple cameras through this. But I have never tried this, so don't know if this is actually supported, and it would require a massive restructuring of the code.

loicvigneron, 2025-04-19:
Thanks for the quick reply.
 
The file is partial, the reason I want to run them in the same container is because I want to create a points cloud. So eventually, the data of both cameras need to be merged.
I'll try to find a another way for this then :) I'll keep you posted if I find something interesting.

I'll close the issue as it's not coming from camera_ros.

Have a nice day,
Loic

closed 2025-04-19 by loicvigneron