christianrauch/camera_ros #115: Errors running on Raspberry Pi 5 + Bookworm + ROS2 Jazzy

fbridge8, 2025-04-23:
Good evening.

I'm facing issues trying to run `camera_ros` on a Raspberry Pi 5 with Bookworm and ROS 2 Jazzy. So my first question is regarding support for this platform.

I have two cameras but in this case I'm selection only the camera module v2 imx-219.

These are the error lines. I'm attaching the complete logs below.

```bash
[0:04:51.954171974] [1535] ERROR IPCUnixSocket ipc_unixsocket.cpp:192 Failed to send: Connection refused
[0:04:51.954192564] [1535] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[0:04:51.954201989] [1535] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:432 Failed to call mapBuffers
...
[0:04:52.053905410] [1535] ERROR IPCUnixSocket ipc_unixsocket.cpp:192 Failed to send: Transport endpoint is not connected
[0:04:52.053922260] [1535] ERROR IPCPipe ipc_pipe_unixsocket.cpp:80 Failed to call async
[0:04:52.053931352] [1535] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:508 Failed to call prepareIsp
```

<details><summary>Complete logs (selecting only one camera)</summary>
<p>

```bash
pi@raspberrypi:~/ros2_jazzy $ LIBCAMERA_LOG_LEVELS=*:DEBUG ros2 run --prefix 'gdb -ex run --args' camera_ros camera_node --ros-args --log-level camera:=debug -p camera:=1 -p width:=640 -p height:=480
GNU gdb (Debian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/pi/ros2_jazzy/install/camera_ros/lib/camera_ros/camera_node...
Starting program: /home/pi/ros2_jazzy/install/camera_ros/lib/camera_ros/camera_node --ros-args --log-level camera:=debug -p camera:=1 -p width:=640 -p height:=480
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff614a040 (LWP 1521)]
[New Thread 0x7ffff593a040 (LWP 1522)]
[New Thread 0x7fffefb7e040 (LWP 1523)]
[New Thread 0x7fffef36e040 (LWP 1524)]
[New Thread 0x7fffdf86e040 (LWP 1525)]
[New Thread 0x7fffdefd6040 (LWP 1526)]
[New Thread 0x7fffde7c6040 (LWP 1527)]
[New Thread 0x7fffddfb6040 (LWP 1528)]
[New Thread 0x7fffdd7a6040 (LWP 1529)]
[New Thread 0x7fffdcf96040 (LWP 1530)]
[New Thread 0x7fffdc6c6040 (LWP 1531)]
[New Thread 0x7fffdbe4e040 (LWP 1532)]
[New Thread 0x7fffdb63e040 (LWP 1533)]
[New Thread 0x7fffdae2e040 (LWP 1534)]
[0:04:49.772290628] [1519] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_pisp.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so is signed
[0:04:49.772360638] [1519] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so'
[0:04:49.772434277] [1519] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_vc4.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so is signed
[0:04:49.772512194] [1519] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
[0:04:49.846953168] [1519]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
[New Thread 0x7fffd9efe040 (LWP 1535)]
[0:04:49.847360692] [1535] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[0:04:49.853463828] [1535] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
[0:04:49.853563001] [1535] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 10 missing dependencies
[0:04:49.854327370] [1535] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[0:04:49.854345349] [1535] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
[0:04:49.854427802] [1535] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media3
[0:04:49.854507922] [1535] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 10 missing dependencies
[0:04:49.855256755] [1535] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[... 670 lines cut ...]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
dtoverlay=pcb4188_vd56g3_cam0
```

</p>
</details> 

System

```bash
pi@raspberrypi:~/ros2_jazzy $ uname -a
Linux raspberrypi 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
```

Any direction would be helpful. Let me know if there's anything else I could provide.

Thanks.

christianrauch, 2025-04-23:
Are you using the libcamera library that comes with the Raspberry Pi OS? Or are you compiling this from scratch on Debian?

Since the ROS node is just a wrapper around libcamera, you should first test if you can get images from the camera using the libcamera examples, such as `cam` and `qcam`.

I remember getting `Call timeout!` errors when IPA modules ran in isolation. From what I see in your log, the shared object modules for the IPAs have an invalid signature (`IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so signature is not valid`). Therefore, they run in isolation (different address spaces) and have to use expensive inter-process communication (IPC). To make this work without running IPA modules in isolation, they should be signed and have a valid signature (you should see a debug message `IPA module [...]/ipa_rpi_pisp.so signature is valid`).

See https://bugs.libcamera.org/show_bug.cgi?id=191 for a previous issue report on this to the libcamera project. I would suggest commenting there that you are also affected by the timeouts on the Raspberry Pi. But ideally, for the best performance, you should use a libcamera version with properly signed IPAs.

fbridge8, 2025-04-24:
I've got a clean image to start over.

> Are you using the libcamera library that comes with the Raspberry Pi OS? Or are you compiling this from scratch on Debian?

Using what came with the system (after update).

camera_ros works fine with that in the following setup:

- Camera module v2
- CAM1 on the Pi

<details><summary>Camera module alone</summary>
<p>

```bash
pi@raspberrypi:~/ros2_jazzy $ ros2 run camera_ros camera_node
[0:43:00.673266960] [99519]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
[0:43:00.674305910] [99537]  INFO RPI pisp.cpp:720 libpisp version v1.0.7 28196ed6edcf 24-04-2025 (17:02:39)
[0:43:00.686569149] [99537]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@1000120000/rp1/i2c@80000/imx219@10 to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[INFO] [1745503536.884532966] [camera]: 
>> cameras:
   0: imx219 (/base/axi/pcie@1000120000/rp1/i2c@80000/imx219@10)
[WARN] [1745503536.884752023] [camera]: no camera selected, using default: "/base/axi/pcie@1000120000/rp1/i2c@80000/imx219@10"
[WARN] [1745503536.884783968] [camera]: set parameter 'camera' to silence this warning
[INFO] [1745503536.885305730] [camera]: 
>> stream formats:
   - R8 (32x32 - 3280x2464)
   - R16 (32x32 - 3280x2464)
   - MONO_PISP_COMP1 (32x32 - 3280x2464)
   - NV21 (32x32 - 3280x2464)
   - SBGGR8 (32x32 - 3280x2464)
   - SBGGR16 (32x32 - 3280x2464)
   - BGGR_PISP_COMP1 (32x32 - 3280x2464)
   - XBGR8888 (32x32 - 3280x2464)
   - BGR888 (32x32 - 3280x2464)
   - RGB888 (32x32 - 3280x2464)
   - XRGB8888 (32x32 - 3280x2464)
   - SGBRG16 (32x32 - 3280x2464)
   - GBRG_PISP_COMP1 (32x32 - 3280x2464)
   - SGRBG16 (32x32 - 3280x2464)
   - GRBG_PISP_COMP1 (32x32 - 3280x2464)
   - SRGGB16 (32x32 - 3280x2464)
   - RGGB_PISP_COMP1 (32x32 - 3280x2464)
   - BGR161616 (32x32 - 3280x2464)
   - RGB161616 (32x32 - 3280x2464)
   - SRGGB8 (32x32 - 3280x2464)
   - SGRBG8 (32x32 - 3280x2464)
   - SGBRG8 (32x32 - 3280x2464)
   - YUYV (32x32 - 3280x2464)
   - UYVY (32x32 - 3280x2464)
[WARN] [1745503536.885447990] [camera]: no pixel format selected, auto-selecting: "XRGB8888"
[WARN] [1745503536.885476935] [camera]: set parameter 'format' to silence this warning
[INFO] [1745503536.885548972] [camera]: 
>> XRGB8888 format sizes:
   - 160x120
   - 240x160
   - 320x240
   - 400x240
   - 480x320
   - 640x360
   - 640x480
   - 720x480
   - 768x480
   - 854x480
   - 720x576
   - 800x600
   - 960x540
   - 1024x576
   - 960x640
   - 1024x600
[... 365 lines cut ...]
/home/pi/ros2_jazzy/install/libcamera/lib/libcamera/ipa_rpi_pisp.so.sign
/home/pi/ros2_jazzy/install/libcamera/lib/libcamera/ipa_rpi_pisp.so
/home/pi/ros2_jazzy/install/libcamera/lib/libpisp.so.1.0.7
/home/pi/ros2_jazzy/install/libcamera/lib/libpisp.so.1
```

</p>
</details> 

<details><summary>Comparison of LD_LIBRARY_PATH before and after source</summary>
<p>

```bash
pi@raspberrypi:~ $ echo $LD_LIBRARY_PATH > before_ld.txt
pi@raspberrypi:~ $ source ~/ros2_jazzy/install/setup.bash 
pi@raspberrypi:~ $ echo $LD_LIBRARY_PATH > after_ld.txt
pi@raspberrypi:~ $ diff before_ld.txt after_ld.txt
1c1
< 
---
> /home/pi/ros2_jazzy/install/rosbag2_storage_mcap/lib:/home/pi/ros2_jazzy/install/rosbag2_compression_zstd/lib:/home/pi/ros2_jazzy/install/mcap_vendor/lib:/home/pi/ros2_jazzy/install/stereo_image_proc/lib:/home/pi/ros2_jazzy/install/rosbag2_transport/lib:/home/pi/ros2_jazzy/install/rosbag2_storage_sqlite3/lib:/home/pi/ros2_jazzy/install/rosbag2_compression/lib:/home/pi/ros2_jazzy/install/rosbag2_cpp/lib:/home/pi/ros2_jazzy/install/rosbag2_storage/lib:/home/pi/ros2_jazzy/install/camera_ros/lib:/home/pi/ros2_jazzy/install/image_view/lib:/home/pi/ros2_jazzy/install/image_publisher/lib:/home/pi/ros2_jazzy/install/depth_image_proc/lib:/home/pi/ros2_jazzy/install/image_proc/lib:/home/pi/ros2_jazzy/install/camera_info_manager/lib:/home/pi/ros2_jazzy/install/camera_calibration_parsers/lib:/home/pi/ros2_jazzy/install/yaml_cpp_vendor/opt/yaml_cpp_vendor/lib:/home/pi/ros2_jazzy/install/visualization_msgs/lib:/home/pi/ros2_jazzy/install/robot_state_publisher/lib:/home/pi/ros2_jazzy/install/kdl_parser/lib:/home/pi/ros2_jazzy/install/urdf/lib:/home/pi/ros2_jazzy/install/urdfdom/lib:/home/pi/ros2_jazzy/install/turtlesim/lib:/home/pi/ros2_jazzy/install/trajectory_msgs/lib:/home/pi/ros2_jazzy/install/image_rotate/lib:/home/pi/ros2_jazzy/install/tf2_ros/lib:/home/pi/ros2_jazzy/install/tf2_msgs/lib:/home/pi/ros2_jazzy/install/tf2_eigen_kdl/lib:/home/pi/ros2_jazzy/install/laser_geometry/lib:/home/pi/ros2_jazzy/install/tf2/lib:/home/pi/ros2_jazzy/install/test_msgs/lib:/home/pi/ros2_jazzy/install/stereo_msgs/lib:/home/pi/ros2_jazzy/install/std_srvs/lib:/home/pi/ros2_jazzy/install/shape_msgs/lib:/home/pi/ros2_jazzy/install/quality_of_service_demo_cpp/lib:/home/pi/ros2_jazzy/install/point_cloud_transport/lib:/home/pi/ros2_jazzy/install/map_msgs/lib:/home/pi/ros2_jazzy/install/image_transport/lib:/home/pi/ros2_jazzy/install/image_tools/lib:/home/pi/ros2_jazzy/install/image_geometry/lib:/home/pi/ros2_jazzy/install/cv_bridge/lib:/home/pi/ros2_jazzy/install/sensor_msgs/lib:/home/pi/ros2_jazzy/install/nav_msgs/lib:/home/pi/ros2_jazzy/install/message_filters/lib:/home/pi/ros2_jazzy/install/logging_demo/lib:/home/pi/ros2_jazzy/install/diagnostic_msgs/lib:/home/pi/ros2_jazzy/install/geometry_msgs/lib:/home/pi/ros2_jazzy/install/demo_nodes_cpp_native/lib:/home/pi/ros2_jazzy/install/demo_nodes_cpp/lib:/home/pi/ros2_jazzy/install/composition/lib:/home/pi/ros2_ja [... cut ...] me/pi/ros2_jazzy/install/iceoryx_hoofs/lib:/home/pi/ros2_jazzy/install/iceoryx_binding_c/lib:/home/pi/ros2_jazzy/install/gz_math_vendor/opt/gz_math_vendor/lib:/home/pi/ros2_jazzy/install/gz_utils_vendor/opt/gz_utils_vendor/lib:/home/pi/ros2_jazzy/install/gz_cmake_vendor/opt/gz_cmake_vendor/lib:/home/pi/ros2_jazzy/install/fastrtps/lib:/home/pi/ros2_jazzy/install/fastcdr/lib:/home/pi/ros2_jazzy/install/google_benchmark_vendor/lib:/home/pi/ros2_jazzy/install/ament_index_cpp/lib
```

</p>
</details>

fbridge8, 2025-04-24:
So I've removed libcamera from `ros2_jazzy/src`, removed it from the build and install along with camera_ros, ran rosdep and colcon build once again and the result is exactly what I was seeing in the errors in my first message (but libcamera-hello still works with sourced env now).

<details><summary>camera_ros camera 0</summary>
<p>

```bash
pi@raspberrypi:~/ros2_jazzy $ ros2 run camera_ros camera_node --ros-args -p camera:=0
[2:44:36.922085472] [269888]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
[2:44:36.931816350] [269906]  INFO RPI pisp.cpp:720 libpisp version v1.1.0 e7974a156008 27-01-2025 (21:50:51)
[2:44:36.933719503] [269906]  WARN CameraSensorProperties camera_sensor_properties.cpp:473 No static properties available for 'vd56g3'
[2:44:36.933735225] [269906]  WARN CameraSensorProperties camera_sensor_properties.cpp:475 Please consider updating the camera sensor properties database
[2:44:36.966405885] [269906]  WARN CameraSensor camera_sensor_legacy.cpp:501 'vd56g3 10-0010': No sensor delays found in static properties. Assuming unverified defaults.
[2:44:36.966868534] [269906]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@1000120000/rp1/i2c@88000/fox@10 to CFE device /dev/media1 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[2:44:36.966944849] [269906]  INFO RPI pisp.cpp:720 libpisp version v1.1.0 e7974a156008 27-01-2025 (21:50:51)
[2:44:37.028142862] [269906]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@1000120000/rp1/i2c@80000/imx219@10 to CFE device /dev/media4 and ISP device /dev/media3 using PiSP variant BCM2712_C0
[INFO] [1745518492.208938013] [camera]: 
>> stream formats:
   - R8 (32x32 - 1124x1364)
   - R16 (32x32 - 1124x1364)
   - MONO_PISP_COMP1 (32x32 - 1124x1364)
   - NV21 (32x32 - 1124x1364)
   - SBGGR8 (32x32 - 1124x1364)
   - SBGGR16 (32x32 - 1124x1364)
   - BGGR_PISP_COMP1 (32x32 - 1124x1364)
   - XBGR8888 (32x32 - 1124x1364)
   - BGR888 (32x32 - 1124x1364)
   - RGB888 (32x32 - 1124x1364)
   - XRGB8888 (32x32 - 1124x1364)
   - SGBRG16 (32x32 - 1124x1364)
   - GBRG_PISP_COMP1 (32x32 - 1124x1364)
   - SGRBG16 (32x32 - 1124x1364)
   - GRBG_PISP_COMP1 (32x32 - 1124x1364)
   - SRGGB16 (32x32 - 1124x1364)
   - RGGB_PISP_COMP1 (32x32 - 1124x1364)
   - BGR161616 (32x32 - 1124x1364)
   - RGB161616 (32x32 - 1124x1364)
   - SRGGB8 (32x32 - 1124x1364)
   - SGRBG8 (32x32 - 1124x1364)
   - SGBRG8 (32x32 - 1124x1364)
   - YUYV (32x32 - 1124x1364)
   - UYVY (32x32 - 1124x1364)
[WARN] [1745518492.209266143] [camera]: no pixel format selected, auto-selecting: "XRGB8888"
[WARN] [1745518492.209301958] [camera]: set parameter 'format' to silence this warning
[INFO] [1745518492.209349903] [camera]: 
>> XRGB8888 format sizes:
   - 160x120
   - 240x160
   - 320x240
   - 400x240
   - 480x320
   - 640x360
   - 640x480
   - 720x480
   - 768x480
   - 854x480
   - 720x576
   - 800x600
   - 960x540
   - 1024x576
   - 960x640
   - 1024x600
   - 1024x768
[WARN] [1745518492.209441014] [camera]: no dimensions selected, auto-selecting: "800x600"
[WARN] [1745518492.209465218] [camera]: set parameter 'width' or 'height' to silence this warning
[2:44:37.029566013] [269888]  INFO Camera camera.cpp:1202 configuring streams: (0) 800x600-XRGB8888
[2:44:37.029761921] [269906]  INFO RPI pisp.cpp:1484 Sensor: /base/axi/pcie@1000120000/rp1/i2c@88000/fox@10 - Selected sensor format: 1024x768-Y10_1X10 - Selected CFE format: 1024x768-PC1M
[INFO] [1745518492.210325294] [camera]: camera "/base/axi/pcie@1000120000/rp1/i2c@88000/fox@10" configured with 800x600-XRGB8888 stream
[INFO] [1745518492.210763277] [camera]: using default calibration URL
[INFO] [1745518492.210835129] [camera]: camera calibration URL: file:///home/pi/.ros/camera_info/vd56g3__base_axi_pcie_1000120000_rp1_i2c_88000_fox_10_800x600.yaml
[ERROR] [1745518492.210956074] [camera_calibration_parsers]: Unable to open camera calibration file [/home/pi/.ros/camera_info/vd56g3__base_axi_pcie_1000120000_rp1_i2c_88000_fox_10_800x600.yaml]
[... 135 lines cut ...]
[2:44:49.194112430] [269944] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:261 Failed to call start
[2:44:49.198663590] [269944] ERROR IPCUnixSocket ipc_unixsocket.cpp:192 Failed to send: Connection refused
[2:44:49.198708664] [269944] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[2:44:49.198720182] [269944] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:432 Failed to call mapBuffers
[2:44:49.277539629] [269944] ERROR IPCUnixSocket ipc_unixsocket.cpp:192 Failed to send: Transport endpoint is not connected
[2:44:49.277575388] [269944] ERROR IPCPipe ipc_pipe_unixsocket.cpp:80 Failed to call async
[2:44:49.277587314] [269944] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:508 Failed to call prepareIsp
^C[INFO] [1745518559.449703061] [rclcpp]: signal_handler(signum=2)
[2:45:44.273751181] [269944] ERROR IPCUnixSocket ipc_unixsocket.cpp:192 Failed to send: Transport endpoint is not connected
[2:45:44.273787496] [269944] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[2:45:44.273797922] [269944] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:317 Failed to call stop
[2:45:44.273877793] [269944] ERROR IPCUnixSocket ipc_unixsocket.cpp:192 Failed to send: Transport endpoint is not connected
[2:45:44.273897515] [269944] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[2:45:44.273907108] [269944] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:470 Failed to call unmapBuffers
[... cut ...]
[2:45:44.278472230] [269944] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[2:45:44.278490619] [269944] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:470 Failed to call unmapBuffers
[2:45:44.278686638] [269944] ERROR IPCUnixSocket ipc_unixsocket.cpp:192 Failed to send: Transport endpoint is not connected
[2:45:44.278715638] [269944] ERROR IPCPipe ipc_pipe_unixsocket.cpp:80 Failed to call async
```

</p>
</details> 

This only happens with the second camera (and ipa package installed). Any direction or process I could try from here?

fbridge8, 2025-04-24:
> What does "what came with the system" mean? Are you using the standard Raspberry Pi OS? Who is providing and maintaining your libcamera package?

The built in system package of libcamera. As I mentioned earlier, everything works fine with this, so I didn't built it from source. This is what I meant above with "All good (this proves that the local libcamera works right?" local being the system package*

The issue arises once I install ST's official IPA modules, available at https://www.st.com/en/embedded-software/stsw-img502.html

This is what my libcamera installations look like.

```bash
pi@raspberrypi:~ $ dpkg -l | grep libcamera
ii  libcamera-dev:arm64                          0.4.0+rpt20250213-1              arm64        complex camera support library (development files)
ii  libcamera-ipa:arm64                          0.4.0+rpt20250213+st20250401-1   arm64        complex camera support library (IPA modules)
ii  libcamera0.4:arm64                           0.4.0+rpt20250213-1              arm64        complex camera support library
```

<details><summary>Logs with libcamera debug</summary>
<p>

```bash
pi@raspberrypi:~ $ LIBCAMERA_LOG_LEVELS=*:DEBUG ros2 run --prefix 'gdb -ex run --args' camera_ros camera_node --ros-args --log-level camera:=debug -p camera:=1 -p width:=640 -p height:=480
GNU gdb (Debian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/pi/ros2_jazzy/install/camera_ros/lib/camera_ros/camera_node...
(No debugging symbols found in /home/pi/ros2_jazzy/install/camera_ros/lib/camera_ros/camera_node)
Starting program: /home/pi/ros2_jazzy/install/camera_ros/lib/camera_ros/camera_node --ros-args --log-level camera:=debug -p camera:=1 -p width:=640 -p height:=480
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff614a040 (LWP 270327)]
[New Thread 0x7ffff593a040 (LWP 270328)]
[New Thread 0x7fffefb7e040 (LWP 270329)]
[New Thread 0x7fffef36e040 (LWP 270330)]
[New Thread 0x7fffde4ae040 (LWP 270331)]
[New Thread 0x7fffddc9e040 (LWP 270332)]
[New Thread 0x7fffdb48e040 (LWP 270333)]
[New Thread 0x7fffd6c7e040 (LWP 270334)]
[New Thread 0x7fffd63e6040 (LWP 270335)]
[New Thread 0x7fffd5bd6040 (LWP 270336)]
[New Thread 0x7fffd53c6040 (LWP 270337)]
[New Thread 0x7fffd4bb6040 (LWP 270338)]
[New Thread 0x7fffd43a6040 (LWP 270339)]
[New Thread 0x7fffd3ad6040 (LWP 270340)]
[New Thread 0x7fffd325e040 (LWP 270341)]
[New Thread 0x7fffd2a4e040 (LWP 270342)]
[New Thread 0x7fffd223e040 (LWP 270343)]
[4:46:50.893003187] [270325] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_pisp.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so is signed
[4:46:50.893154817] [270325] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so'
[4:46:50.893236114] [270325] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_vc4.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so is signed
[4:46:50.893303670] [270325] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
[4:46:50.967869498] [270325]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
[New Thread 0x7fffd130e040 (LWP 270344)]
[4:46:50.968285222] [270344] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[4:46:50.974364238] [270344] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media1
[4:46:50.974455239] [270344] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 10 missing dependencies
[4:46:50.975206687] [270344] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[4:46:50.975223873] [270344] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: rp1-cfe
[4:46:50.975301947] [270344] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media4
[4:46:50.975382559] [270344] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media4 due to 10 missing dependencies
[... 562 lines cut ...]
[4:46:53.276130486] [270344] DEBUG V4L2 v4l2_videodevice.cpp:1731 /dev/video4[57:cap]: Dequeuing buffer 3
[4:46:53.276152190] [270344] DEBUG RPI pisp.cpp:1712 Stream CFE Image buffer dequeue, buffer id 4, timestamp: 17213266995000
[4:46:53.276169504] [270344] DEBUG RPiDelayedControls delayed_controls.cpp:220 Reading Exposure to 1058 at index 1
[4:46:53.276185838] [270344] DEBUG RPiDelayedControls delayed_controls.cpp:220 Reading Horizontal Blanking to 2808 at index 1
[4:46:53.276202690] [270344] DEBUG RPiDelayedControls delayed_controls.cpp:220 Reading Analogue Gain to 0 at index 1
[... cut ...]
[4:46:53.276385172] [270344] DEBUG RPISTREAM rpi_stream.cpp:271 Queuing buffer 6 for CFE Config
[4:46:53.276404284] [270344] DEBUG V4L2 v4l2_videodevice.cpp:2011 /dev/video7[58:out]: Queueing buffer 5
[4:46:53.276474414] [270344] DEBUG V4L2 v4l2_videodevice.cpp:1731 /dev/video6[59:cap]: Dequeuing buffer 3
[4:46:53.276495006] [270344] DEBUG RPI pisp.cpp:1712 Stream CFE Stats buffer dequeue, buffer id 4, timestamp: 17213266995000
```

</p>
</details> 

I'll check with ST what's the situation with their IPA modules and will report back (may take a while).

christianrauch, 2025-04-24:
> > What does "what came with the system" mean? Are you using the standard Raspberry Pi OS? Who is providing and maintaining your libcamera package?
> 
> The built in system package of libcamera. As I mentioned earlier, everything works fine with this, so I didn't built it from source. This is what I meant above with "All good (this proves that the local libcamera works right?" local being the system package*
> 
> The issue arises once I install ST's official IPA modules, available at https://www.st.com/en/embedded-software/stsw-img502.html
> 
> This is what my libcamera installations look like.
> 
> pi@raspberrypi:~ $ dpkg -l | grep libcamera
> ii  libcamera-dev:arm64                          0.4.0+rpt20250213-1              arm64        complex camera support library (development files)
> ii  libcamera-ipa:arm64                          0.4.0+rpt20250213+st20250401-1   arm64        complex camera support library (IPA modules)
> ii  libcamera0.4:arm64                           0.4.0+rpt20250213-1              arm64        complex camera support library

So, these Debian packages are provided by ST? Are you sure that this is the library that you use with the camera_ros node and that you are not mixing up different libcamera installations, e.g. from the ROS workspace?

> Logs with libcamera debug

Those are the logs with the ROS node and not the `libcamera-hello` executable I was asking above. These logs show the same `signature is not valid` issue for `/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so`.

I can only repeat this once again, that you either have to resolve the timeout issue by asking in the libcamera bug or use IPA modules with a valid signature. If you are mixing up libcamera packages and e.g. load IPA modules from a different package, the signature validation will also fail.

> I'll check with ST what's the situation with their IPA modules and will report back (may take a while).
> 
> edit: even though that only happens with ROS2 and camera_ros. The system libcamera does not report any problem with signatures. I'll try to build libcamera from source next.

fbridge8, 2025-04-24:
> So, these Debian packages are provided by ST?

Only `libcamera-ipa:arm64 0.4.0+rpt20250213+st20250401-1` comes from ST.

> Are you sure that this is the library that you use with the camera_ros node and that you are not mixing up different libcamera installations, e.g. from the ROS workspace?

I'm not sure. How can I check that exactly?

<details><summary>libcamera-hello debug</summary>
<p>

```bash
pi@raspberrypi:~ $ LIBCAMERA_LOG_LEVELS=*:DEBUG libcamera-hello -t 500 >> before_source.txt
[5:56:25.226107417] [270640] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_pisp.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so is signed
[5:56:25.226182195] [270640] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so'
[5:56:25.226262232] [270640] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_vc4.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so is signed
[5:56:25.226283418] [270640] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
[5:56:25.226294881] [270640]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
[5:56:25.226420067] [270643] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[5:56:25.235671416] [270643] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media1
[5:56:25.235799731] [270643] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 10 missing dependencies
[5:56:25.236948219] [270643] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[5:56:25.236970997] [270643] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: rp1-cfe
[5:56:25.237081554] [270643] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media4
[5:56:25.237190721] [270643] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media4 due to 10 missing dependencies
[5:56:25.238318153] [270643] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media4 found
[5:56:25.238338838] [270643] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media4: rp1-cfe
[5:56:25.238419024] [270643] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpi-hevc-dec" created from /dev/media2
[5:56:25.238438376] [270643] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 1 missing dependencies
[5:56:25.238566488] [270643] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[5:56:25.238584692] [270643] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: rpi-hevc-dec
[5:56:25.238678118] [270643] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media0
[5:56:25.238698378] [270643] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 8 missing dependencies
[5:56:25.238789563] [270643] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media3
[5:56:25.238809582] [270643] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 8 missing dependencies
[5:56:25.239668883] [270643] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[5:56:25.239689105] [270643] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: pispbe
[5:56:25.240557185] [270643] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[5:56:25.240577444] [270643] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: pispbe
[5:56:25.240943909] [270643] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/vc4'
[5:56:25.240979965] [270643] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[5:56:25.241044021] [270643] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[5:56:25.241069095] [270643] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/pisp'
[5:56:25.241079854] [270643] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "rp1-cfe"
[5:56:25.241107984] [270643] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "pispbe"
[5:56:25.241214392] [270643] DEBUG ClockRec clock_recovery.cpp:69 configure 100 2000 10 50000
[5:56:25.241223466] [270643] DEBUG ClockRec clock_recovery.cpp:83 reset
[5:56:25.241332856] [270643]  INFO RPI pisp.cpp:720 libpisp version v1.1.0 e7974a156008 27-01-2025 (21:50:51)
[5:56:25.242087878] [270643] DEBUG CameraSensor camera_sensor_raw.cpp:210 vd56g3 10-0010: unsupported number of sinks (0) or sources (1)
[5:56:25.242126379] [270643] DEBUG V4L2 v4l2_device.cpp:699 'vd56g3 10-0010': Control: Exposure (0x00980911)
[5:56:25.242149731] [270643] DEBUG V4L2 v4l2_device.cpp:699 'vd56g3 10-0010': Control: Horizontal Flip (0x00980914)
[5:56:25.242161916] [270643] DEBUG V4L2 v4l2_device.cpp:699 'vd56g3 10-0010': Control: Vertical Flip (0x00980915)
[5:56:25.242172342] [270643] DEBUG V4L2 v4l2_device.cpp:699 'vd56g3 10-0010': Control: Temperature in celsius (0x00981920)
[... cut ...]
[5:56:25.921319914] [270643] DEBUG Event event_dispatcher_poll.cpp:215 next timer 0x7fff72719390 expires in 1.999980593
```

</p>
</details> 

The same `signature is not valid` happens, but it doesn't break.

I also noticed that (irrespective of before or after sourcing), in the very first lines of these logs you can see:

```bash
[5:56:25.226294881] [270640]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
```

This version doesn't match what I have installed. I'll check where's that coming from.

Thanks for your time and help ❤

christianrauch, 2025-04-24:
> > So, these Debian packages are provided by ST?
> 
> Only `libcamera-ipa:arm64 0.4.0+rpt20250213+st20250401-1` comes from ST.

I guess if the libraries come from different sources, they use different signing keys. The upstream IPA modules are signed at build time with a newly generated key. If you use other IPAs from somewhere else, they were probably signed with a different key, so it is not possible to verify them without the matching key.

> > Are you sure that this is the library that you use with the camera_ros node and that you are not mixing up different libcamera installations, e.g. from the ROS workspace?
> 
> I'm not sure. How can I check that exactly?

Well, just don't install the same library multiple times :-) Search in the filesystem if the same so files exist multiple times.

> libcamera-hello debug
> The same `signature is not valid` happens, but it doesn't break.

I see. Same invalid signature, but no timeouts. In this case, check with the libcamera developers why the timeouts happen in one case and not the other.

kbingham, 2025-04-25:
If ST are supplying a separate IPA - they /certainly/ can not be using the same signing key - and the ST IPA will be treated as 'untrusted' and therefore always forced into isolation.

As this means the logs are currently not then visible - something is crashing inside of ST's IPA - and we're not seeing /what/.

I'm afraid - this is definitely something that has to be escallated to ST - and ideally - request that ST upstream - or otherwise publish their changes.

Otherwise you are installing binaries that are crashing and now causing you pain :-(

christianrauch, 2025-05-15:
@fbridge8 Can I assume that your issue has been fixed (https://community.st.com/t5/imaging-sensors/invalid-signature-error-in-ipa-module-for-vd56g3/m-p/796819)? 

In any case, I think there is nothing `camera_ros` as a consumer of `libcamera`, or even `libcamera` itself, can do about weird third-party IPAs. The isolation is behaving as intended and according to https://github.com/raspberrypi/libcamera/issues/258#issuecomment-2829931433, the `Call timeout!` and other IPC issues stem from the isolated IPA process crashing or otherwise not responding.

christianrauch, 2025-06-11:
I assume this issue is not relevant any more.

closed 2025-06-11 by christianrauch