christianrauch/camera_ros #117: Issue with Image Timestamps — Occasionally in the Future

dkloeseratb, 2025-04-30:
Hi, and thanks for your work on this driver!

We're encountering an issue with the image message timestamps produced by the driver. Occasionally, the timestamps appear to be in the future, which causes problems with downstream consumers such as tf calculations and synchronization logic.

While investigating, we came across this code snippet, where a time offset between the system clock and the device clock is calculated at startup and then added to the timestamp of every image:

https://github.com/christianrauch/camera_ros/blob/1accefc9c3c4f51029b148fff9d5ecdc91c1894b/src/CameraNode.cpp#L614-L619

We're unsure about the rationale behind this approach. Why not simply use the system time at the moment the image is received? Or alternatively, if the device provides a reliable timestamp, why not just use the device's time directly?

christianrauch, 2025-04-30:
> Why not simply use the system time at the moment the image is received?

Because this will not be the "true" time when the image was generated on the sensor. E.g., if the camera generates evenly spaced images and sends them to the PC, and the node receives these, e.g. as a burst, they will share a very similar timestamp from the PC reference time, even though they were correctly and evenly generated on the device. The time on the sensor is the "true" time that is independent of any communication delay.

> Or alternatively, if the device provides a reliable timestamp, why not just use the device's time directly?

I assume that the device will have its own time source and will not be aware of the "real" time. Hence, I assume that the device can only be used for relative time stamps with a device local reference, but not for absolute points in time that relate to other sensors or the PC with its own reference time.

Unless there is a way to set the time reference on the device itself, we have to apply the offset as time reference conversion manually. I am not aware of an alternative way to do this with libcamera. But let me know if you find one. There may also be more advanced time synchronisation methods to adjust the two time sources (PC and device). I would be happy to accept a PR implementing this.

> Occasionally, the timestamps appear to be in the future, which causes problems with downstream consumers such as tf calculations and synchronization logic.

This would mean that the time on the device jumps. Have you seen these time jumps in the metadata of the image (`metadata.timestamp`)?

dkloeseratb, 2025-05-05:
Thanks for your response. It make it much clearer to me, why you implemented it this way.

I will have another look into the effects that we observer...

closed 2025-05-05 by dkloeseratb