christianrauch/camera_ros #118: Errors running camera_node after update

dbacher42, 2025-05-06:
I installed this package via the DEB packages onto a series of Raspberry Pis. Today I did a sudo apt update && upgrade on one of them, which updated camera_ros to version 0.4.0 and libcamera to version 0.5.0. I can no longer launch the camera_node after the update. When I run the package with ros2 run camera_ros camera_node, I get the following output to the console:

``
[1:20:55.039685089] [5723] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
``
``
[1:20:55.039779050] [5723] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
``

From there, the output is normal until:

``
terminate called after throwing an instance of 'std::out_of_range'
  what():  unordered_map::at
``

I have another Raspberry Pi that is still running version 0.3.0 and it doesn't have the issue. Any idea what could be causing this?

christianrauch, 2025-05-07:
`ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info` is a "harmless error" as `v4l2-compat.so` is indeed not an IPA module. I reported this confusing warning at https://bugs.libcamera.org/show_bug.cgi?id=268. You will get the same "error" with the standard Ubuntu package `libcamera-v4l2`. E.g. installing libcamera without IPAs on Ubuntu 24.04 and running the `cam` example:
```sh
sudo apt install libcamera-tools libcamera-v4l2
cam -l
```
will also print:
```
ERROR IPAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
ERROR IPAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
 WARN IPAManager ipa_manager.cpp:154 No IPA found in '/usr/lib/x86_64-linux-gnu/libcamera'
 INFO Camera camera_manager.cpp:284 libcamera v0.2.0
```

So, this "error" can be ignored and should not cause further issues and any other issue must originate from somewhere else.

Can you provide more information about your hardware setup and debug information as per documentation?

dbacher42, 2025-05-07:
I'm running on a Raspberry Pi 5 running Ubuntu 24.04.2 with ROS2 Jazzy. The camera I'm using is a Microsoft Lifecam Cinema, just a random USB webcam I had lying around. It was working just fine with this package until I updated to version 0.4.0.

I installed gdb and ran in debug mode as described in the documentation:

 This was the output:
```
[2:51:26.612390138] [39991]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0+59-d83ff0a4
[New Thread 0xffffd600e040 (LWP 40007)]
[INFO] [1746651929.719003673] [camera]: 
>> cameras:
   0: Microsoft LifeCam Cinema(TM): (/base/axi/pcie@120000/rp1/usb@300000-1:1.0-045e:075d)
[WARN] [1746651929.719204488] [camera]: no camera selected, using default: "/base/axi/pcie@120000/rp1/usb@300000-1:1.0-045e:075d"
[WARN] [1746651929.719239765] [camera]: set parameter 'camera' to silence this warning
[DEBUG] [1746651929.725226891] [camera]: default Viewfinder stream configuration: "1280x720-MJPEG"
[DEBUG] [1746651929.725703317] [camera]: 
>> stream formats:
   - MJPEG (160x120 - 1280x720)
   - YUYV (160x120 - 1280x800)
[INFO] [1746651929.725934298] [camera]: 
>> stream formats:
   - MJPEG (160x120 - 1280x720)
   - YUYV (160x120 - 1280x800)
[WARN] [1746651929.726008780] [camera]: no pixel format selected, auto-selecting: "MJPEG"
[WARN] [1746651929.726080928] [camera]: set parameter 'format' to silence this warning
[INFO] [1746651929.726215706] [camera]: 
>> MJPEG format sizes:
   - 160x120
   - 176x144
   - 320x240
   - 416x240
   - 352x288
   - 640x360
   - 640x480
   - 800x448
   - 800x600
   - 960x544
   - 1280x720
[2:51:26.754040591] [39991]  INFO Camera camera.cpp:1205 configuring streams: (0) 1280x720-MJPEG
[INFO] [1746651929.742760622] [camera]: camera "/base/axi/pcie@120000/rp1/usb@300000-1:1.0-045e:075d" configured with 1280x720-MJPEG stream
[INFO] [1746651929.744370250] [camera]: using default calibration URL
[INFO] [1746651929.744639121] [camera]: camera calibration URL: file:///home/dbacher/.ros/camera_info/MicrosoftLifeCamCinema_TM____base_axi_pcie_120000_rp1_usb_300000_1_1_0_045e_075d_1280x720.yaml
[DEBUG] [1746651929.744757880] [camera]: reading camera calibration from /home/dbacher/.ros/camera_info/MicrosoftLifeCamCinema_TM____base_axi_pcie_120000_rp1_usb_300000_1_1_0_045e_075d_1280x720.yaml
[DEBUG] [1746651929.747321230] [camera]: declare 'ExposureTime' (type: integer)
[DEBUG] [1746651929.748386878] [camera]: declare 'Saturation' (type: double)
[DEBUG] [1746651929.749023340] [camera]: declare 'ExposureTimeMode' (type: integer)
[DEBUG] [1746651929.749232396] [camera]: declare 'Contrast' (type: double)
[DEBUG] [1746651929.749639173] [camera]: declare 'Brightness' (type: double)
terminate called after throwing an instance of 'std::out_of_range'
  what():  unordered_map::at

Thread 1 "camera_node" received signal SIGABRT, Aborted.
Download failed: Invalid argument.  Continuing without source file ./nptl/./nptl/pthread_kill.c.
__pthread_kill_implementation (threadid=281474842202144, signo=signo@entry=6, no_tid=no_tid@entry=0)
    at ./nptl/pthread_kill.c:44
warning: 44	./nptl/pthread_kill.c: No such file or directory
```

The problem must have something to do with the std::out_of_range error at the end but I'm not sure what could be causing that.

christianrauch, 2025-05-07:
You have to provide more information to identify the exact issue.

1. There is no backtrace in the log. Please show the backtrace or upload a core dump. If you use the binary version from the Debian package, you have to install the debug symbols from `ros-jazzy-camera-ros-dbgsym`. Or you compile the same version from source.

2. Can you provide steps to reproduce this? Does this also happen with other cameras? Can you list the camera properties (formats, controls, ...) via the `cam` utility (you probably have to compile libcamera from source)?
 
3. Can you reproduce this with earlier versions and the latest (current `main`) version of `camera_ros`? Which is the latest working version of `camera_ros` where this issue does not occur?

dbacher42, 2025-05-07:
1.  Here is the backtrace:
```
#0  __pthread_kill_implementation (threadid=281474842202144, signo=signo@entry=6, no_tid=no_tid@entry=0)
    at ./nptl/pthread_kill.c:44
#1  0x0000fffff76c7670 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  0x0000fffff767cb3c in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x0000fffff7667e00 in __GI_abort () at ./stdlib/abort.c:79
#4  0x0000fffff78ad4d8 in __gnu_cxx::__verbose_terminate_handler ()
    at ../../../../src/libstdc++-v3/libsupc++/vterminate.cc:95
#5  0x0000fffff78aa570 in __cxxabiv1::__terminate (handler=<optimized out>)
    at ../../../../src/libstdc++-v3/libsupc++/eh_terminate.cc:48
#6  0x0000fffff78a0e04 in std::terminate () at ../../../../src/libstdc++-v3/libsupc++/eh_terminate.cc:58
#7  0x0000fffff78aa908 in __cxxabiv1::__cxa_throw (obj=<optimized out>, 
    tinfo=0xfffff7a77c10 <typeinfo for std::out_of_range>, dest=0xfffff78c2be0 <std::out_of_range::~out_of_range()>)
    at ../../../../src/libstdc++-v3/libsupc++/eh_throw.cc:98
#8  0x0000fffff78a11a4 in std::__throw_out_of_range (__s=0xffffef50d6e8 "unordered_map::at")
    at ../../../../../src/libstdc++-v3/src/c++11/functexcept.cc:86
#9  0x0000ffffef4fd548 in std::__detail::_Map_base<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, rclcpp::ParameterValue>, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, rclcpp::ParameterValue> >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true>, true>::at (this=0xaaaaab0e9218, __k="ExposureTime")
    at /usr/include/c++/13/bits/hashtable_policy.h:789
#10 0x0000ffffef4fccec in std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, rclcpp::ParameterValue, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, rclcpp::ParameterValue> > >::at (
    this=0xaaaaab0e9218, __k="ExposureTime") at /usr/include/c++/13/bits/unordered_map.h:1004
#11 0x0000ffffef4fbffc in ParameterConflictHandler::resolve_overrides (this=0xaaaaab0e9218, 
    p=std::unordered_map with 5 elements = {...})
    at /home/dbacher/ros2_ws/src/camera_ros/src/ParameterConflictHandler.cpp:138
#12 0x0000ffffef4e6a58 in ParameterHandler::declare (this=0xaaaaab0e9210, controls=...)
    at /home/dbacher/ros2_ws/src/camera_ros/src/ParameterHandler.cpp:193
#13 0x0000ffffef41c770 in camera::CameraNode::CameraNode (this=0xaaaaab0e8a40, options=...)
    at /home/dbacher/ros2_ws/src/camera_ros/src/CameraNode.cpp:494
#14 0x0000ffffef4c392c in std::_Construct<camera::CameraNode, rclcpp::NodeOptions const&> (__p=0xaaaaab0e8a40)
    at /usr/include/c++/13/bits/stl_construct.h:119
#15 0x0000ffffef4c3318 in std::allocator_traits<std::allocator<void> >::construct<camera::CameraNode, rclcpp::NodeOptions const&> (__p=0xaaaaab0e8a40) at /usr/include/c++/13/bits/alloc_traits.h:661
#16 std::_Sp_counted_ptr_inplace<camera::CameraNode, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<rclcpp::NodeOptions const&> (this=0xaaaaab0e8a30, __a=...) at /usr/include/c++/13/bits/shared_ptr_base.h:604
#17 0x0000ffffef4c2d84 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<camera::CameraNode, std::allocator<void>, rclcpp::NodeOptions const&> (this=0xffffffffc920, __p=@0xffffffffc918: 0x0, __a=...)
    at /usr/include/c++/13/bits/shared_ptr_base.h:971
#18 0x0000ffffef4c2a20 in std::__shared_ptr<camera::CameraNode, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<void>, rclcpp::NodeOptions const&> (this=0xffffffffc918, __tag=...) at /usr/include/c++/13/bits/shared_ptr_base.h:1712
#19 0x0000ffffef4c2760 in std::shared_ptr<camera::CameraNode>::shared_ptr<std::allocator<void>, rclcpp::NodeOptions const&> (this=0xffffffffc918, __tag=...) at /usr/include/c++/13/bits/shared_ptr.h:464
#20 0x0000ffffef4c25a8 in std::make_shared<camera::CameraNode, rclcpp::NodeOptions const&> ()
    at /usr/include/c++/13/bits/shared_ptr.h:1010
#21 0x0000ffffef4c2430 in rclcpp_components::NodeFactoryTemplate<camera::CameraNode>::create_node_instance (
    this=0xaaaaab0e5920, options=...)
    at /opt/ros/jazzy/include/rclcpp_components/rclcpp_components/node_factory_template.hpp:45
#22 0x0000aaaaaaab8e34 in main (argc=8, argv=0xffffffffde18)
    at /home/dbacher/ros2_ws/build/camera_ros/rclcpp_components/node_main_camera_node.cpp:69
```

2. The steps to reproduce are just running the node with `ros2 run camera_ros camera_node`. The controls listed by `LIBCAMERA_LOG_LEVELS=*:DEBUG cam -l` are:
```
[3:46:57.346658673] [51130] DEBUG V4L2 v4l2_device.cpp:636 /dev/video0[20:cap]: Control: Brightness (0x00980900)
[3:46:57.346714414] [51130] DEBUG V4L2 v4l2_device.cpp:636 /dev/video0[20:cap]: Control: Contrast (0x00980901)
[... cut ...]
[3:46:57.346932747] [51130] DEBUG V4L2 v4l2_device.cpp:636 /dev/video0[20:cap]: Control: Focus, Automatic Continuous (0x009a090c)
[3:46:57.346950303] [51130] DEBUG V4L2 v4l2_device.cpp:636 /dev/video0[20:cap]: Control: Zoom, Absolute (0x009a090d)
[3:46:57.346981173] [51130] DEBUG V4L2 v4l2_videodevice.cpp:632 /dev/video0[20:cap]: Opened device usb-xhci-hcd.1-1: uvcvideo: Microsoft® LifeCam Cinema(TM):
```

3. The issue only started occurring once I update to camera_ros 0.4.0 and libcamera 0.5.0. It works on camera_ros 0.3.0 and libcamera 0.4.0.

christianrauch, 2025-05-08:
Thanks for the backtrace. I can now see that the issue is related to non-working conflict check for `ExposureTime` and `ExposureTimeMode`. 

This check:
https://github.com/christianrauch/camera_ros/blob/0b2c24fe204fb23cb8b2df6a71f47efbcae63418/src/ParameterConflictHandler.cpp#L108-L114
should evaluate to true, store the current `ExposureTime`, and disable it, before reading it from the storage again here:
https://github.com/christianrauch/camera_ros/blob/0b2c24fe204fb23cb8b2df6a71f47efbcae63418/src/ParameterConflictHandler.cpp#L134-L140

This check either falsely evaluates to "false", and the exposure is not stored before it is read again, or it correctly evaluates to "false" but we wrongly restore the `ExposureTime`. In either case, we access `ExposureTime` in `store` when it was not stored before, which causes this access error.

I assume this must be related to different default values.

Can you list the controls via `cam -c 1 --list-controls`? Or can you in another way, e.g. by printing additional debug information, provide details about the controls with their min, max and default values?

Also, can you reproduce the issue with older libcamera versions (0.4 instead of 0.5)?

christianrauch, 2025-05-08:
Since I am not able to reproduce this locally with the hardware I have, I won't be able to properly debug this.

I created documentation on how to generate core dumps (https://github.com/christianrauch/camera_ros/pull/121). If you can share a core dump of the crash, I will be able to inspect the state of the process much better. Please follow the new instructions added via the PR and share the core dump.

In a nutshell:
```sh
# install dbgsym packages
sudo apt install ros-$ROS_DISTRO-libcamera-dbgsym ros-$ROS_DISTRO-camera-ros-dbgsym
# enable core dumps
sudo sysctl -w kernel.core_pattern=/tmp/%e-%t.core
ulimit -c unlimited
# reproduce crash and generate core dump at /tmp/camera_node-*.core
LIBCAMERA_LOG_LEVELS=*:DEBUG ros2 run --prefix "gdb -ex 'set pagination off' -ex run -ex backtrace --args" camera_ros camera_node --ros-args --log-level camera:=debug
```

Then, upload the file `/tmp/camera_node-*.core`.

dbacher42, 2025-05-09:
Here's the output of `cam -c 1 --list-controls`:
```
[8:40:53.757889712] [38144]  WARN IPAManager ipa_manager.cpp:154 No IPA found in '/usr/lib/aarch64-linux-gnu/libcamera'
[8:40:53.757991027] [38144]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0
Using camera /base/axi/pcie@120000/rp1/usb@300000-1:1.0-045e:075d as cam0
Control: ExposureTime: [500..2000000]
Control: AeEnable: [false..true]
Control: Saturation: [0.000000..2.409639]
Control: Contrast: [0.500000..1.500000]
Control: Brightness: [-0.844262..1.000000]
```

The issue occurs on camera_ros version 0.4.0 regardless of the libcamera version.

I've attached the core dump below:

[camera_node-1746760844.zip](https://github.com/user-attachments/files/20114493/camera_node-1746760844.zip)

christianrauch, 2025-05-09:
Thanks for the additional information.

The `cam` command above is using an older libcamera version (0.2) and therefore still has the `AeEnable` control instead of `ExposureTimeMode`. Are you having the crashes with this libcamera version? The bloom package is using binary version 0.5 from the ROS repos. Can you clarify which versions of what you are using, where you get the binary packages from, or which source repos you use?

The core file and the log above tell me that you run this on `aarch64`? Are you using Ubuntu on a Raspberry Pi? Can you reproduce this on a regular `x86-64` system? Also, let me know which git tag or commit you are using if you compile from source.

dbacher42, 2025-05-09:
Sorry about the confusion. I had uninstalled the libcamera-tools debian package which for some reason uses libcamera v0.2.0, even though I had libcamera v0.5.0 installed. Possibly because I had specifically installed the ros-jazzy-libcamera package. I tried building libcamera v0.5.0 from source and the issue still occurs. Here's the new output from `cam -c 1 --list-controls`:
```
[0:12:16.608165025] [7912]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0+40-a7994150
Using camera /base/axi/pcie@120000/rp1/usb@300000-1:1.0-045e:075d as cam0
Control: [inout] libcamera::ExposureTimeMode:
  - ExposureTimeModeAuto (0)
  - ExposureTimeModeManual (1) [default]
Control: [inout] libcamera::ExposureTime: [500..2000000]
Control: [inout] libcamera::Saturation: [0.000000..2.409639]
Control: [inout] libcamera::Contrast: [0.500000..1.500000]
Control: [inout] libcamera::Brightness: [-0.844262..1.000000]
```
Since the error is occurring in the ParameterConflictHandler.cpp, I reverted to the last commit before that file was last changed, that being `4c98ed3`.  That solved the issue. The issue occurs in commits from `684f122` onward.

christianrauch, 2025-05-09:
Thanks a lot for all this information. Eventually, I was able to process the core dump on a Raspberry Pi. From the data in the core dump, and later your newer `cam` log, I could see that the default value for `ExposureTimeMode` in your case is `ExposureTimeModeManual`, where for most UVC cameras I used, it is `ExposureTimeModeAuto`.

I could reproduce your issue locally and fixed it in #123 . @dbacher42 Can you test if this PR fixes your issue?

dbacher42, 2025-05-09:
That worked! The issue no longer occurs and I was able to capture images from my camera like I could before. Thanks so much for your help!

closed 2025-05-09 by dbacher42

christianrauch, 2025-06-30:
FYI: The `Symbol ipaModuleInfo not found` and `v4l2-compat.so: IPA module has no valid info` "ERROR" is now gone since libcamera v0.5.1. See the [libcamera v0.5.1release notes](https://git.libcamera.org/libcamera/libcamera.git/tag/?h=v0.5.1):

> Notably for packagers - IPA modules now have their own subdirectory
> which should prevent undesirable surrupticious error messages that would
> occur if packagers choose to install the V4L2 adaptation layer in the
> same folder as the IPA modules.

PR #123 (2025-05-09): only restore 'ExposureTime' if it is not set
If the `ExposureTimeMode` is `ExposureTimeModeManual` by default, we are not encountering conflicts in the default configuration and hence we do not remove the `ExposureTime` from the default parameter values in `resolve_defaults` and store it. In `resolve_overrides` we only check `ExposureTimeModeManual` to decide if we have to restore the `ExposureTime`
https://github.com/christianrauch/camera_ros/blob/ab763a25646b1c1f87d7856a5c1b732672c13903/src/ParameterConflictHandler.cpp#L134-L140 assuming that if `ExposureTimeModeManual` is `ExposureTimeModeM [... cut]