christianrauch/camera_ros #125: Build error on ROS2 Jazzy when building with -O3

0x62, 2025-05-31:
I'm getting a build error on ROS2 Jazzy. It looks like the issue was introduced in 72a74fff4c269d5086caea2ee8344c7ea0a6e131

```
[Processing: camera_ros, rviz_default_plugins, test_communication, test_rclcpp]
--- stderr: camera_ros
In file included from /usr/include/c++/12/ios:40,
                 from /usr/include/c++/12/ostream:38,
                 from /usr/include/c++/12/iostream:39,
                 from /home/ben/ros2/install/rclcpp/include/rclcpp/rclcpp/parameter.hpp:18,
                 from /home/ben/ros2/src/christianrauch/camera_ros/src/ParameterConflictHandler.hpp:2,
                 from /home/ben/ros2/src/christianrauch/camera_ros/src/ParameterHandler.hpp:2,
                 from /home/ben/ros2/src/christianrauch/camera_ros/src/ParameterHandler.cpp:1:
In static member function ‘static constexpr std::char_traits<char>::char_type* std::char_traits<char>::copy(char_type*, const char_type*, std::size_t)’,
    inlined from ‘static constexpr void std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::_S_copy(_CharT*, const _CharT*, size_type) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]’ at /usr/include/c++/12/bits/basic_string.h:423:21,
    inlined from ‘constexpr std::__cxx11::basic_string<_CharT, _Traits, _Allocator>& std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::_M_replace(size_type, size_type, const _CharT*, size_type) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]’ at /usr/include/c++/12/bits/basic_string.tcc:532:22,
    inlined from ‘constexpr std::__cxx11::basic_string<_CharT, _Traits, _Alloc>& std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::replace(size_type, size_type, const _CharT*, size_type) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]’ at /usr/include/c++/12/bits/basic_string.h:2171:19,
    inlined from ‘constexpr std::__cxx11::basic_string<_CharT, _Traits, _Alloc>& std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::insert(size_type, const _CharT*) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]’ at /usr/include/c++/12/bits/basic_string.h:1928:22,
    inlined from ‘constexpr std::__cxx11::basic_string<_CharT, _Traits, _Allocator> std::operator+(const _CharT*, __cxx11::basic_string<_CharT, _Traits, _Allocator>&&) [with _CharT = char; _Traits = char_traits<char>; _Alloc = allocator<char>]’ at /usr/include/c++/12/bits/basic_string.h:3541:36,
    inlined from ‘rcl_interfaces::msg::SetParametersResult format_result(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&)’ at /home/ben/ros2/src/christianrauch/camera_ros/src/ParameterHandler.cpp:38:30:
/usr/include/c++/12/bits/char_traits.h:431:56: error: ‘void* __builtin_memcpy(void*, const void*, long unsigned int)’ accessing 9223372036854775810 or more bytes at offsets [2, 9223372036854775807] and 1 may overlap up to 9223372036854775813 bytes at offset -3 [-Werror=restrict]
  431 |         return static_cast<char_type*>(__builtin_memcpy(__s1, __s2, __n));
      |                                        ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
gmake[2]: *** [CMakeFiles/param.dir/build.make:146: CMakeFiles/param.dir/src/ParameterHandler.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:169: CMakeFiles/param.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```

The build command I used is:

```
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-march=armv8-a+crc -mtune=cortex-a72 -O3" -DCMAKE_C_FLAGS="-march=armv8-a+crc -mtune=cortex-a72 -O3"
```

Updating `format_result` to concatenate the string more verbosely fixed it:

```
if (msgs.size() > 1) {
  result.reason += "("; 
  result.reason += std::to_string(i); 
  result.reason += ") ";
}
```

christianrauch, 2025-05-31:
I cannot reproduce this on `x86_64` with `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-O3" -DCMAKE_C_FLAGS="-O3"`. Could it be that this is something specific to the architecture or compiler? Can you reproduce this on the CI?

christianrauch, 2025-06-20:
@0x62 I can't reproduce this with the `-O3` and the ROS build farm can't reproduce this on the arm64 builds either. Can you identify which compiler flags reproduces the issue and reproduce this on the CI?

0x62, 2025-06-29:
I encountered this building on a Pi 5. I've been unable to reproduce this on CI or x86_64. I did a full update and rebuild last night, which was successful with the same build command so will close this.

closed 2025-06-29 by 0x62